aaronwmorris/indi-allsky #1257: Fisheye/Panorama not flipped

RoySalisbury, 2024-05-12:
I have both "Flip Image Vertically" and "Flip Image Horizontally" enabled for my camera.  However, it does not look like the panorama generation takes these settings into account. 

[image]

Also, as a "feature" it might be nice to add the cardinal markers (N/S/E/W) at the bottom of the image.

Thanks.

aaronwmorris, 2024-05-12:
The panorama is generated after the image is flipped, so the flipped image is used.

I will look into adding the cardinal directions to the panorama.

RoySalisbury, 2024-05-13:
I guess it just looks odd.  I have the panorama set so that the center is looking south.  IN that case, I would expect the sun/moon to rise from the left -> right (east -> west).  However, the image is flipped ... It goes from right -> left.

The cardinal directions for this image would be N-W-S-E-N .. and would expect it to be N-E-S-W-N ...

[image]

aaronwmorris, 2024-05-13:
I never really paid attention to how it was generated.  This is just how the [fish2pano](https://github.com/bluthen/fish2pano) python module works.

RoySalisbury, 2024-05-13:
I took a look at that project.  In it he has 2 sample images.  From them you can tell that the result is actually a mirror image (the text on the image is mirrored).  

Wonder if you can just do a mirror of the result to get it back to normal.  Add a switch to the config to enable/disable it.

aaronwmorris, 2024-05-13:
I'm guessing it is just reading the image counter-clockwise.  I'll look into adding an option to flip the image horizontally.

aaronwmorris, 2024-05-14:
Nevermind, after reviewing the images, its reading them clockwise.

aaronwmorris, 2024-05-15:
Merged #1264 with the ability to add cardinal directions to panoramas.  There are options to flip the panorama as well.

I had to setup some special test images to figure out details about the panorama.  The first important point was the the starting point of the panorama, which is 90 degrees clockwise and the progression is clockwise from that point.  The image needs to be rotated -90 degrees to get the bottom of the image to be the center of the panorama.

RoySalisbury, 2024-05-17:
I've updated to the latest code.  It looks good so far.  

Thanks for the update!

closed 2024-05-26 by aaronwmorris