aaronwmorris/indi-allsky #1258: Fisheye/Panorama usable image area

RoySalisbury, 2024-05-12:
Not sure at what point in the processing your generating the panorama or how much of the original image.  However, might I suggest that you limit the panorama generation to only the area defined by the indicated image circle.  There seems to be a lot of "dead space" at the bottom of the image but if the panorama was limited to the image circle (even if it the square boundries of the circle) there might be less dead space.

[image]

aaronwmorris, 2024-05-12:
The settings for the panorama (under the Processing tab) allow you to set a size for the image circle.

RoySalisbury, 2024-05-13:
OK.. I spent about 30 min tweaking the panorama settings .. finally got something thats much better.  Its kinda confusing since there are basically 3 different areas to setup the image circle and they are all just slightly different.

I set the center of t panorama image to be due south.

[image]

aaronwmorris, 2024-05-13:
Yes, I agree it can be confusing with so many different image circle options, however, each option has their own uses.

There may be cases where you want the circle for the panorama to be different from the cardinal direction if you want to offset each by a different value.

RoySalisbury, 2024-05-17:
Your comments here along with the update in #1257 I think resolves this issue.

closed 2024-05-17 by RoySalisbury