aaronwmorris/indi-allsky #1268: Hello. I'm not sure if I can write here or better opening a new thread.

aaronwmorris, 2024-05-18:
Hello. I'm not sure if I can write here or better opening a new thread.
I have exactly the same issue with the same camera after 29/04, and Ubuntu 20.04 LTS
Now I have "No image FOR 15 MIN" since then.

I installed the ASI Studio software and the camera works fine. I got images etc.
It's just the Indi AllSky software that's not working.
It has been working perfectly for years!

Any idea?

_Originally posted by @Errabundo66 in https://github.com/aaronwmorris/indi-allsky/issues/1189#issuecomment-2118396658_

aaronwmorris, 2024-05-18:
Can you provide the output of `./misc/support_info.sh`?

Errabundo66, 2024-05-18:
Hi Aaron. 
There’s no need. 
As I post in another thread, I solved it with your CLI instruction timedatectl set-ntp true. !

I changed some weeks ago the time/hour at the config site in order to show UTC time. (In Spain we are at GMT+2). And it seems to be the cause of the error. 

Once the NTP server is enabled again, all works again. 
I also updated to the indi-AllSky last version. 

Thanks again!
Fernando

closed 2024-05-18 by aaronwmorris