aaronwmorris/indi-allsky #1270: Trying to do a new install on a Pi 3b+ libopenexr-dev : Breaks: libilmbase-dev (< 2.5.7-3~)

ikeysolomon, 2024-05-25:
Hi I am trying install on a Pi 3b+ with zwo asi120mc 
am receiving the following error: 

Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libimath-dev : Conflicts: libilmbase-dev
 libopenexr-dev : Breaks: libilmbase-dev (< 2.5.7-3~)
E: Unable to correct problems, you have held broken packages.

Any clues on this one? 
I would like to try your software as the allsky by Thomas Jacquin seems to give me the dreaded bug associated with the zwo asi120mc. 
Thank you for your time.

aaronwmorris, 2024-05-26:
You may be using Raspberry Pi OS 12 (bookworm) 32-bit which has some package dependency problems.  Please try the 64-bit version.

ikeysolomon, 2024-05-26:
Thank you this solved my problem. Would you mind updating releasing notes. Perhaps this was noted and I missed it however.

aaronwmorris, 2024-05-26:
There is a note on the main page, but I will try to sprinkle some more warnings in the wiki.  I was hoping the dependency problem would have been resolved by now.

closed 2024-05-28 by ikeysolomon