aaronwmorris/indi-allsky #1271: ZWO drivers.

ikeysolomon, 2024-05-26:
Hi,
I have a Zwo asi120mc I am trying to get going.
Installing the latest release, I  am given two options  asi single and asi all. Unfortunately I chose asi single. When I went into the local page after install It said that no camera was detected. 

Is there a way to install the other asi driver without too much fuss? 

Thank you.... (getting closer haha)

aaronwmorris, 2024-05-26:
You can re-run setup.sh to change the driver being used.  The ASI120 should work with either indi_asi_ccd or indi_asi_single_ccd.

I would reboot after changing the driver.  Also, if you disconnect the camera, you will need to restart the indiserver to re-initialize the connection to the camera.

```
systemctl --user restart indiserver
```

ikeysolomon, 2024-05-27:
Rebooting solved the issue, thanks heaps!

closed 2024-05-27 by ikeysolomon