aaronwmorris/indi-allsky #1272: Timelapse issue

clambert24, 2024-05-27:
Hi All. 
I have just changed from the TJ software to the indi one by aaronwmorris due to having problems viewing timelapse videos etc. 
This one works better, but I noticed that the video was only about 2 hours long...
Another thing I have is that it came up with "Image queue exceeded maximum threshold depth."
What could cause this or how to rectify?
[image]

aaronwmorris, 2024-05-27:
I saw you post on Facebook maybe 10 minutes after you posted, but I am unable to post anything to the Allsky group for some reason.

Basically, indi-allsky uses a queue to process images as they are taken.  Normally, images are processed fairly quickly, so the queue size is almost always 0, but if the system slows down, the queue can increase and this is just a warning regarding a possible slow down.

Can you post the output of the script `./misc/support_info.sh`?

clambert24, 2024-05-27:
Ok, I hate to ask but how do I do that? 
I'm a complete novice when it comes to things like this...🙄

aaronwmorris, 2024-05-27:
Within the indi-allsky folder, just run `./misc/support_info.sh` and just copy the output here.

clambert24, 2024-05-27:
ok, so managed to run that once i found the folder, but then the termal disappeared . did it save the log somewhere?

aaronwmorris, 2024-05-27:
You can run it as many times as you need.  I am going to work on a web based interface for this as well.

clambert24, 2024-05-27:
Yeah I got that, but where is the output from the thing you told me to run do I can copy and paste it?

aaronwmorris, 2024-05-27:
It is usually displayed in the output of the command in the terminal.

clambert24, 2024-05-27:
thats the thing. i execute the command in a terminal, it outputted lots of info, then when it got to the end the terminal just closed. 
ok so tried just opening it, and it gave this info now. 

#!/bin/bash

#set -x  # command tracing
set -o errexit
set -o nounset

PATH=/usr/local/bin:/usr/bin:/bin
export PATH

function catch_error() {
    echo
    echo
    echo "\`\`\`"  # markdown
    echo "###############"
    echo "###  ERROR  ###"
    echo "###############"
    echo
    echo "The script exited abnormally, please try to run again..."
    echo
    echo
    exit 1
}
trap catch_error ERR

function catch_sigint() {
    echo
    echo
    echo "\`\`\`"  # markdown
    echo "###############"
    echo "###  ERROR  ###"
    echo "###############"
    echo
    echo "The script was interrupted, please run the script again to finish..."
    echo
    echo
    exit 1
}
trap catch_sigint SIGINT

if [ ! -f "/etc/os-release" ]; then
    echo
    echo "Unable to determine OS from /etc/os-release"
    echo
    exit 1
fi

source /etc/os-release

DISTRO_ID="${ID:-unknown}"
DISTRO_VERSION_ID="${VERSION_ID:-unknown}"
CPU_ARCH=$(uname -m)
CPU_BITS=$(getconf LONG_BIT)
CPU_TOTAL=$(grep -c "^proc" /proc/cpuinfo)
MEM_TOTAL=$(grep MemTotal /proc/meminfo | awk "{print \$2}")

if [ -f "/proc/device-tree/model" ]; then
    SYSTEM_MODEL=$(cat /proc/device-tree/model)
else
    SYSTEM_MODEL="Generic PC"
fi

if which indiserver >/dev/null 2>&1; then
    INDISERVER=$(which indiserver)
else
    INDISERVER="not found"
fi

SCRIPT_DIR=$(dirname "$0")
cd "$SCRIPT_DIR/.."
ALLSKY_DIRECTORY=$PWD
cd "$OLDPWD"

# go ahead and prompt for password
sudo true

echo "#################################"
echo "### indi-allsky support info  ###"
echo "#################################"

sleep 3

echo "\`\`\`"  # markdown
echo
echo "Distribution: $DISTRO_ID"
echo "Release: $DISTRO_VERSION_ID"
echo "Arch: $CPU_ARCH"
echo "Bits: $CPU_BITS"
echo
echo "CPUs: $CPU_TOTAL"
echo "Memory: $MEM_TOTAL kB"
echo
echo "System: $SYSTEM_MODEL"

echo
uname -a

echo
echo "Uptime"
uptime

echo
echo "Memory"
free

echo
echo "Filesystems"
df -k

echo
echo "sysctl info"
sudo sysctl vm.swappiness

echo
echo "system python: $(python3 -V)"

echo
echo "indiserver: $INDISERVER"
echo

if [ -f "/etc/astroberry.version" ]; then
    echo "Detected Astroberry server"
    echo
fi

echo
echo "User info"
id
echo

echo "Process info"
# shellcheck disable=SC2009
ps auxwww | grep indi | grep -v grep || true
echo

echo "USB info"
lsusb
echo

echo "USB Permissions"
find /dev/bus/usb -ls || true
echo

echo "video device Permissions"
ls -l /dev/video* || true
echo

echo "v4l info"
v4l2-ctl --list-devices || true
echo

echo "Module info"
lsmod
echo

echo "git status"
git status | head -n 100
echo

echo "git log"
git log -n 1 | head -n 100
echo

if pkg-config --exists libindi; then
    DETECTED_INDIVERSION=$(pkg-config --modversion libindi)
    echo "indi version: $DETECTED_INDIVERSION"
    echo
else
    echo "indi version: not detected"
    echo
fi

echo "indi packages"
dpkg -l | grep libindi || true
echo

echo "indi connections"
ss -ant | grep 7624 || true
echo

if which indi_getprop >/dev/null 2>&1; then
    echo "Detected indi properties"
    indi_getprop -v 2>&1 || true
    echo
fi

if pkg-config --exists libcamera; then
    DETECTED_LIBCAMERA=$(pkg-config --modversion libcamera)
    echo "libcamera version: $DETECTED_LIBCAMERA"
    echo
else
    echo "libcamera: not detected"
    echo
fi

echo "libcamera packages"
dpkg -l | grep libcamera || true
echo

echo "libcamera cameras"
if which rpicam-hello >/dev/null 2>&1; then
    echo "rpicam-hello: $(which rpicam-hello)"
    rpicam-hello --list-cameras || true
    echo
elif which libcamera-hello >/dev/null 2>&1; then
    echo "libcamera-hello: $(which libcamera-hello)"
    libcamera-hello --list-cameras || true
    echo
else
    echo "libcamera-hello not available"
    echo
fi

echo "python packages"
dpkg -l | grep python || true
echo

if [ -d "${ALLSKY_DIRECTORY}/virtualenv/indi-allsky" ]; then
    echo "Detected indi-allsky virtualenv"

    # shellcheck source=/dev/null
    source "${ALLSKY_DIRECTORY}/virtualenv/indi-allsky/bin/activate"
    echo "virtualenv python: $(python3 -V)"
    echo "virtualenv PATH: $PATH"

    if which flask >/dev/null 2>&1; then
        echo "flask command: $(which flask)"
    else
        echo "flask: not found"
    fi

    echo
    echo "virtualenv python modules"
    pip freeze

    echo "\`\`\`"  # markdown

    echo
    echo "indi-allsky config (passwords redacted)"
    INDI_ALLSKY_CONFIG=$("${ALLSKY_DIRECTORY}/config.py" dump)

    echo "\`\`\`json"  # markdown
    # Remove all secrets from config
    echo "$INDI_ALLSKY_CONFIG" | jq --arg redacted "REDACTED" '.FILETRANSFER.PASSWORD = $redacted | .FILETRANSFER.PASSWORD_E = $redacted | .S3UPLOAD.SECRET_KEY = $redacted | .S3UPLOAD.SECRET_KEY_E = $redacted | .MQTTPUBLISH.PASSWORD = $redacted | .MQTTPUBLISH.PASSWORD_E = $redacted | .SYNCAPI.APIKEY = $redacted | .SYNCAPI.APIKEY_E = $redacted | .PYCURL_CAMERA.PASSWORD = $redacted | .PYCURL_CAMERA.PASSWORD_E = $redacted'

    deactivate
    echo
else
    echo "indi-allsky virtualenv is not created"
    echo
fi

echo "\`\`\`"  # markdown
echo "#################################"
echo "###     end support info      ###"
echo "#################################"

aaronwmorris, 2024-05-27:
Okay, lets try this:

```
./misc/support_info.sh > support_info.txt
```

That should write the file to `support_info.txt`

clambert24, 2024-05-28:
Hi. 
Ok obviously I'm being completely thick and don't understand where that file is, or where to type what you just said. 
[image]

aaronwmorris, 2024-05-28:
I just added a web view to be able to query the support info.  You would just need to upgrade to the latest code.

https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky

clambert24, 2024-05-28:
ok. upgraded and managed to run the support file. this is what it said. 

/home/pi/indi-allsky/misc/support_info.sh: line 61: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 929096 kB

System: Raspberry Pi 3 Model B Rev 1.2

Linux Allsky 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux

Uptime
 21:00:35 up 23:23,  2 users,  load average: 8.17, 5.24, 3.62

Memory
               total        used        free      shared  buff/cache   available
Mem:          929096      583216       58744        2512      355440      345880
Swap:        4194300      357632     3836668

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev              322448        0    322448   0% /dev
tmpfs              92912     1220     91692   2% /run
/dev/mmcblk0p2  29462984 17818252  10128036  64% /
tmpfs             464548      168    464380   1% /dev/shm
tmpfs               5120       12      5108   1% /run/lock
tmpfs              76800        0     76800   0% /home/pi/allsky/tmp
/dev/mmcblk0p1    522230    76382    445848  15% /boot/firmware
tmpfs              92908       44     92864   1% /run/user/1000

sysctl info
vm.swappiness = 60

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

User info
uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),33(www-data),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),115(lpadmin),993(gpio),994(i2c),995(spi),999(systemd-journal)

Process info
pi           990  0.2  0.1   9620  1536 ?        Ss   May27   3:31 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_simulator_ccd
pi           994  0.1  0.2  41236  2176 ?        S    May27   2:31 indi_simulator_telescope
pi           995  0.1  0.2 123424  2048 ?        Sl   May27   2:02 indi_simulator_ccd
pi        372176  6.3  2.4 580740 22576 ?        SNsl 20:56   0:16 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
pi        372235  1.1  1.9 499664 18340 ?        Rl   20:56   0:02 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
pi        372237 44.1 22.1 766720 206024 ?       Rl   20:56   1:44 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
pi        372245  0.0  1.2 382036 11284 ?        SN   20:56   0:00 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
pi        372697  1.4  2.2  33388 21068 ?        SNs  20:58   0:01 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
pi        372737 13.4 12.6 578616 117736 ?       SNl  20:58   0:15 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
pi        373028  0.3  0.3   6800  2944 ?        SN   21:00   0:00 /bin/bash /home/pi/indi-allsky/misc/support_info.sh

USB info
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

USB Permissions
      124      0 drwxr-xr-x   3 root     root           60 Jan  1  1970 /dev/bus/usb
      125      0 drwxr-xr-x   2 root     root          100 Jan  1  1970 /dev/bus/usb/001
      126      0 crw-rw-r--   1 root     root     189,   0 May 27 21:37 /dev/bus/usb/001/001
[... 1018 lines cut ...]
    "GPIO_PIN_3": "D27",
    "GPIO_PIN_4": "D22"
  },
  "DEW_HEATER": {
    "CLASSNAME": "",
    "ENABLE_DAY": false,
    "PIN_1": "D12",
    "LEVEL_DEF": 100
  },
  "TEMP_SENSOR": {
    "A_CLASSNAME": "",
    "A_PIN_1": "D5",
    "A_USER_VAR_SLOT": 10,
    "A_I2C_ADDRESS": "0x77",
    "B_CLASSNAME": "",
    "B_PIN_1": "D6",
    "B_USER_VAR_SLOT": 15,
    "B_I2C_ADDRESS": "0x76"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-05-28:
Excellent.  Everything looks good there.

Do you have a heatsink and fan attached?

One thing that can happen is if the CPU goes above 80, it will start to severely throttle your CPU performance.  When you look at the `Image Lag` view, what does it say about your image processing times?

clambert24, 2024-05-28:
No not got a fan etc, can't see me needing one as I don't think it gets that hot. Please see attached pics for reference.  
[image]
[image]

aaronwmorris, 2024-05-28:
Do you run your raspberry outside with the camera or is it inside?

aaronwmorris, 2024-05-29:
Keep in mind that the temperature in the picture is the temperature of the camera, not the temperature of the CPU.  The temperature of the CPU is going to be probably 20-30 degrees higher, at least.

When the timelapse is generated, the CPU utilization and heat go up considerably, which can cause throttling, which will slow down image processing (which happens concurrently with the timelapse generation).

clambert24, 2024-05-29:
It's out side. 
There is a hole in the bottom where I made a mistake of my cable entry so just left it open. Ok that's interesting to know about the heat, I didn't realise it could get that hot, and what being outside I thought with wind etc it would cool it down enough.  See attached pic of my setup. 
If you think it needs a fan, any recommendations on type of fan or what connections to use etc? 
[image]

aaronwmorris, 2024-05-30:
You definitely need a fan and heatsink, if only for those times you are generating the timelapse.  I just use the cheapest 5v fan and heatsink that can be purchased on Amazon or eBay.

clambert24, 2024-06-02:
Ok. 
So iv added a heat sink and fan, yet the system is still only generating about 2 hours worth of timelapse, yet if I go into images it looks like it has taken pictures all throughout the night 🤔

aaronwmorris, 2024-06-02:
Are you located in the US South?  Just double checking your latitude and longitude.

clambert24, 2024-06-02:
Nope. I'm in the UK. Is it saying its wrong? If so that might be why it's starting/finishing at weird times

aaronwmorris, 2024-06-02:
The Support info has latitude 33 and longitude -84.  You can correct that in the `Location` tab

aaronwmorris, 2024-06-03:
The two things you need to verify, your system timezone and the latitude and longitude.  Also, make sure you have not transposed latitude and longitude.

If you have to change either of these, I would suggest a full reboot.

Once changed, it may take 12 hours to fully correct.  Depending on the direction of the time change, older images may be mixed in with the next timelapse.

clambert24, 2024-06-04:
Ok, so it now looks like it has sorted itself out with time zone etc and last night I had a recording from 10 PMto 3:50 AM. 
Is there a way to define the hours it records, or is it automatically set with sunrise/sunset etc?

clambert24, 2024-06-04:
https://github.com/aaronwmorris/indi-allsky/assets/83610871/3cdfc44f-0637-4b47-a051-107b4113f7b3

aaronwmorris, 2024-06-04:
The start and stop time is based on sunrise and sunset.  The default setting for switching between night and day is when the sun is -6 degrees below the horizon.  This is configurable on the `Location` tab as the `Sun altitude`

ikeysolomon, 2024-06-12:
In the example video you posted above there seems to be some "rocking" in the beginning rather  than a smooth transition. I noticed this too only a couple of my timelapses. I don't have it checked in the settings but seems to happen still to some degree. Any hope of ironing out?

aaronwmorris, 2024-06-13:
@ikeysolomon This behavior is normal because the exposure algorithm is adjusting the exposure based on the changes to the sky brightness.  The exposure algorithm is **reactive** so the exposure follows changes to brightness.

Because this transition is the day/night boundary, there are continuous massive changes to brightness and you get this "stepped" exposure.  There are practically no clouds, so this is the worst case scenario (clouds are a source of brightness for most people living closer to cities).

There is a way improve this by using a detection mask.  The detection mask can outline the areas of the image that are the true "sky" so the software can fully measure the whole sky brightness.  The default detection area is the central 50% of the image.  Since the sun rising/setting at the edge of the image, the edge changes brightness before the center and it can delay the change to the exposure.

closed 2024-09-14 by aaronwmorris