aaronwmorris/indi-allsky #1283: Daytime purple hue.

clambert24, 2024-05-29:
Any ideas how I can reduce the purple on the daytime images? 
[image]

aaronwmorris, 2024-05-29:
What kind of camera are you using?

clambert24, 2024-05-29:
[image]

aaronwmorris, 2024-05-29:
Do you happen to know if you got the version with the IR-Cut filter or not?  This could be due to the IR-cut filter not being on the camera.  There is a different settings file that can be set for better color balance if that is the case.

The other possibility is that the wrong Bayer pattern is being used.  I am assuming you selected IMX477 as the driver?

clambert24, 2024-05-29:
I'm not sure about what version it is, I'm guessing it's not. 
Yes I wasn't sure what version to use, but using Google it told me to use the IMX477

aaronwmorris, 2024-05-30:
If you are using the Raspi 3 for this, try adding the following to the `Night libcamera extra options` and `Day libcamera extra options` options on the Camera tab.  Then restart.

```
--tuning-file /usr/share/libcamera/ipa/rpi/vc4/imx477_noir.json
```

clambert24, 2024-05-30:
Ok, so I copied and pasted that as instructed,  but it's still purple 😔
[image]

aaronwmorris, 2024-05-30:
Can you send me the output of `./misc/support_info.sh`?  If you have the latest code, there is also a web view that can generate this info under `Tools`.

clambert24, 2024-05-30:
/home/pi/indi-allsky/misc/support_info.sh: line 61: warning: command substitution: ignored null byte in input ################################# ### indi-allsky support info ### ################################# ``` Distribution: debian Release: 12 Arch: aarch64 Bits: 64 CPUs: 4 Memory: 929100 kB System: Raspberry Pi 3 Model B Rev 1.2 Linux Allsky 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux Uptime 18:39:31 up 21:32, 2 users, load average: 3.38, 2.83, 2.57 Memory total used free shared buff/cache available Mem: 929100 384152 325324 312 285740 544948 Swap: 4194300 553472 3640828 Filesystems Filesystem 1K-blocks Used Available Use% Mounted on udev 322452 0 322452 0% /dev tmpfs 92912 1188 91724 2% /run /dev/mmcblk0p2 29462984 24303304 3642984 87% / tmpfs 464548 172 464376 1% /dev/shm tmpfs 5120 12 5108 1% /run/lock tmpfs 76800 0 76800 0% /home/pi/allsky/tmp /dev/mmcblk0p1 522230 76382 445848 15% /boot/firmware tmpfs 92908 44 92864 1% /run/user/1000 sysctl info vm.swappiness = 60 system python: Python 3.11.2 indiserver: /usr/local/bin/indiserver User info uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),33(www-data),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),115(lpadmin),993(gpio),994(i2c),995(spi),999(systemd-journal) Process info pi 884 0.1 0.2 9620 2048 ? Ss May29 2:27 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_simulator_ccd pi 994 0.1 0.2 41236 2432 ? S May29 1:40 indi_simulator_telescope pi 995 0.1 0.2 123424 2560 ? Sl May29 1:23 indi_simulator_ccd pi 2310 0.0 2.1 800828 19920 ? SNsl May29 1:14 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run pi 4910 0.0 0.3 33388 3404 ? SNs May29 0:21 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi pi 4915 0.1 4.1 932092 38416 ? SNl May29 1:25 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi pi 141608 0.5 2.4 802116 23060 ? SNl 17:31 0:23 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run pi 141609 44.6 20.0 984188 186064 ? RNl 17:31 30:27 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run pi 141610 0.3 6.6 811096 61560 ? SN 17:31 0:12 /home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run pi 150589 0.4 0.3 6800 3072 ? SN 18:39 0:00 /bin/bash /home/pi/indi-allsky/misc/support_info.sh USB info Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub USB Permissions 124 0 drwxr-xr-x 3 root root 60 Jan 1 1970 /dev/bus/usb 125 0 drwxr-xr-x 2 root root 100 Jan 1 1970 /dev/bus/usb/001 126 0 crw-rw-r-- 1 root root 189, 0 May 29 21:06 /dev/bus/usb/001/001 141 0 crw-rw-r-- 1 root root 189, 1 May 29 21:06 /dev/bus/usb/001/002 151 0 crw-rw-r-- 1 root root 189, 2 May 29 21:06 /dev/bus/usb/001/003 video device Permissions crw-rw----+ 1 root video 81, 13 May 29 21:06 /dev/video0 crw-rw----+ 1 root video 81, 14 May 29 21:06 /dev/video1 crw-rw----+ 1 root video 81, 0 May 29 21:06 /dev/video10 crw-rw----+ 1 root video 81, 1 May 29 21:06 /dev/video11 crw-rw----+ 1 root video 81, 3 May 29 21:06 /dev/video12 crw-rw----+ 1 root video 81, 2 May 29 21:06 /dev/video13 crw-rw----+ 1 root video 81, 4 May 29 21:06 /dev/video14 crw-rw----+ 1 root video 81, 5 May 29 21:06 /dev/video15 crw-rw----+ 1 root video 81, 7 May 29 21:06 /dev/video16 crw-rw----+ 1 root video 81, 6 May 29 21:06 /dev/video18 crw-rw----+ 1 root video 81, 9 May 29 21:06 /dev/video20 crw-rw----+ 1 root video 81, 10 May 29 21:06 /dev/video21 crw-rw----+ 1 root video 81, 11 May 29 21:06 /dev/video22 crw-rw----+ 1 root video 81, 12 May 29 21:06 /dev/video23 crw-rw----+ 1 root video 81, 8 May 29 21:06 /dev/video31 v4l info unicam (platform:3f801000.csi): 	/dev/video0 	/dev/video1 	/dev/media3 bcm2835-codec-decode (platform:bcm2835-codec): 	/dev/video10 	/dev/video11 	/dev/video12 	/dev/video18 	/dev/video31 	/dev/media1 bcm2835-isp (platform:bcm2835-isp): 	/dev/video13 	/dev/video14 	/dev/video15 	/dev/video16 	/dev/video20 	/dev/video21 	/dev/video22 	/dev/video23 	/dev/media0 	/dev/media2 Module info Module Size Used by tcp_diag 12288 0 inet_diag 24576 1 tcp_diag rfcomm 53248 4 snd_seq_dummy 12288 0 snd_hrtimer 12288 1 snd_seq 81920 7 snd_seq_dummy snd_seq_device 16384 1 snd_seq cmac 12288 3 algif_hash 12288 1 aes_arm64 12288 3 aes_generic 32768 1 aes_arm64 algif_skcipher 12288 1 af_alg 28672 6 algif_hash,algif_skcipher bnep 24576 2 brcmfmac_wcc 12288 0 brcmfmac 348160 1 brcmfmac_wcc brcmutil 24576 1 brcmfmac cfg80211 995328 1 brcmfmac imx477 24576 3 bcm2835_unicam 49152 8 hci_uart 49152 0 binfmt_misc 16384 1 btbcm 24576 1 hci_uart raspberrypi_hwmon 12288 0 bluetooth 606208 33 hci_uart,btbcm,bnep,rfcomm v4l2_dv_timings 40960 1 bcm2835_unicam v4l2_fwnode 20480 2 imx477,bcm2835_unicam v4l2_async 20480 3 v4l2_fwnode,imx477,bcm2835_unicam bcm2835_v4l2 40960 0 bcm2835_codec 45056 0 bcm2835_isp 28672 10 ecdh_generic 16384 2 bluetooth v4l2_mem2mem 53248 1 bcm2835_codec bcm2835_mmal_vchiq 36864 3 bcm2835_codec,bcm2835_v4l2,bcm [... cut ...]  "D12", "LEVEL_DEF": 100, "THOLD_ENABLE": false, "MANUAL_TARGET": 0, "TEMP_USER_VAR_SLOT": 10, "LEVEL_LOW": 33, "LEVEL_MED": 66, "LEVEL_HIGH": 100, "THOLD_DIFF_LOW": 15, "THOLD_DIFF_MED": 10, "THOLD_DIFF_HIGH": 5 }, "TEMP_SENSOR": { "A_CLASSNAME": "", "A_PIN_1": "D5", "A_USER_VAR_SLOT": 10, "A_I2C_ADDRESS": "0x77", "B_CLASSNAME": "", "B_PIN_1": "D6", "B_USER_VAR_SLOT": 15, "B_I2C_ADDRESS": "0x76" } } ``` ################################# ### end support info ### #################################

aaronwmorris, 2024-05-30:
Great, thanks!  If you could enable FITS image capture for a few frames and zip up the file and post it here, I would like to look at the raw data.

clambert24, 2024-05-30:
ok. so i enabled FITS image capture for about 15 minutes, but now dont know where that file got saved to

aaronwmorris, 2024-05-31:
You can look through the `Images` view under Media around the time period where you captured the FITS files.  If a FITS file was captured, there will be a link under the image you may use to download.

clambert24, 2024-05-31:
hi. i hope you can see this and it works?

[ccd2_20240531_190627.zip](https://github.com/user-attachments/files/15517714/ccd2_20240531_190627.zip)

aaronwmorris, 2024-05-31:
It is possible the wrong bayer pattern is being used.  You can test this yourself by going to the FITS Processor and test different debayer patterns.

The correct pattern for the IMX477 is BGGR and I can roughly replicate the reddish hue by selecting GRBG or GBRG.  I do not understand why your system would be selecting the wrong pattern.  In the config you can override the `Bayer Pattern` by changing "Auto Detect" to BGGR on the Image tab and see if that corrects the problem.

Again, this does not make sense because the FITS file you provided has BGGR in the FITS header and the resulting image has a green bias (can be fixed with SCNR), which is what I would expect for your camera.

clambert24, 2024-06-01:
ok, so this is confusing, i changed the bayer pattern to BGGR and that didnt work, so i randomly dissabled SCNR and now the image is correct and working ok. i had turned that to average neutral to cure a green issue. very odd but its now ok. thank you for your help

[image]

closed 2024-07-17 by aaronwmorris