aaronwmorris/indi-allsky #1290: Creating Synthetic Exposure/FITS file by adding successive exposures

gaitskell, 2024-06-01:
We have been looking in detail at the photometry from our Oculus Pro Camera (Starlight Xpress) with 150 deg FOV.
The typical light entering the camera from an apparent magnitude 0 star (Arcturus, Vega) will cause the brightest pixel to enter the non-linear regime for exposure times greater than 1 sec.  
Is it possible to create a mode that stacks images from successive exposures

aaronwmorris, 2024-06-05:
I have been thinking about this.  The stacking module would be an obvious place to do this.  It is already designed to store a sequence of images (in memory) and perform an action on them.  Performing a sum would not be very difficult.  I do not know if it is necessarily appropriate, but I am still considering it.

As for the timing of the exposures... that can be a mixed bag.  Most of the INDI servers are pretty good, but I have had one or two that gave problems.  For instance, with my SVBony SV305, I had issues with about 33% of the images taking double the exposure time to complete.  The exposure appeared to be correct, but for some reason, sometimes a 15s exposure would take 30s to complete.  There are also times that some exposures arrive in less time than the requested exposure.  This is the reason I started tracking the time it takes to receive exposures in the Lag view.

The exposure issue is why I shifted the capture process to a non-blocking loop early in development.  Instead of waiting 15s after requesting a 15s exposure, indi-allsky immediately starts continuously polling the exposure control to check the status of the camera.  (It also gives the opportunity to perform other tasks while waiting on exposures).

There is one other thing to consider in this case.  Downloading the image from the camera will slow down the process considerably.  You cannot start an exposure until the previous exposure is complete and downloaded.  Downloading the exposure requires time, and it can range from 0.5s to 3-4s depending on the camera and size of the frame.  If it takes 2s to download an image, it would require 90s to take 30 1s exposures.

A video streaming mode would negate the download time, but that is out of scope for indi-allsky.

gaitskell, 2024-06-05:
Thanks for the notes. The CCD download overhead is unavoidable, I will check the logs on our system to see what time that is consuming - I assume it is reported. We would make the bandwidth/storage consumed for FITS transfer more optimal as discussed with stacking at capture time.

gaitskell, 2024-06-30:
@aaronwmorris  Trust all is well. Have you been able to see how to make the stacking of Synthetic FITS file possible at the front end? Is there a good way to implement this? Thanks, Rick

aaronwmorris, 2024-07-04:
I am still thinking about this.  I had another thought that this could be a new "camera" type.  For instance, if you want a 30s exposure, it would take 30 1s exposures and return a single FITS frame with float32 data.

The challenge is the exposure feedback which adjusts the exposure based on current sky conditions.  Most people use indi-allsky for the pretty pictures, not necessarily objective scientific data.  :-)

I am still batting around ideas.

gaitskell, 2024-07-16:
I think that anyone using front end stacking to create synthetic FITS would be happy with a specified exposure time - specifically you don't want the CCD exposure times changing based on exposure feedback since there are so many different ways to define an optimium exposure. Setting a fixed time per exposure works well in most scenarios.

aaronwmorris, 2024-07-17:
I am still looking at this.  I made a first attempt at creating a new camera driver and realized it would be more complex than I initially realized.

indi-allsky was designed around INDI (duh), which for the most part functions asynchronously.  When configuration is changed, you have to check the status of the control to know when the new config is complete (like slewing a mount or taking an exposure).

The challenge I ran into is if I have to abstract the exposure controls, I have to do so asynchronously.  This likely means I will have to create a second indi client connection in a thread to manage the exposures, separate from the main camera connection.

gaitskell, 2024-07-25:
As an alternative one could create a seperate process that goes through the available local FITS files and generates combined  synthetic files (by adding). 

The upload/transfer processes would then be configured to only upload *_synth.fits rather than all the local fits files.

aaronwmorris, 2024-08-14:
Okay, I have made some progress on this.  I have developed a new camera type which will be the "INDI Accumulator".  Basically, the synthetic exposures will be summed prior to any processing and indi-allsky basically receives a single 32-bit unsigned FITS file with the synthetic result.

For the final image, the results are expected to roughly align with uint16 data.  Any values greater than 65535 are cut off to 65535 and the image is processed as if the FITS were a normal uint16.  However, the saved FITS file will have the original uint32 data.

The dark frame processing also supports using the synthetic data to remove dark current, etc, but this is optional.

This is currently in my dev branch and I am going to let this run for a few days on my systems to make sure there are no major issues.

aaronwmorris, 2024-08-15:
Merged #1456 with the new Accumulator camera.

closed 2025-02-20 by aaronwmorris