aaronwmorris/indi-allsky #1298: Lat and Lon do not update with GPS Location

degan6, 2024-06-04:
The system lat and lon do not seem to update when the GPS is enabled. I can see the correct coordinates in the log, but the config remains the same.

I have built a portable allsky camera and would like to avoid having to set the lat and log each time I deploy it. This would help me track where the camera was placed.

Thank you for this great software.

degan6, 2024-06-05:
I created this pull request that adds a line to the location config page.

https://github.com/aaronwmorris/indi-allsky/pull/1302

I think, ideally, the location should be updated based on the GPS location, but I get that could be a tall feature request affecting a few people.

I hope this small change will save others time.

If I am completely wrong, let me know and I'll take another look.

Thank you again for putting this project online! It is making my life better!

aaronwmorris, 2024-06-07:
I finally re-read the line you added, but I believe that to be incorrect.  If the location changes according to GPS, it should update both the location in the status as well as the location in the config.

There are rules that prevent location updates until the Latitude and Longitude have changed by at least 0.1 degrees, which is roughly 7 miles.

degan6, 2024-06-07:
Thanks for taking a look @aaronwmorris. I'll take another and test it again.

Part of me thinks this might be a bug with the UI, where the config gets updated but the config page doesn't show the new value somehow.

aaronwmorris, 2024-06-07:
The config should be loaded with pretty much every page load.

aaronwmorris, 2024-06-20:
I just realized there is another possibility...  when the config is loaded, it loads the latest config based on a date column in the database.  If you changed the time where the new time is older than the current system date, the config can get stuck on the "latest" config which is technically in the future.  This would not correct until the new time has surpassed the old time, which might be a few hours.

aaronwmorris, 2024-07-17:
I have taken some additional steps to ensure this does not happen.  The configs are now stored with the UTC timestamp instead of a localtime timestamp.  This will ensure timezone changes do not cause configs to be stored in the "future".

Closing this for now.  Let me know if there is anything else I can do.

closed 2024-07-17 by aaronwmorris