aaronwmorris/indi-allsky #1299: System fails to operate and take images

AstroSmoke, 2024-06-04:
Hi,
This my 2nd main build with indi-allsky,

My 1st attempt, As an overall experience of my first time, using a Pi5, 128GB Kingston SD card and an ASI290MC

It was a bit of a fumble through the install process, mostly allow anything auto to happen and try not to 2nd guess options, I eventually had a system operational.

Although I had a number of problems during my 1st setup, I had images and time lapses working.
Mainly the issues I had were getting settings to apply, and then have a change and or difference with the image. 
I completely gave up with focus, and used sharpcap to focus the camera, then returned to the software.
I also gave up trying to color correct and used AWB.  
But in the end I manged to capture some reasonable images at night, however at dusk and dawn and day the images were completely blown.

I found trying to change settings frustrating and some what difficult. Eventually I had reasonable night images.

I note one thing that is different with install one, I used indi as a camera, rather than the IMX290 option. But in both cases I used the ASI single ccd option during the install process. However camera info never worked using indi.

I then manged to get a Pi M.2 hat and a 515GB NVMe and using the latest Pi5 firmware installed a fresh copy of PiOS and then installed indi-allsky onto the NVMe drive. Note I set the options, dtparam=nvme and dtparam=pciex1_gen=3 to enable the NVMe.

I took note during the install process and I saw what looked like a number of install errors in the indi component.
This occurred with both installs regardless of the storage media.

In the 2nd install, I tried to make sure the camera was linked, at first the system would not find the camera, despite, reinstalling the whole system 3 times and making sure location and camera selection.

I left the system alone operating for 1/2 hour and suddenly the camera showed up in the top left indicator of the web service.
Camera info also also started working.
However no image was taken, ever.
I noted the long and lat were incorrect, and tried to enter correct long and lat for my timezone.
It took an hour of repeated attempts to enter the long and lat values, they just would not take, entering via a network pc and using the Pi5 chrome on the pi itself.
eventually the long and lat suddenly took, I have no idea what I did differently.

Now the system still displays DOWN, and in services all are enabled and running.
Numerous reboots, etc. No change. 
I even reinstalled the whole system twice more, thinking a possible error or broken code during download.
This resulted in the same outcome, system DOWN and no images taken.
  
I've attached my logs and system info to see if you can find any reason for my difficulties. I suspect this not everyone's experience, just some behaviour in relation to the devices used. 
 
[Systeminfo.zip](https://github.com/user-attachments/files/15544693/Systeminfo.zip)

[log.zip](https://github.com/user-attachments/files/15544684/log.zip)

regards
AstroSmoke

AstroSmoke, 2024-06-04:
further information, and setting change.

The system suddenly starting working, when: I set the camera interface to 'INDI'
Is there something wrong with the IMX290 option, or is this not intended for ZWO ASI290 cameras?

aaronwmorris, 2024-06-04:
The IMX290 option is for libcamera devices which are camera modules connected to the CSI port on a Raspberry Pi.  You would want INDI for the ZWO ASI290MC.

When you change any of the settings, you almost always have to issue a "Reload" to restart the indi-allsky processes for them to take effect.

For various reasons, the location is not read directly from the config.  The location is stored with the camera configuration in the database.  When indi-allsky starts, it synchronizes the location information from the config to the camera definition.  So a reload/restart is necessary to update the location information.  Reloading takes about 2-3 minutes to fully complete.

If you ever physically unplug the camera, the indiserver process needs to be restarted as well.  Rebooting should correct this in most cases, but you can also restart from the Services tab under System.

I appreciate the log, it shows I had a typo in the CFA pattern for the libcamera IMX290, even though you would not use that interface.

AstroSmoke, 2024-06-05:
Hi, thanks for the reply.
I did use the tick box to reload on config changes. 

A couple of suggestions,

If you could add a feed back illumination icon, after you make a change an icon displays RED, when update complete change to Green, Or a written 'update in progress', then 'changes complete'.
There is some feed back text, in green, that is hard to see. There signals need a background highlight or different color.

Despite the guide, description of 'choice' options during the install could do with deeper explanation, choosing the right camera choice had me obviously confused. There was two options in relation to ASI cameras, and I choose the single ccd, but kinda rolled the dice which one to choose. Of course as you pointed out I messed up selecting the IMX290 during the install.

When it came to Lat and Long coords, I am sure I tried reload, rebooting, and everything you suggested, for some reason it would not commit. I wasn't sure how many decimal places was causing a problem. Again a feed back loop would help, because of the lag between entry and response. We get so used to instant gratification.

Focus, I like the idea, as this is needed in the other allsky software, however I still struggled. What i like to do is wind up the gain to 300+ to run high speed FPS to observe the focus happen. When way out of focus with a slow system you can spend hours just trying to get close, then with large image lag where you might be one or two images behind the movement you end up in a dance. I.m using the ASI lens that can zoom and focus and as I mention above, I had to jump out to sharpcap, or ASI software on the laptop to dial in the focus.
There are focus options spread over image and focus, It is such a thing to get set up that you could create a separate Tab specifically with more timing function that could override AWB and other features that slow the image output down.
It could be a place to setup auto focus routines at a later stage.

regards

aaronwmorris, 2024-06-06:
I will have to change how the status reporting functions to give feedback on the reload, but it is a good idea.  The web interface has no direct interaction with the capture process, so I have to communicate between them using the database.

The focus functionality does require patience.  Given how indi-allsky performs image capture, there is always going to be a 1-2s lag between taking the image and seeing the image.  The indi-allsky image capture process is asynchronous from the image processing (and the web interface is also independent).  The image processing is automatically reduced when in focus mode, but the biggest delay is downloading the image from the camera.  Some cameras only take 0.5s to download the image... some require +4s to download.  Recently, I added some logic to slow down the capture process if the image processing queue falls behind, so this should prevent the image from being something like 10s behind.

I prefer using Firecapture to focus as well.  Realtime video feedback does make things easier.  Unfortunately, neither SharpCap or Firecapture will work with CSI connected cameras (Raspberry Pis), so I wanted to give some focus functionality.  It is not a great option, but it is better than nothing.

aaronwmorris, 2024-06-07:
Merged #1307 with some better feedback when the system is starting, running, and reloading.

closed 2024-09-14 by aaronwmorris