aaronwmorris/indi-allsky #1310: Raspberry CPU Temp for Fan Control

giangitesta, 2024-06-07:
Hi, I need some help. I would like to use my Raspberry's CPU temperature as a target for controlling the fans. It's possible?

Regards

aaronwmorris, 2024-06-07:
I will work on that today.

aaronwmorris, 2024-06-07:
Merged #1311 .  For Raspberry Pi 3/4/5, the CPU temperature should be `System Temp 10`.

aaronwmorris, 2024-06-12:
@giangitesta What kind of fan do you have connected?

giangitesta, 2024-06-12:
Hi,
i connected a 40 mm x 10 mm 12V DC Brushless fan and two 60mm fans always at 12 V DC controlled through a logic level mosfet in ON/OFF mode.
Unfortunately, I haven't had a chance to update yet to test. I hope to do it this weekend.
Greetings and thanks for your great work.

[image]
[image]

aaronwmorris, 2024-06-12:
Very nice!  I just wanted to make sure you were not using something like the PoE HAT with an i2c controllable fan.

aaronwmorris, 2024-07-17:
@giangitesta I just wanted to follow up to make this was working for you.

giangitesta, 2024-07-18:
Good morning Aaron,
sorry for the delay in replying.
You can consider the update working.
Thanks again and regards.

G

giangitesta, 2024-07-18:
Good morning @aaronwmorris ,
sorry for the delay in replying.
You can consider the update working.
Thanks again and regards.

G

closed 2024-07-18 by giangitesta