aaronwmorris/indi-allsky #1313: Memory swap full.

clambert24, 2024-06-08:
Ok. Decided to re install everything on a new bugger sd card as was running out of storage quickly. 
Now it's saying memory swap full. What does this mean and hiw to fix it? 
[image]

clambert24, 2024-06-08:
Yup that sorted it, thank you. Got it down to 30%

clambert24, 2024-06-14:
Ok, so my problem wasn't resolved. 
Turned out my pi 3B had an issue and kept shutting down/hanging after about 30 minutes.  
Found a spare pi 4 so had to re install everything from scratch is I thi k I messed up the swap file issue and didn't know how to solv it. 
I'm back up running again on a pi4 with a 64mb sd card, but still have a 100% full swap file. 
I'm not sure exactly how to fix that or exactly what to type/do.

aaronwmorris, 2024-06-14:
When the swap is full, can you provide the output of `./misc/support_info.sh` ?

clambert24, 2024-06-14:
it was 100% earlier, but now its 98%. 
hopefully this gives you an idea.

/home/admin/indi-allsky/misc/support_info.sh: line 61: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 1891248 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

Time
Fri 14 Jun 18:48:28 BST 2024

System timezone
Europe/London

Uptime
 18:48:28 up 19:26,  2 users,  load average: 0.39, 0.42, 0.37

Memory
               total        used        free      shared  buff/cache   available
Mem:         1891248      990484      257564       84932      797928      900764
Swap:         204796      201688        3108

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev              672452        0    672452   0% /dev
tmpfs             189128     1412    187716   1% /run
/dev/mmcblk0p2  60801648 13733240  43960056  24% /
tmpfs             945624      584    945040   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
/dev/mmcblk0p1    522230    76402    445828  15% /boot/firmware
tmpfs             189124       40    189084   1% /run/user/1000

sysctl info
vm.swappiness = 60

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether d8:3a:dd:26:a6:9d brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.86/24 brd 192.168.0.255 scope global dynamic noprefixroute eth0
       valid_lft 59652sec preferred_lft 59652sec
    inet6 fe80::70be:dec7:e0a4:233b/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d8:3a:dd:26:a6:9e brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.176/24 brd 192.168.0.255 scope global dynamic noprefixroute wlan0
       valid_lft 59666sec preferred_lft 59666sec
    inet6 fe80::450f:ad5c:2244:4447/64 scope link noprefixroute 
[... 1080 lines cut ...]
    "THOLD_DIFF_HIGH": 10,
    "THOLD_ENABLE": false
  },
  "TEMP_SENSOR": {
    "A_CLASSNAME": "",
    "A_PIN_1": "D5",
    "A_USER_VAR_SLOT": 10,
    "A_I2C_ADDRESS": "0x77",
    "B_CLASSNAME": "",
    "B_PIN_1": "D6",
    "B_USER_VAR_SLOT": 15,
    "B_I2C_ADDRESS": "0x76",
    "C_CLASSNAME": "",
    "C_PIN_1": "D16",
    "C_USER_VAR_SLOT": 20,
    "C_I2C_ADDRESS": "0x40",
    "OPENWEATHERMAP_APIKEY": "REDACTED",
    "OPENWEATHERMAP_APIKEY_E": "REDACTED"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-06-14:
Setting the swappiness kernel setting is a pretty easy thing to keep the swap usage lower, that is on the memory optimizations page.

You probably need to bump up the swap to 512MB as well.

clambert24, 2024-06-14:
Ok. 
can you please explain in pure english that a child can understand. sorry but i get very confused on what to type and where.

aaronwmorris, 2024-06-14:
No worries.

For the swappiness, can copy and paste this directly into your console for the raspberry pi.

```bash
echo "vm.swappiness = 1" | sudo tee /etc/sysctl.d/90-swappiness.conf

sudo sysctl --system
```

For the swap, you can follow these instructions.  You do not really need to do this one.

* Edit `/etc/dphys-swapfile`
* Set `CONF_SWAPSIZE=512` to increase to 512MB
```bash
sudo dphys-swapfile swapoff
sudo dphys-swapfile setup
sudo dphys-swapfile swapon
```

clambert24, 2024-06-14:
ok. so i did the top one as you said, and it did something, because now the system is saying i am using 0% swappiness and in the support info it sais iv got 204796 free and zero used. so guessing this will be ok now?
i tried the other one you mentioned about increasing swap size, but couldnt work out how to change settings. when i did the sudo setup it just said (want /var/swap=200MByte, checking existing: keeping it)

aaronwmorris, 2024-06-14:
The default swappiness setting in Linux is `60` which causes the kernel to swap when there is [relatively] low memory "pressure".  This is supposedly good for desktop systems and prioritizes keeping memory free for new applications to be run.  On a server and an allsky system, you do not really want to keep memory free.  It should use all available memory at all times, but not use swap unless it is absolutely necessary.

Setting to `1` allows the kernel to primarily steal memory from file cache instead of swapping to disk.

clambert24, 2024-06-14:
ok thats great thank you so much for explaining that and helping me. i have set it up now in my garden and shall see what results i have in the morning. thank you.

clambert24, 2024-06-15:
Ok, this is not goid. 
Thus morning it said timelapse failed to generate and swap memory is back to 100%

aaronwmorris, 2024-06-15:
Try upgrading to the latest code, just in case.

If you have the GUI running on the Raspi, you may consider turning it off with the following:

```
systemctl set-default multi-user.target

systemctl isolate multi-user.target
```

aaronwmorris, 2024-08-10:
@clambert24 Did you manage to work through this or are the timelapses failing to generate?

closed 2024-09-14 by aaronwmorris

morganjrees, 2025-04-10:
Hi, I seem to be in similar situation where I've applied the suggested fixes. 

After a few days the swap builds up and the page stops loading and have to pull plug to reboot then works for a few days and stops again, assuming its the swap filling and freezing up the system.

Im looking at upgrading to PI5 with 15GB memory but down want that to also fill up and stop after a couple weeks?

[image]

aaronwmorris, 2025-04-10:
@morganjrees There will be high swap usage on a Pi3 due to the memory constraints.  I would suggest at least 1-2GB of swap on that platform.