aaronwmorris/indi-allsky #1318: Not Producing Timelapses or Keos Automatically - ASI678MC

AlaskanAstro, 2024-06-10:
Hello, recently got this project running after migrating over from TJ's all sky. I'm currently trying to sort through my issues and the first is that it doesn't seem to be creating timelapses or keograms automatically. 

I _can_ however use the Generate page to create a timelapse and keo so the hardware seems capable. Here's a short summary of my circumstances and my guesses what may be up:

Raspi 4b 4GB, ASI678MC, 15s day subs, I live at 63N and night time mode is not triggering at all. 

Some guesses from things that have caused issues with other programs: 

- My sunset is after midnight because of timezone and latitude
- I got a Swap Space Full message this morning but I haven't seen that when I manually generated the Timelapse or Keo
- 24 hours of 15s intervals with a 4k cam is a lot
- No switch to night mode even with -3 degree sun altitude trigger this time of year
- 40k video bitrate

Again this is a bit confusing because the manual generation does work. Please let me know if any extra logs would be helpful. Really enjoying the program so far. 

/home/nina/indi-allsky/misc/support_info.sh: line 61: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3883064 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux allsky2 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux

Uptime
 10:13:59 up 1 day, 15:05,  2 users,  load average: 0.19, 0.07, 0.01

Memory
               total        used        free      shared  buff/cache   available
Mem:         3883064     1109360      609176       29896     2271900     2773704
Swap:         204796      202180        2616

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1668360        0   1668360   0% /dev
tmpfs             388308     1324    386984   1% /run
/dev/mmcblk0p2  60801648 25895396  31797900  45% /
tmpfs            1941532      316   1941216   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
/dev/mmcblk0p1    522230    76382    445848  15% /boot/firmware
tmpfs             388304       40    388264   1% /run/user/1000

sysctl info
vm.swappiness = 60

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

User info
uid=1000(nina) gid=1000(nina) groups=1000(nina),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),115(lpadmin),993(gpio),994(i2c),995(spi),999(systemd-journal)

Process info
nina        1005  0.7  0.1  83676  5692 ?        Ss   Jun08  17:29 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_asi_ccd
nina        1023  0.0  0.2  41236 10368 ?        S    Jun08   2:03 indi_simulator_telescope
nina        1024  1.2  2.1 404548 85228 ?        Sl   Jun08  28:56 indi_asi_ccd
nina        1975  0.0  0.2  33412 10812 ?        SNs  Jun08   0:25 /home/nina/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/nina/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
nina        1977  0.0  6.6 1085936 257664 ?      SNl  Jun08   1:57 /home/nina/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/nina/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
nina        2001  0.0  1.6 800944 63660 ?        SNsl Jun08   1:20 /home/nina/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
nina        2431  4.6  3.8 839976 147720 ?       SNl  Jun08 107:01 /home/nina/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
[... 1149 lines cut ...]
    "THOLD_DIFF_LOW": 15,
    "THOLD_DIFF_MED": 10,
    "THOLD_DIFF_HIGH": 5,
    "THOLD_ENABLE": false
  },
  "TEMP_SENSOR": {
    "A_CLASSNAME": "",
    "A_PIN_1": "D5",
    "A_USER_VAR_SLOT": 10,
    "A_I2C_ADDRESS": "0x77",
    "B_CLASSNAME": "",
    "B_PIN_1": "D6",
    "B_USER_VAR_SLOT": 15,
    "B_I2C_ADDRESS": "0x76",
    "C_CLASSNAME": "",
    "C_PIN_1": "D16",
    "C_USER_VAR_SLOT": 20,
    "C_I2C_ADDRESS": "0x40"
  }
}

```
#################################
###     end support info      ###
#################################

AlaskanAstro, 2024-06-10:
Timezone is Alaska, uhhh AKST? GMT-8 right now. We really should have like 3 timezones across the state. My transit time this time of year is around 2:30PM so not only is my sunset late because of being far north it's also very offset because I'm far west into the timezone.

henrikeri, 2024-06-11:
I have the same issue happening since a few days ago. Lat 58N here, GMT+1 with DST (Oslo time)

aaronwmorris, 2024-06-12:
After thinking about this, obviously the problem is the sun never reaches the altitude to trigger the day/night transition, which is what generates the timelapse.

Now that I fully understand the problem, I am considering options.

AlaskanAstro, 2024-06-12:
Two things on that: 1, I could test that tonight with a night trigger of -1 degree if you'd like. 
2, How finely does INDI All Sky calculate sun altitude? My trigger was -3 degrees and I believe my true sun alt went as low as -3.9 or even 4 on the first night I noticed it didn't trigger.

aaronwmorris, 2024-06-12:
indi-allsky performs its astrometric calculations using pyephem, so the raw floating point value is used for the Sun's altitude for the current timestamp.

I went ahead and wrote a script to calculate the Suns min/max altitude for a given day at any lat/long, as well as the overall min/max on the solstices.  5 days ago, the Sun's minimum altitude was -2.7 degrees for your location.

This is definitely a problem in any case.  I need to implement a failsafe to prevent this from happening.

```
$ ./testing/sunAltMinMax.py
WARNING:root:Latitude:  62.9
WARNING:root:Longitude: -160.0
WARNING:root:Now - 2024-06-07 18:31:36.086857+00:00
INFO:root:2024-06-07 22:39:03.879051: 50.0
INFO:root:2024-06-08 10:39:03.879051: -2.7
WARNING:root:Solstice 1
INFO:root:2024-06-20 22:41:48.101761: 50.5
INFO:root:2024-06-21 10:41:48.101761: -2.3
WARNING:root:Soltice 2
INFO:root:2024-12-21 22:38:30.713792: 3.9
INFO:root:2024-12-22 10:38:30.713792: -50.5
```

AlaskanAstro, 2024-06-13:
Cool well I look forward to implementing the fix.

aaronwmorris, 2024-06-13:
When I was first writing the test script, I also noticed a discrepancy between pyephem and skyfield.  Skyfield is supposedly the more modern replacement to pyephem and has the same author.  I may open a ticket with that team to find out if I am doing something wrong.

I tried several combinations of lat/long and in most cases the minimum altitude is within 0.2 degrees for both libraries, but for some reason, your latitude can have a difference of more than a degree.

```
WARNING:root:Latitude:  62.9
WARNING:root:Longitude: -160.0

WARNING:root:pyephem Now - 2024-06-07 19:30:40.486816+00:00
INFO:root:2024-06-07 22:39:03.879051: Max 49.977
INFO:root:2024-06-08 10:39:03.879051: Min -2.713
WARNING:root:pyephem Solstice 1
INFO:root:2024-06-20 22:41:48.101761: Max 50.550
INFO:root:2024-06-21 10:41:48.101761: Min -2.270
WARNING:root:pyephem Solstice 2
INFO:root:2024-12-21 22:38:30.713792: Max 3.857
INFO:root:2024-12-22 10:38:30.713792: Min -50.536

WARNING:root:skyfield Now - 2024-06-07 19:30:40.486816+00:00
INFO:root:2024-06-07 22:39:03.915149+00:00: Max 49.964
INFO:root:2024-06-08 10:39:09.749678+00:00: Min -4.194
WARNING:root:skyfield Solstice 1
INFO:root:2024-06-20 22:41:48.131156+00:00: Max 50.537
INFO:root:2024-06-21 10:41:54.591994+00:00: Min -3.665
WARNING:root:skyfield Solstice 2
INFO:root:2024-12-21 22:38:30.655583+00:00: Max 3.660
INFO:root:2024-12-21 10:38:15.721727+00:00: Min -50.540
```

aaronwmorris, 2024-06-13:
And I have the answer...  Apparently, pyephem is accounting for refraction at the horizon, so the sun is visually at a different altitude than it is physically.  skyfield does not account for refraction below -1 degrees.  Setting the atmospheric pressure to 0 disables the refraction calculation and makes the numbers exactly the same in both libraries.

https://github.com/skyfielders/python-skyfield/discussions/974

aaronwmorris, 2024-06-14:
Back to the issue at hand...

My plan is to add a drop dead time limit for daytime and night time.  If the day/night Sun altitude is not hit, the drop dead time will trigger generating a timelapse for the previous period.  Normally, indi-allsky expects a day then a night then another day, but this will introduce the concept of back-to-back days during summer, and back-to-back nights in winter.

This introduces another challenge... the `dayDate` classification.   The way indi-allsky classifies images for a timelapse is the `dayDate` attribute.  The drop dead limit for day and night cannot be midnight/noon, it needs to be meridian/anti-meridian.  However, timezones being as they are, the anti-meridian sometimes occurs the following day (during summer).  Due to this "misalignment", it means some daytime images that fall after midnight need to be associated with the previous dayDate period.  My current method for calculating dayDate needs to be refactored.

AlaskanAstro, 2024-06-14:
Well I'm excited to see your ideas on this. Thanks so much for tackling it. Sorry I broke your project. 😄 Will definitely be buying you a coffee with the link at some point.

aaronwmorris, 2024-06-21:
I went ahead and pulled the trigger on this merge #1340 .  This should properly handle higher latitudes now.  The meridian is now considered the hard stop between consecutive nights, and the anti-meridian for consecutive days.

There may be gremlins in this code.  I have tested many cases with this, but the behavior is unique to the different timezone and latitude/longitude combinations.

henrikeri, 2024-06-21:
Deployed, I'll let you know how it plays out :)

AlaskanAstro, 2024-06-21:
As someone not super familiar with Git, how do I update this without it being part of a main release?

aaronwmorris, 2024-06-24:
Hello all, if you upgraded to fix the previous issue, I introduced a problem which caused the wrong timelapse to be generated.  Merged #1348 

Please upgrade to the latest release to fix this issue.

aaronwmorris, 2024-06-24:
@AlaskanAstro Unfortunately, I have a very simple release management process.  I generally do not maintain separate feature branches.  With a change like this, do generally develop it on a separate branch for the initial integration, but try to quickly merge to dev/main as quickly as possible.

AlaskanAstro, 2024-06-24:
Got, figured it out and followed your simple wiki instructions. Thank you!

aaronwmorris, 2024-07-05:
@AlaskanAstro @henrikeri Did the updates fix the issues with the timelapses not being generated?

AlaskanAstro, 2024-07-05:
Yes, it worked great. Just 2 days ago it started triggering night again actually at -3 degrees, so I may tweak that back to a more conservative time since the 24 hour videos were working fine.

aaronwmorris, 2024-07-21:
FYI, there was a bug reported by another user due to this fix for this issue:

#1414

aaronwmorris, 2024-08-10:
Closing this for now.  Please let me know if there are any related issue!

closed 2024-08-10 by aaronwmorris