aaronwmorris/indi-allsky #1328: Tomorrow failing

laterrible, 2024-06-15:
Hey team, recent fresh install going pretty well, it's been a while since I last used this and it's better than ever, with one exception: I can't get timelapses to generate.

Both auto and manual attempts fail. Here's the support log. 4b with pi hq camera, running from Poe hat installed on a USB stick I've had success with before.

What goofy things have I setup incorrectly or overlooked?

/home/admin/indi-allsky/misc/support_info.sh: line 61: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 1890224 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux indi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

Time
Sun 16 Jun 07:17:05 AEST 2024

System timezone
Australia/Sydney

Uptime
 07:17:05 up 12:12,  2 users,  load average: 0.94, 0.78, 0.81

Memory
               total        used        free      shared  buff/cache   available
Mem:         1890224      734312      240996        9596      990748     1155912
Swap:        1048572      307160      741412

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev              672172        0    672172   0% /dev
tmpfs             189024     1340    187684   1% /run
/dev/sda2       58556632 42306340  13255912  77% /
tmpfs             945112      308    944804   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
/dev/sda1         522230    75372    446858  15% /boot/firmware
tmpfs             189020       36    188984   1% /run/user/1000

sysctl info
vm.swappiness = 60

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

User info
uid=1000(admin) gid=1000(admin) groups=1000(admin),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),115(lpadmin),993(gpio),994(i2c),995(spi),999(systemd-journal)

Process info
avahi        605  0.3  0.1   7672  3328 ?        Ss   Jun15   2:18 avahi-daemon: running [indi.local]
admin       1054  0.1  0.1   9620  2816 ?        Ss   Jun15   0:58 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_simulator_ccd
admin       1073  0.0  0.4  41232  8576 ?        S    Jun15   0:39 indi_simulator_telescope
admin       1074  0.0  0.4 123336  8576 ?        Sl   Jun15   0:31 indi_simulator_ccd
admin       2014  0.0  3.2 801672 60652 ?        SNsl Jun15   0:29 /home/admin/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
admin       2336  0.0  1.0  33408 19012 ?        SNs  Jun15   0:07 /home/admin/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/admin/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
admin       2338  0.0  5.3 672340 101060 ?       SNl  Jun15   0:23 /home/admin/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/admin/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
admin       2364  0.2  2.8 802104 52992 ?        SNl  Jun15   1:56 /home/admin/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
admin       2366  4.0 11.8 991784 224272 ?       SNl  Jun15  27:06 /home/admin/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
admin       2367  0.6 12.9 950444 244608 ?       SNl  Jun15   4:16 /home/admin/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
admin       7981  0.6  0.1   6800  3072 ?        SN   07:17   0:00 /bin/bash /home/admin/indi-allsky/misc/support_info.sh
[... 1088 lines cut ...]
    "THOLD_DIFF_LOW": 0,
    "THOLD_DIFF_MED": 5,
    "THOLD_DIFF_HIGH": 10,
    "THOLD_ENABLE": false
  },
  "TEMP_SENSOR": {
    "A_CLASSNAME": "",
    "A_PIN_1": "D5",
    "A_USER_VAR_SLOT": 10,
    "A_I2C_ADDRESS": "0x77",
    "B_CLASSNAME": "",
    "B_PIN_1": "D6",
    "B_USER_VAR_SLOT": 15,
    "B_I2C_ADDRESS": "0x76",
    "C_CLASSNAME": "",
    "C_PIN_1": "D16",
    "C_USER_VAR_SLOT": 20,
    "C_I2C_ADDRESS": "0x40"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-06-16:
Try upgrading.  I merged #1327 last night which might be the cause.

laterrible, 2024-06-16:
Sadly no dice. Ran the upgrade just now. It has generated a couple timelapses in the last week, all of which have very quirky artefacts (red outlines, horrible exposure etc). Scaling is set to 50%, so expecting t isn't a memory-related thing.

I'll dig into the logs, see if I can find anything else sticking out. Thanks Aaron

aaronwmorris, 2024-06-16:
If ffmpeg generates an error, it should be logged.  All timelapse errors would be under the Video worker.

You can search for Video worker related errors with:

```
grep Video /var/log/indi-allsky/indi-allsky.log
```

laterrible, 2024-06-16:
Ah nice, thanks, saved me hunting around. OK, so that log points at a few things namely

[warning] deprecated pixel format used, make sure you did set range correctly\n[libx264 @ 0x557f91ade0] [error] width not divisible by 2 (1521x 1140)\n[error] Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height\n'

over and over again.

I'll have a closer look at the config and settings, see where it's gone wrong. Thanks again Aaron

closed 2024-06-16 by laterrible

aaronwmorris, 2024-06-16:
Beat me to it...

Okay, that helps, I think the cause is the image scaling combined with the video scaling.  Since you are scaling the image by 75%, then ffmpeg is supposed to scale that by 50%, but the resulting resolution is not divisible by 2.

I have protections in indi-allsky to make sure that any action in its own control that changes the image dimensions, the resulting image wlll be divisible by 2, but this is a case I cannot manage because ffmpeg is the cause.

aaronwmorris, 2024-06-16:
The IMX477 specific scaling options set a specific size and do not calculate the scale mathematically.  That is why it works.