aaronwmorris/indi-allsky #1329: ImportError: numpy.core.multiarray failed to import

TonAmSi, 2024-06-17:
Hello to all,
I have the following problem (see error message).

What else can I try to get setup.sh to run?
I have tried various sources on the www, including https://numpy.org/devdocs/user/troubleshooting-importerror.html
- But so far I could not find the solution.

I am using a Raspberry Pi 5 (4 GB) with a Zwo ASI 385MC.

Many thanks in advance, Amy 

----------------------------------
AttributeError: _ARRAY_API not found
Error: While importing 'app', an ImportError was raised:

Traceback (most recent call last):
  File "/home/darkstar/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/cli.py", line 245, in locate_app
    __import__(module_name)
  File "/home/darkstar/indi-allsky/app.py", line 3, in <module>
    from indi_allsky.flask import create_app
  File "/home/darkstar/indi-allsky/indi_allsky/flask/__init__.py", line 17, in <module>
    from .views import bp_allsky  # noqa: E402
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/darkstar/indi-allsky/indi_allsky/flask/views.py", line 22, in <module>
    from ..processing import ImageProcessor
  File "/home/darkstar/indi-allsky/indi_allsky/processing.py", line 11, in <module>
    import cv2
  File "/home/darkstar/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/home/darkstar/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/cv2/__init__.py", line 153, in bootstrap
    native_module = importlib.import_module("cv2")
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: numpy.core.multiarray failed to import

aaronwmorris, 2024-06-17:
Looks like numpy 2.0 was released and it is not compatible with the rest of the modules.

TonAmSi, 2024-06-17:
Yes, that is definitely the case! 😁
Maybe make a note of this special feature in the wiki?

aaronwmorris, 2024-06-17:
I Just updated the requirements to prevent numpy 2.0 from being installed.

Merged #1331

closed 2024-07-05 by aaronwmorris