aaronwmorris/indi-allsky #1333: TSL2591 needs adjustment to gain during the day

Vertikar, 2024-06-19:
Hey Aaron, found an issue with the default gain setting during the day.

### Issue: 
Too much light during the day for the default setting of 25x gain - self.tsl2591.gain = adafruit_tsl2591.GAIN_MED  
Was working fine overnight until about 10am here, and i can see its started throwing an exception in the logs below.

### Fix:
Change gain to self.tsl2591.gain = adafruit_tsl2591.GAIN_LOW, have tested this during the day, will see how it goes tonight, and if it needs to be changed overnight. Shared the working logs below for this too.

**Failing Logs:**

```
Jun 19 10:04:28 allsky [INFO] MainProcess-1256/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 1658.3209, visible: 959990864, ir: 14648, full: 960005512
Jun 19 10:04:28 allsky [INFO] MainProcess-1256/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 3.5c, object temp: -15.1c
Jun 19 10:04:28 allsky [INFO] MainProcess-1256/Sensor-1 sensor.saferun() [147]: Sensor SQM: 4.53439
Jun 19 10:05:29 allsky [INFO] MainProcess-1256/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 1705.7076, visible: 978079267, ir: 14924, full: 978094191
Jun 19 10:05:29 allsky [INFO] MainProcess-1256/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 3.5c, object temp: -15.1c
Jun 19 10:05:29 allsky [INFO] MainProcess-1256/Sensor-1 sensor.saferun() [147]: Sensor SQM: 4.50380
Jun 19 10:06:29 allsky [INFO] MainProcess-1256/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 1748.2441, visible: 1014059143, ir: 15473, full: 1014074616
Jun 19 10:06:29 allsky [INFO] MainProcess-1256/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 3.5c, object temp: -15.2c
Jun 19 10:06:29 allsky [INFO] MainProcess-1256/Sensor-1 sensor.saferun() [147]: Sensor SQM: 4.47705
Jun 19 10:07:29 allsky [INFO] MainProcess-1256/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 1763.9244, visible: 1029001593, ir: 15701, full: 1029017294
Jun 19 10:07:29 allsky [INFO] MainProcess-1256/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 3.4c, object temp: -15.1c
Jun 19 10:07:29 allsky [INFO] MainProcess-1256/Sensor-1 sensor.saferun() [147]: Sensor SQM: 4.46736
Jun 19 10:08:30 allsky [INFO] MainProcess-1256/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 1764.6881, visible: 1039684070, ir: 15864, full: 1039699934
Jun 19 10:08:30 allsky [INFO] MainProcess-1256/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 3.5c, object temp: -14.7c
Jun 19 10:08:30 allsky [INFO] MainProcess-1256/Sensor-1 sensor.saferun() [147]: Sensor SQM: 4.46689
Jun 19 10:09:30 allsky [ERROR] MainProcess-1256/Sensor-1 sensor.update_sensors() [367]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jun 19 10:09:30 allsky [INFO] MainProcess-1256/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 3.6c, object temp: -15.1c
Jun 19 10:09:30 allsky [INFO] MainProcess-1256/Sensor-1 sensor.saferun() [147]: Sensor SQM: 4.46689
Jun 19 10:10:30 allsky [ERROR] MainProcess-1256/Sensor-1 sensor.update_sensors() [367]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jun 19 10:10:30 allsky [INFO] MainProcess-1256/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 3.7c, object temp: -14.9c
Jun 19 10:10:30 allsky [INFO] MainProcess-1256/Sensor-1 sensor.saferun() [147]: Sensor SQM: 4.46689
Jun 19 10:11:30 allsky [ERROR] MainProcess-1256/Sensor-1 sensor.update_sensors() [367]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jun 19 10:11:30 allsky [INFO] MainProcess-1256/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 3.8c, object temp: -15.0c
Jun 19 10:11:30 allsky [INFO] MainProcess-1256/Sensor-1 sensor.saferun() [147]: Sensor SQM: 4.46689
Jun 19 10:12:31 allsky [ERROR] MainProcess-1256/Sensor-1 sensor.update_sensors() [367]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jun 19 10:12:31 allsky [INFO] MainProcess-1256/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 3.9c, object temp: -14.8c
Jun 19 10:12:31 allsky [INFO] MainProcess-1256/Sensor-1 sensor.saferun() [147]: Sensor SQM: 4.46689
Jun 19 10:13:31 allsky [ERROR] MainProcess-1256/Sensor-1 sensor.update_sensors() [367]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
```

**Working logs:**

```
Jun 19 11:55:58 allsky [INFO] MainProcess-9115/MainThread allsky._startSensorWorker() [500]: Starting Sensor-1 worker
Jun 19 11:55:58 allsky [WARNING] MainProcess-9115/Sensor-1 lightSensorTsl2591.__init__() [59]: Initializing TSL2591 I2C light sensor device @ 0x29
Jun 19 11:55:59 allsky [WARNING] MainProcess-9115/Sensor-1 tempSensorMlx90614.__init__() [57]: Initializing MLX90614 I2C temperature device @ 0x5a
Jun 19 11:55:59 allsky [INFO] MainProcess-9115/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 5036.5152, visible: 119867749, ir: 1829, full: 119869578
Jun 19 11:55:59 allsky [INFO] MainProcess-9115/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 25.4c, object temp: -12.9c
Jun 19 11:56:02 allsky [WARNING] MainProcess-9115/Sensor-1 sensor.night_day_change() [155]: Day/Night change
Jun 19 11:56:02 allsky [INFO] MainProcess-9115/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 5040.2688, visible: 119671140, ir: 1826, full: 119672966
Jun 19 11:56:02 allsky [INFO] MainProcess-9115/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 25.3c, object temp: -17.7c
Jun 19 11:56:02 allsky [INFO] MainProcess-9115/Sensor-1 sensor.saferun() [147]: Sensor SQM: 3.32743
Jun 19 11:57:03 allsky [WARNING] MainProcess-9115/Sensor-1 sensor.night_day_change() [155]: Day/Night change
Jun 19 11:57:03 allsky [INFO] MainProcess-9115/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 4989.1872, visible: 119212371, ir: 1819, full: 119214190
Jun 19 11:57:03 allsky [INFO] MainProcess-9115/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 24.6c, object temp: -12.9c
Jun 19 11:57:03 allsky [INFO] MainProcess-9115/Sensor-1 sensor.saferun() [147]: Sensor SQM: 3.33849
Jun 19 11:58:03 allsky [INFO] MainProcess-9115/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 4956.7104, visible: 118491460, ir: 1808, full: 118493268
Jun 19 11:58:03 allsky [INFO] MainProcess-9115/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 25.0c, object temp: -13.5c
Jun 19 11:58:03 allsky [INFO] MainProcess-9115/Sensor-1 sensor.saferun() [147]: Sensor SQM: 3.34558
```

aaronwmorris, 2024-06-19:
My plan is to add these sensor settings to the config page eventually, but I have not decided on the best way to do that yet.

Vertikar, 2024-06-20:
So I had a look last night, and it looks like using the low gain setting, it won't go below 4 lux, in the log below I flicked it back to medium before 23:25:42 and did a restart. Included some more logs further down showing it decreasing to 4 then not going any further.

```
Jun 19 23:19:18 allsky [INFO] MainProcess-9115/Sensor-5 lightSensorTsl2591.update() [24]: TSL2591 - lux: 4.0800, visible: 1, ir: 0, full: 1
Jun 19 23:19:18 allsky [INFO] MainProcess-9115/Sensor-5 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: -1.8c, object temp: -7.1c
Jun 19 23:19:18 allsky [INFO] MainProcess-9115/Sensor-5 sensor.saferun() [147]: Sensor SQM: 11.05691
Jun 19 23:20:18 allsky [INFO] MainProcess-9115/Sensor-5 lightSensorTsl2591.update() [24]: TSL2591 - lux: 4.0800, visible: 1, ir: 0, full: 1
Jun 19 23:20:18 allsky [INFO] MainProcess-9115/Sensor-5 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: -1.8c, object temp: -7.4c
Jun 19 23:20:18 allsky [INFO] MainProcess-9115/Sensor-5 sensor.saferun() [147]: Sensor SQM: 11.05691
Jun 19 23:21:18 allsky [INFO] MainProcess-9115/Sensor-5 lightSensorTsl2591.update() [24]: TSL2591 - lux: 4.0800, visible: 1, ir: 0, full: 1
Jun 19 23:21:18 allsky [INFO] MainProcess-9115/Sensor-5 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: -1.8c, object temp: -7.2c
Jun 19 23:21:18 allsky [INFO] MainProcess-9115/Sensor-5 sensor.saferun() [147]: Sensor SQM: 11.05691
Jun 19 23:22:19 allsky [INFO] MainProcess-9115/Sensor-5 lightSensorTsl2591.update() [24]: TSL2591 - lux: 4.0800, visible: 1, ir: 0, full: 1
Jun 19 23:22:19 allsky [INFO] MainProcess-9115/Sensor-5 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: -1.7c, object temp: -7.2c
Jun 19 23:22:19 allsky [INFO] MainProcess-9115/Sensor-5 sensor.saferun() [147]: Sensor SQM: 11.05691
Jun 19 23:23:19 allsky [INFO] MainProcess-9115/Sensor-5 lightSensorTsl2591.update() [24]: TSL2591 - lux: 4.0800, visible: 1, ir: 0, full: 1
Jun 19 23:23:19 allsky [INFO] MainProcess-9115/Sensor-5 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: -1.6c, object temp: -7.1c
Jun 19 23:23:19 allsky [INFO] MainProcess-9115/Sensor-5 sensor.saferun() [147]: Sensor SQM: 11.05691
Jun 19 23:24:19 allsky [INFO] MainProcess-9115/Sensor-5 lightSensorTsl2591.update() [24]: TSL2591 - lux: 4.0800, visible: 1, ir: 0, full: 1
Jun 19 23:24:19 allsky [INFO] MainProcess-9115/Sensor-5 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: -1.6c, object temp: -7.2c
Jun 19 23:24:19 allsky [INFO] MainProcess-9115/Sensor-5 sensor.saferun() [147]: Sensor SQM: 11.05691
Jun 19 23:25:19 allsky [INFO] MainProcess-9115/Sensor-5 lightSensorTsl2591.update() [24]: TSL2591 - lux: 4.0800, visible: 1, ir: 0, full: 1
Jun 19 23:25:19 allsky [INFO] MainProcess-9115/Sensor-5 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: -1.6c, object temp: -7.0c
Jun 19 23:25:19 allsky [INFO] MainProcess-9115/Sensor-5 sensor.saferun() [147]: Sensor SQM: 11.05691
Jun 19 23:25:29 allsky [INFO] MainProcess-9115/MainThread allsky._stopSensorWorker() [530]: Stopping Sensor worker
Jun 19 23:25:31 allsky [WARNING] MainProcess-9115/Sensor-5 sensor.saferun() [117]: Goodbye
Jun 19 23:25:41 allsky [INFO] MainProcess-17721/MainThread allsky._startSensorWorker() [500]: Starting Sensor-1 worker
Jun 19 23:25:41 allsky [WARNING] MainProcess-17721/Sensor-1 lightSensorTsl2591.__init__() [59]: Initializing TSL2591 I2C light sensor device @ 0x29
Jun 19 23:25:42 allsky [WARNING] MainProcess-17721/Sensor-1 tempSensorMlx90614.__init__() [57]: Initializing MLX90614 I2C temperature device @ 0x5a
Jun 19 23:25:42 allsky [INFO] MainProcess-17721/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 0.1632, visible: 1, ir: 0, full: 1
Jun 19 23:25:42 allsky [INFO] MainProcess-17721/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: -1.6c, object temp: -7.0c
Jun 19 23:25:45 allsky [WARNING] MainProcess-17721/Sensor-1 sensor.night_day_change() [155]: Day/Night change
Jun 19 23:25:45 allsky [INFO] MainProcess-17721/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 0.1632, visible: 1, ir: 0, full: 1
Jun 19 23:25:45 allsky [INFO] MainProcess-17721/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: -1.6c, object temp: -6.9c
Jun 19 23:25:45 allsky [INFO] MainProcess-17721/Sensor-1 sensor.saferun() [147]: Sensor SQM: 14.55176
Jun 19 23:26:45 allsky [INFO] MainProcess-17721/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 0.1632, visible: 1, ir: 0, full: 1
Jun 19 23:26:45 allsky [INFO] MainProcess-17721/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: -1.5c, object temp: -6.9c
Jun 19 23:26:45 allsky [INFO] MainProcess-17721/Sensor-1 sensor.saferun() [147]: Sensor SQM: 14.55176
Jun 19 23:27:45 allsky [INFO] MainProcess-17721/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 0.1632, visible: 1, ir: 0, full: 1
Jun 19 23:27:45 allsky [INFO] MainProcess-17721/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: -1.5c, object temp: -6.7c
Jun 19 23:27:45 allsky [INFO] MainProcess-17721/Sensor-1 sensor.saferun() [147]: Sensor SQM: 14.55176
Jun 19 23:28:46 allsky [INFO] MainProcess-17721/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 0.1632, visible: 1, ir: 0, full: 1
Jun 19 23:28:46 allsky [INFO] MainProcess-17721/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: -1.4c, object temp: -6.9c
[... cut ...]
Jun 19 17:41:35 allsky [INFO] MainProcess-9115/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 2.5c, object temp: -14.5c
Jun 19 17:41:35 allsky [INFO] MainProcess-9115/Sensor-1 sensor.saferun() [147]: Sensor SQM: 11.05691
Jun 19 17:42:35 allsky [INFO] MainProcess-9115/Sensor-1 lightSensorTsl2591.update() [24]: TSL2591 - lux: 4.0800, visible: 1, ir: 0, full: 1
Jun 19 17:42:35 allsky [INFO] MainProcess-9115/Sensor-1 tempSensorMlx90614.update() [21]: MLX90614 - ambient temp: 2.5c, object temp: -14.3c
```

aaronwmorris, 2024-06-20:
I still have my TSL2591 hooked up and it performs the same.  4 lux on low gain appears to be the minimum.  I will work on the interface options.

Vertikar, 2024-06-21:
For reference, I bumped the gain to HIGH last night, and that maxed out after 127 lux this morning. it had a minimum of 0.0458 at 6:45am before the moon had fully set and before the sun had risen above 9 degrees. 
Think it will need some testing when the moon is new too, as I suspect the maximum gain setting maybe useful then

```
Jun 21 08:01:33 allsky [INFO] MainProcess-49393/Sensor-2 lightSensorTsl2591.update() [24]: TSL2591 - lux: 121.9577, visible: 798576752, ir: 12185, full: 798588937
Jun 21 08:02:33 allsky [INFO] MainProcess-49393/Sensor-2 lightSensorTsl2591.update() [24]: TSL2591 - lux: 123.7357, visible: 844911343, ir: 12892, full: 844924235
Jun 21 08:03:33 allsky [INFO] MainProcess-49393/Sensor-2 lightSensorTsl2591.update() [24]: TSL2591 - lux: 127.0276, visible: 912873184, ir: 13929, full: 912887113
Jun 21 08:04:33 allsky [ERROR] MainProcess-49393/Sensor-2 sensor.update_sensors() [367]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jun 21 08:05:34 allsky [ERROR] MainProcess-49393/Sensor-2 sensor.update_sensors() [367]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jun 21 08:06:34 allsky [ERROR] MainProcess-49393/Sensor-2 sensor.update_sensors() [367]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jun 21 08:07:34 allsky [ERROR] MainProcess-49393/Sensor-2 sensor.update_sensors() [367]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jun 21 08:08:34 allsky [ERROR] MainProcess-49393/Sensor-2 sensor.update_sensors() [367]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
```

Vertikar, 2024-06-21:
> I still have my TSL2591 hooked up

How have you got yours setup, with the camera under the clear dome?
still got mine external slightly weather protected until I get my enclosure fully sorted.

astrotater, 2024-06-21:
Is it possible to use an LDR to automatically set the gain of the TSL2591?
I think Robert Brown's SQM meter uses this approach.

Vertikar, 2024-06-21:
> Is it possible to use an LDR to automatically set the gain of the TSL2591? I think Robert Brown's SQM meter uses this approach.

Not sure if this is necessary if we know the ranges of the gains, it could be done in software, changing the gain as it approaches the min and max ranges of each gain setting before it overflows

aaronwmorris, 2024-07-02:
@Vertikar Merged #1373 which includes day and night settings for the TSL2561 and TSL2591 sensors.

Vertikar, 2024-07-03:
> @Vertikar Merged #1373 which includes day and night settings for the TSL2561 and TSL2591 sensors.

Looks to be working for day time so far, I'll see how it goes tonight and let you know

```
Jul  3 11:04:27 allsky [WARNING] MainProcess-11701/Sensor-1 sensor.night_day_change() [154]: Day/Night change
Jul  3 11:04:27 allsky [INFO] MainProcess-11701/Sensor-1 lightSensorTsl2591.update_sensor_settings() [58]: [B] Switching TSL2591 to day mode
Jul  3 11:04:28 allsky [INFO] MainProcess-11701/Sensor-1 lightSensorTsl2591.update() [29]: [B] TSL2591 - lux: 3148.6176, visible: 54199573, ir: 827, full: 54200400
Jul  3 11:04:28 allsky [INFO] MainProcess-11701/Sensor-1 tempSensorMlx90614.update() [22]: [C] MLX90614 - ambient temp: 5.7c, object temp: -1.6c
Jul  3 11:04:28 allsky [INFO] MainProcess-11701/Sensor-1 sensor.saferun() [146]: Sensor SQM: 3.83826
Jul  3 11:04:31 allsky [INFO] MainProcess-11701/MainThread allsky._queueManualTasks() [1211]: Checking for manually submitted tasks
```

Vertikar, 2024-07-03:
I only just updated it now, hence why it only flicked to day mode at 11am

aaronwmorris, 2024-07-03:
Yeah, it starts in a state where it does not know if it is day or night.  When it detects the right setting, it will switch to the right mode.

Vertikar, 2024-07-04:
Opened a new issue, it went from night to day with no issues as we saw, but day to night looks to have not worked for me for some reason.

https://github.com/aaronwmorris/indi-allsky/issues/1376

aaronwmorris, 2024-07-04:
Merged #1376 to fix the typo causing the wrong gain to be used for daytime.

Vertikar, 2024-07-04:
Back to this issue as I think the other one can be closed off now. It transitioned back to night fine again tonight:

```
ul  4 17:39:58 allsky [INFO] MainProcess-29996/Sensor-1 lightSensorTsl2591.update() [29]: [Sensor B] TSL2591 - lux: 0.7507, visible: 655371, ir: 10, full: 655381
Jul  4 17:39:58 allsky [INFO] MainProcess-29996/Sensor-1 tempSensorMlx90614.update() [22]: [Sensor C] MLX90614 - ambient temp: 4.9c, object temp: -6.5c
Jul  4 17:39:58 allsky [INFO] MainProcess-29996/Sensor-1 sensor.saferun() [149]: Sensor SQM: 12.89486
Jul  4 17:40:13 allsky [INFO] MainProcess-29996/Sensor-1 lightSensorTsl2591.update() [29]: [Sensor B] TSL2591 - lux: 0.8552, visible: 589835, ir: 9, full: 589844
Jul  4 17:40:13 allsky [INFO] MainProcess-29996/Sensor-1 tempSensorMlx90614.update() [22]: [Sensor C] MLX90614 - ambient temp: 4.9c, object temp: -6.5c
Jul  4 17:40:13 allsky [INFO] MainProcess-29996/Sensor-1 sensor.saferun() [149]: Sensor SQM: 12.75343
Jul  4 17:40:28 allsky [WARNING] MainProcess-29996/Sensor-1 sensor.night_day_change() [157]: Day/Night change
Jul  4 17:40:28 allsky [INFO] MainProcess-29996/Sensor-1 lightSensorTsl2591.update_sensor_settings() [54]: [Sensor B] Switching TSL2591 to night mode
Jul  4 17:40:29 allsky [INFO] MainProcess-29996/Sensor-1 lightSensorTsl2591.update() [29]: [Sensor B] TSL2591 - lux: 0.4304, visible: 241306948, ir: 3682, full: 241310630
Jul  4 17:40:29 allsky [INFO] MainProcess-29996/Sensor-1 tempSensorMlx90614.update() [22]: [Sensor C] MLX90614 - ambient temp: 4.9c, object temp: -6.4c
Jul  4 17:40:29 allsky [INFO] MainProcess-29996/Sensor-1 sensor.saferun() [149]: Sensor SQM: 13.49882
Jul  4 17:40:45 allsky [INFO] MainProcess-29996/Sensor-1 lightSensorTsl2591.update() [29]: [Sensor B] TSL2591 - lux: 0.4131, visible: 224070772, ir: 3419, full: 224074191
Jul  4 17:40:45 allsky [INFO] MainProcess-29996/Sensor-1 tempSensorMlx90614.update() [22]: [Sensor C] MLX90614 - ambient temp: 4.8c, object temp: -6.7c
Jul  4 17:40:45 allsky [INFO] MainProcess-29996/Sensor-1 sensor.saferun() [149]: Sensor SQM: 13.54354
Jul  4 17:41:00 allsky [INFO] MainProcess-29996/Sensor-1 lightSensorTsl2591.update() [29]: [Sensor B] TSL2591 - lux: 0.3967, visible: 207948719, ir: 3173, full: 207820817
Jul  4 17:41:00 allsky [INFO] MainProcess-29996/Sensor-1 tempSensorMlx90614.update() [22]: [Sensor C] MLX90614 - ambient temp: 4.9c, object temp: -6.7c
```

But earlier after I'd pulled in your latest update I was still getting some overflows, like the gain was still wrong, example logs here:

```
ul  4 15:35:49 allsky [INFO] MainProcess-29996/Sensor-1 sensor.saferun() [149]: Sensor SQM: 4.88870
Jul  4 15:36:04 allsky [ERROR] MainProcess-29996/Sensor-1 sensor.update_sensors() [434]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jul  4 15:36:04 allsky [INFO] MainProcess-29996/Sensor-1 tempSensorMlx90614.update() [22]: [Sensor C] MLX90614 - ambient temp: 13.5c, object temp: -1.8c
Jul  4 15:36:04 allsky [INFO] MainProcess-29996/Sensor-1 sensor.saferun() [149]: Sensor SQM: 4.88870
Jul  4 15:36:19 allsky [ERROR] MainProcess-29996/Sensor-1 sensor.update_sensors() [434]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jul  4 15:36:19 allsky [INFO] MainProcess-29996/Sensor-1 tempSensorMlx90614.update() [22]: [Sensor C] MLX90614 - ambient temp: 13.5c, object temp: -1.6c
Jul  4 15:36:19 allsky [INFO] MainProcess-29996/Sensor-1 sensor.saferun() [149]: Sensor SQM: 4.88870
Jul  4 15:36:34 allsky [INFO] MainProcess-29996/Sensor-1 lightSensorTsl2591.update() [29]: [Sensor B] TSL2591 - lux: 1227.3684, visible: 1146833184, ir: 17499, full: 1146850683
Jul  4 15:36:34 allsky [INFO] MainProcess-29996/Sensor-1 tempSensorMlx90614.update() [22]: [Sensor C] MLX90614 - ambient temp: 13.4c, object temp: -1.6c
Jul  4 15:36:34 allsky [INFO] MainProcess-29996/Sensor-1 sensor.saferun() [149]: Sensor SQM: 4.86112
Jul  4 15:36:49 allsky [ERROR] MainProcess-29996/Sensor-1 sensor.update_sensors() [434]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jul  4 15:36:49 allsky [INFO] MainProcess-29996/Sensor-1 tempSensorMlx90614.update() [22]: [Sensor C] MLX90614 - ambient temp: 13.4c, object temp: -1.5c
Jul  4 15:36:49 allsky [INFO] MainProcess-29996/Sensor-1 sensor.saferun() [149]: Sensor SQM: 4.86112
Jul  4 15:37:04 allsky [INFO] MainProcess-29996/Sensor-1 lightSensorTsl2591.update() [29]: [Sensor B] TSL2591 - lux: 1211.0419, visible: 1139361907, ir: 17385, full: 1139379292
Jul  4 15:37:04 allsky [INFO] MainProcess-29996/Sensor-1 tempSensorMlx90614.update() [22]: [Sensor C] MLX90614 - ambient temp: 13.3c, object temp: -2.1c
Jul  4 15:37:04 allsky [INFO] MainProcess-29996/Sensor-1 sensor.saferun() [149]: Sensor SQM: 4.87566
Jul  4 15:37:20 allsky [INFO] MainProcess-29996/Sensor-1 lightSensorTsl2591.update() [29]: [Sensor B] TSL2591 - lux: 1170.7185, visible: 1108035146, ir: 16907, full: 1108052053
Jul  4 15:37:20 allsky [INFO] MainProcess-29996/Sensor-1 tempSensorMlx90614.update() [22]: [Sensor C] MLX90614 - ambient temp: 13.3c, object temp: -1.8c
Jul  4 15:37:20 allsky [INFO] MainProcess-29996/Sensor-1 sensor.saferun() [149]: Sensor SQM: 4.91243
Jul  4 15:37:35 allsky [INFO] MainProcess-29996/Sensor-1 lightSensorTsl2591.update() [29]: [Sensor B] TSL2591 - lux: 1137.7064, visible: 1080312945, ir: 16484, full: 1080329429
Jul  4 15:37:35 allsky [INFO] MainProcess-29996/Sensor-1 tempSensorMlx90614.update() [22]: [Sensor C] MLX90614 - ambient temp: 13.3c, object temp: -1.6c
```
I'll have another look tomorrow when it transitions to day and see what happens.

aaronwmorris, 2024-07-04:
I merged #1378 which will show the gain settings being used when switching between day and night.

I also added some commented out lines to the `update()` function which will read the gain and integration time registers and log them for every update.  You can uncomment these to verify the device is actually configured.  I have not tested these on a real device yet.

Vertikar, 2024-07-05:
> I merged #1378 which will show the gain settings being used when switching between day and night.
>
> I also added some commented out lines to the `update()` function which will read the gain and [...]

Have pulled this down and uncommented it, will check it out later today and see how it goes

aaronwmorris, 2024-07-05:
Merged #1380 .  After testing this on my Pi with a TSL2591, I realized the settings constants are integers, so I updated the logging accordingly.  This was mainly to add the contant's integers to the descriptions in the config page.

Vertikar, 2024-07-08:
Looking all good after a few day night transitions, happy to close it off if its all good on your side too

aaronwmorris, 2024-07-08:
Fantastic!  Closing for now.  Please re-open if you run into any problems

closed 2024-07-08 by aaronwmorris