aaronwmorris/indi-allsky #1336: multi camera support with raspi5

gmke, 2024-06-19:
Hi Aaron,
I started to play with my raspi5 and its two camera interfaces. I connected one RPI HQ camera and an ov5647 based camera. Both cameras show up in the selector below the left up logo. Changing the camera there changes nothing in the display. Going into config/Camera, I can change the interface to libcamera OV5647, but after a save/reload, the HQ(imx477) camera still gets displayed. The only way to use the ov5647 camera was to add --camera 1 to the libcamera extra options. I guess I'm to dump to configure it correctly or some multicam sw features are buggy or missing. 
Would be nice to operate two cameras with one system, not only switchable but also in parallel. 
Killer feature would be to stitch the two pics together for an allsky panorama using two super widefield lenses instead of a circle fisheye lens :-)

CS, Markus

aaronwmorris, 2024-06-20:
I will work on a camera port selector for libcamera.

As for multi-camera support.  I considered multi-camera support early in the development, but I kind of abandoned the idea due to the added complexity.  I may still focus some effort on running two instances of indi-allsky on the same device, which could still be viewed through the same web interface.  I completely agree that it would be really cool, but I do not want to sacrifice stability for a rarely used feature.

gmke, 2024-06-20:
thank you Aaron. Making the camera selection actually work would be really nice. You could put the multi-camera support into a separate application managing multiple instances of indi-allsky (maybe containerized) by adding a kind of remote control API. 

CS, Markus

aaronwmorris, 2024-06-20:
I have had others manage two cameras on the same Pi5 by running one instance on the Pi5 and the second instance under a docker container.

I do need to update the docker-compose script to make it easier to run multiple docker instances of indi-allsky.

aaronwmorris, 2024-06-26:
Merged #1364 with the camera selection in the config.

TonAmSi, 2024-07-22:
This is an interesting topic.
I have just connected two ZWO cameras via USB3 on a trial basis. Both are recognized, but both are down. 
Is there a way to start one of the two and still leave both connected via USB?

Here is my website for this: allsky.astrodon.space

aaronwmorris, 2024-07-22:
Are the cameras both the same model or different models?

TonAmSi, 2024-07-22:
Hi Aaron,
different models: Asi385MC and Asi715MC
Greetings, Amy
[image]

aaronwmorris, 2024-07-22:
The default behavior of indi-allsky is to query a list of cameras from the indiserver and connect to the first camera returned.  I do not know how INDI decides which camera is "first".  It could change randomly.

In cases where you have two cameras connected, there is a field `INDI Camera Name` on the Camera tab where you can input the camera name, eg "ZWO CCD ASI385MC" and it will only connect to a camera by that name.

Having different cameras makes things easier because the names should remain consistent.  If you have two of the same kind, INDI may append a 1 or 2 at the end of the name and you still do not necessarily know which is which and they order may swap.

TonAmSi, 2024-07-22:
Thanks for the tip!
Entering the name worked.
But exposure and gain are not responding - i'll have to find out why.
[image]

aaronwmorris, 2024-07-22:
Can you post the support info here?

TonAmSi, 2024-07-22:
Here is the info:
/home/darkstar/indi-allsky/misc/support_info.sh: line 61: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 4142240 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux allsky 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

Time
Mon 22 Jul 22:20:37 CEST 2024

System timezone
Europe/Berlin

Uptime
 22:20:37 up 12 min,  3 users,  load average: 0.34, 0.46, 0.31

Memory
               total        used        free      shared  buff/cache   available
Mem:         4142240     1568448     1184848       40320     1495952     2573792
Swap:         204784           0      204784

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1896144        0   1896144   0% /dev
tmpfs             414224     6064    408160   2% /run
/dev/mmcblk0p2 491543056 25093476 441466028   6% /
tmpfs            2071120      960   2070160   1% /dev/shm
tmpfs               5120       48      5072   1% /run/lock
/dev/loop2         34560    34560         0 100% /snap/snapd/21761
/dev/loop0         61184    61184         0 100% /snap/core20/2321
/dev/loop1         44032    44032         0 100% /snap/certbot/3832
/dev/mmcblk0p1    522230    77320    444910  15% /boot/firmware
tmpfs             414224      176    414048   1% /run/user/1000

sysctl info
vm.swappiness = 60

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 2c:cf:67:3a:6b:5c brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:3a:6b:5d brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.209/24 brd 192.168.2.255 scope global dynamic noprefixroute wlan0
       valid_lft 1813660sec preferred_lft 1813660sec
    inet6 2003:e3:cf40:9f95:5bd1:b7b3:d6eb:e0b/64 scope global dynamic noprefixroute 
       valid_lft 172765sec preferred_lft 86365sec
    inet6 fe80::fa8c:a1af:3c26:5760/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
[... 1268 lines cut ...]
    "MQTT_PORT": 8883,
    "MQTT_USERNAME": "indi-allsky",
    "MQTT_TLS": true,
    "MQTT_CERT_BYPASS": true,
    "TSL2561_GAIN_NIGHT": 1,
    "TSL2561_GAIN_DAY": 0,
    "TSL2561_INT_NIGHT": 1,
    "TSL2561_INT_DAY": 1,
    "TSL2591_GAIN_NIGHT": "GAIN_MED",
    "TSL2591_GAIN_DAY": "GAIN_LOW",
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-07-22:
I do not see why the gain and exposure would not work.  Given your location, at the time you were running the test, it was probably still daytime which would make gain 0 the correct setting.  It is probably night by now.

TonAmSi, 2024-07-22:
I also think that the system still believes it is daytime. The gain is now at 70, but exposure is still at 0.00003

TonAmSi, 2024-07-22:
I shone a flashlight on the camera's sensor and now: exposure = 20
I think the problem is solved :-))

aaronwmorris, 2024-07-22:
Okay, I see.  Effectively, it tricked itself into thinking it was generating a useful image.  This should not have happened (duh) so there is obviously some condition here I was not expecting.

This used to be a much bigger problem, before I set the default exposure to a more reasonable value.  Unfortunately, I think another feature I implemented did not help... When you restart indi-allsky, it reuses the last known "good" exposure value so there is no recalculation of the exposure during the timelapse.  It just got stuck at a bad value.

TonAmSi, 2024-07-22:
No problem!
I just figured that if the system wanted daytime - it would get bright light. 😁
But now I can at least try to focus. 
Since the sensor has a pixel size of 1.4 - I still have to look for a suitable lens. Unfortunately, Stardot or Arecont 1.55 is no longer available. But I'll think of something. 👍
But thanks for your help and clear skies!

aaronwmorris, 2024-07-22:
Take a look at the Camera Simulator on your install.  The ASI715 is a smaller sensor, so you will probably need one of the "Small" lenses.

The M12 1.44mm f2.0 1/2.5" lens might be a good fit.  I can be found on Amazon, eBay, and Aliexpress for fairly cheap.  For the ASI715, it would also require an M12 to CS mount adapter.

https://allsky.aarmor.net/indi-allsky/camerasimulator?lens=m12_f2.0_1.44mm_1-2.5&sensor=imx715&offset_x=0&offset_y=0

Lens:  https://www.aliexpress.com/i/3256804312194997.html

Adapter:  https://www.aliexpress.us/item/3256805299289441.html

TonAmSi, 2024-07-22:
Oh, wow. Thank you! 
But: Unfortunately not available in Germany. 🤷🏻‍♀️

But so far the stars look quite good in focus.
[image]

closed 2024-09-14 by aaronwmorris