aaronwmorris/indi-allsky #1338: Having Trouble Getting DHT22 to Show Up

AlaskanAstro, 2024-06-19:
Hello, new problem from me: 

I have a DHT22 temp/humidity sensor and I am kind of out of guesses as to why I can't get it to work in indi-allsky. 
I have it plugged into a bench test raspi4 identical to my regular one and I have a fresh install of indi-allsky. I've got the Sensors tab filled out correctly with GPIO pin location and I'm not sure what else I would need to do. Nothing but CPU Temp shows up under the System Information tab under Environment. 

I followed this guide and was able to get good readings on this Pi, so I know the sensor and hardware are compatible and working. https://pimylifeup.com/raspberry-pi-humidity-sensor-dht22/

Any suggestions? Am I just looking in the

aaronwmorris, 2024-06-20:
Can you dump the output of the Support Info page (under Tools)?  For simple tests of the sensor, you can also run the sensor test script.  I did not have a lot of luck with my DHT21, so I will give that another try.

```
source virtualenv/indi-allsky/bin/activate

./misc/sensor_test.py
```

AlaskanAstro, 2024-06-20:
Here you go. I got a "bash ... no such file or directory" copying those commands into my SSH window. 

/home/nina/indi-allsky/misc/support_info.sh: line 61: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7997480 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux skytest 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

Time
Wed 19 Jun 17:04:49 AKDT 2024

System timezone
America/Anchorage

Uptime
 17:04:49 up 2 min,  3 users,  load average: 0.25, 0.27, 0.11

Memory
               total        used        free      shared  buff/cache   available
Mem:         7997480      703152     6862944       17064      563312     7294328
Swap:         102396           0      102396

Filesystems
Filesystem     1K-blocks    Used Available Use% Mounted on
udev             3725800       0   3725800   0% /dev
tmpfs             799748    1320    798428   1% /run
/dev/mmcblk0p2  60801648 9525532  48167764  17% /
tmpfs            3998740     308   3998432   1% /dev/shm
tmpfs               5120      16      5104   1% /run/lock
/dev/mmcblk0p1    522230   75372    446858  15% /boot/firmware
tmpfs             799748      36    799712   1% /run/user/1000

sysctl info
vm.swappiness = 60

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether d8:3a:dd:19:14:05 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d8:3a:dd:19:14:06 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.199/24 brd 192.168.1.255 scope global dynamic noprefixroute wlan0
       valid_lft 7056sec preferred_lft 7056sec
    inet6 fe80::3528:265c:8e83:4470/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

User info
uid=1000(nina) gid=1000(nina) groups=1000(nina),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),115(lpadmin),993(gpio),994(i2c),995(spi),999(systemd-journal)

[... 925 lines cut ...]
    "THOLD_DIFF_HIGH": 10,
    "THOLD_ENABLE": false
  },
  "TEMP_SENSOR": {
    "A_CLASSNAME": "blinka_temp_sensor_dht22",
    "A_PIN_1": "D5",
    "A_USER_VAR_SLOT": 10,
    "A_I2C_ADDRESS": "0x77",
    "B_CLASSNAME": "",
    "B_PIN_1": "D6",
    "B_USER_VAR_SLOT": 15,
    "B_I2C_ADDRESS": "0x76",
    "C_CLASSNAME": "",
    "C_PIN_1": "D16",
    "C_USER_VAR_SLOT": 20,
    "C_I2C_ADDRESS": "0x40",
    "OPENWEATHERMAP_APIKEY": "REDACTED",
    "OPENWEATHERMAP_APIKEY_E": "REDACTED"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-06-20:
Thanks for the info.

You would have to run those commands from the indi-allsky folder.

AlaskanAstro, 2024-06-20:
Ok, I got this as a result which seems promising. 

[image]

AlaskanAstro, 2024-06-20:
Alright it's working! I have the temp reading out on the overlay. It still doesn't show up under the environment page but I don't care all that much as long as it actually works.

aaronwmorris, 2024-06-20:
Excellent, I am glad the sensor is working!  You are correct about the order of the data values.

AlaskanAstro, 2024-06-20:
Unless you wanted to troubleshoot why the sensors don't show up on the system/environment page, I'd be happy calling this issue closed. It works, I just had to trust that it was there blindly while setting up. In fact I have 2 DHT22s on the same pi both giving unique readouts in the 10/11 and 15/16 slots. Super excited about having a temp/hum for inside and outside my housing.

aaronwmorris, 2024-07-03:
I will see what I can do about getting the sensors to display on the system page.

AlaskanAstro, 2024-07-04:
I recognize that indi-allsky is not supposed to become graphana but it would neat if it's not a lot of work to see the sensor readouts on the charts page too.

aaronwmorris, 2024-07-04:
I have already added the ability to graph the sensors in the charts page.  You may define up to 4 additional graphs from the available sensors in the Admin tab.  #1363 

I will add a note on the config view to point to these settings.

AlaskanAstro, 2024-07-04:
Perfect, had to do the simple upgrade and it's working now. Thanks! Perhaps a tweak for the future but the ability to rename or add to a name for charts could be useful if people have multiple temp sensors, etc.

aaronwmorris, 2024-07-04:
I will see what I can do.

aaronwmorris, 2024-07-04:
Merged #1375 which adds customizable sensor labels.

AlaskanAstro, 2024-07-04:
worked great. thank you!

closed 2024-08-10 by aaronwmorris