aaronwmorris/indi-allsky #1339: Request: Add offset to daytime disable dew_heater

keno242, 2024-06-21:
I've implemented an ON/OFF relay for controlling my dew heater and have the indi-allsy 'Dew Heater - Standard' working - thanks for adding all this in, it's been timely for us down-under! 

Is there a way I can add a delay/offset to the when daytime turn-off control of the dew-heater takes place as my dome temp drops very low before the sun has a chance to warm up the dome. 

You'll see it stars at 9deg, then dips to 3.6deg with the dew_heater off and then get back up to 9deg after about 90mins. Daytime was triggered at 06:43am local time.

```
2024-06-21T06:43:03.292103+08:00 allsky [WARNING] MainProcess-7351/Sensor-1088 dewHeaterStandard.state() [53]: Set dew heater state: 0%
```

See below output from this morning as an example (ignore the BME values, my BME is currently inside my allsky enclosure) but the dome temp sensor (an analog TMP36) is accurate and located in my dome.

```
2024-06-21 06:38:24 - BME280 Temperature: 09.47°C, Pressure: 982.65hPa, Humidity: 65.32%, Dew Point: 02.53°C, Dome Temperature (avg 30s): 09.78°C, Relay State: 1
2024-06-21 06:38:54 - BME280 Temperature: 09.47°C, Pressure: 982.65hPa, Humidity: 65.32%, Dew Point: 02.53°C, Dome Temperature (avg 30s): 09.93°C, Relay State: 1
2024-06-21 06:39:25 - BME280 Temperature: 09.46°C, Pressure: 982.70hPa, Humidity: 65.25%, Dew Point: 02.51°C, Dome Temperature (avg 30s): 09.88°C, Relay State: 1
2024-06-21 06:39:55 - BME280 Temperature: 09.46°C, Pressure: 982.70hPa, Humidity: 65.25%, Dew Point: 02.51°C, Dome Temperature (avg 30s): 09.88°C, Relay State: 1
2024-06-21 06:40:25 - BME280 Temperature: 09.46°C, Pressure: 982.70hPa, Humidity: 65.15%, Dew Point: 02.49°C, Dome Temperature (avg 30s): 09.93°C, Relay State: 1
2024-06-21 06:40:55 - BME280 Temperature: 09.46°C, Pressure: 982.70hPa, Humidity: 65.15%, Dew Point: 02.49°C, Dome Temperature (avg 30s): 09.93°C, Relay State: 1
2024-06-21 06:41:25 - BME280 Temperature: 09.44°C, Pressure: 982.71hPa, Humidity: 65.27%, Dew Point: 02.50°C, Dome Temperature (avg 30s): 09.88°C, Relay State: 1
2024-06-21 06:41:55 - BME280 Temperature: 09.44°C, Pressure: 982.71hPa, Humidity: 65.27%, Dew Point: 02.50°C, Dome Temperature (avg 30s): 09.88°C, Relay State: 1
2024-06-21 06:42:25 - BME280 Temperature: 09.42°C, Pressure: 982.70hPa, Humidity: 65.08%, Dew Point: 02.44°C, Dome Temperature (avg 30s): 09.78°C, Relay State: 1
2024-06-21 06:42:55 - BME280 Temperature: 09.42°C, Pressure: 982.70hPa, Humidity: 65.08%, Dew Point: 02.44°C, Dome Temperature (avg 30s): 09.93°C, Relay State: 1
2024-06-21 06:43:25 - BME280 Temperature: 09.40°C, Pressure: 982.72hPa, Humidity: 65.18%, Dew Point: 02.44°C, Dome Temperature (avg 30s): 08.18°C, Relay State: 0
2024-06-21 06:43:56 - BME280 Temperature: 09.40°C, Pressure: 982.72hPa, Humidity: 65.18%, Dew Point: 02.44°C, Dome Temperature (avg 30s): 07.23°C, Relay State: 0
2024-06-21 06:44:26 - BME280 Temperature: 09.35°C, Pressure: 982.67hPa, Humidity: 65.17%, Dew Point: 02.39°C, Dome Temperature (avg 30s): 07.13°C, Relay State: 0
2024-06-21 06:44:56 - BME280 Temperature: 09.35°C, Pressure: 982.67hPa, Humidity: 65.17%, Dew Point: 02.39°C, Dome Temperature (avg 30s): 06.98°C, Relay State: 0
2024-06-21 06:45:26 - BME280 Temperature: 09.33°C, Pressure: 982.69hPa, Humidity: 65.38%, Dew Point: 02.40°C, Dome Temperature (avg 30s): 06.83°C, Relay State: 0
2024-06-21 06:45:56 - BME280 Temperature: 09.33°C, Pressure: 982.69hPa, Humidity: 65.38%, Dew Point: 02.40°C, Dome Temperature (avg 30s): 06.78°C, Relay State: 0
2024-06-21 06:46:26 - BME280 Temperature: 09.29°C, Pressure: 982.68hPa, Humidity: 65.57%, Dew Point: 02.40°C, Dome Temperature (avg 30s): 06.73°C, Relay State: 0
2024-06-21 06:46:56 - BME280 Temperature: 09.29°C, Pressure: 982.68hPa, Humidity: 65.57%, Dew Point: 02.40°C, Dome Temperature (avg 30s): 06.78°C, Relay State: 0
2024-06-21 06:47:26 - BME280 Temperature: 09.30°C, Pressure: 982.71hPa, Humidity: 65.92%, Dew Point: 02.48°C, Dome Temperature (avg 30s): 06.48°C, Relay State: 0
2024-06-21 06:47:56 - BME280 Temperature: 09.30°C, Pressure: 982.71hPa, Humidity: 65.92%, Dew Point: 02.48°C, Dome Temperature (avg 30s): 06.58°C, Relay State: 0
2024-06-21 06:48:27 - BME280 Temperature: 09.31°C, Pressure: 982.70hPa, Humidity: 66.09%, Dew Point: 02.53°C, Dome Temperature (avg 30s): 06.63°C, Relay State: 0
2024-06-21 06:48:57 - BME280 Temperature: 09.31°C, Pressure: 982.70hPa, Humidity: 66.09%, Dew Point: 02.53°C, Dome Temperature (avg 30s): 06.63°C, Relay State: 0
2024-06-21 06:49:27 - BME280 Temperature: 09.32°C, Pressure: 982.72hPa, Humidity: 66.11%, Dew Point: 02.54°C, Dome Temperature (avg 30s): 06.53°C, Relay State: 0
2024-06-21 06:49:57 - BME280 Temperature: 09.32°C, Pressure: 982.72hPa, Humidity: 66.11%, Dew Point: 02.54°C, Dome Temperature (avg 30s): 06.48°C, Relay State: 0
2024-06-21 06:50:27 - BME280 Temperature: 09.36°C, Pressure: 982.70hPa, Humidity: 65.98%, Dew Point: 02.56°C, Dome Temperature (avg 30s): 06.18°C, Relay State: 0
2024-06-21 06:50:57 - BME280 Temperature: 09.36°C, Pressure: 982.70hPa, Humidity: 65.98%, Dew Point: 02.56°C, Dome Temperature (avg 30s): 06.38°C, Relay State: 0
2024-06-21 06:51:27 - BME280 Temperature: 09.40°C, Pressure: 982.70hPa, Humidity: 65.65%, Dew Point: 02.53°C, Dome Temperature (avg 30s): 06.18°C, Relay State: 0
2024-06-21 06:51:57 - BME280 Temperature: 09.40°C, Pressure: 982.70hPa, Humidity: 65.65%, Dew Point: 02.53°C, Dome Temperature (avg 30s): 06.03°C, Relay State: 0
2024-06-21 06:52:27 - BME280 Temperature: 09.43°C, Pressure: 982.65hPa, Humidity: 65.75%, Dew Point: 02.58°C, Dome Temperature (avg 30s): 06.08°C, Relay State: 0
2024-06-21 06:52:58 - BME280 Temperature: 09.43°C, Pressure: 982.65hPa, Humidity: 65.75%, Dew Point: 02.58°C, Dome Temperature (avg 30s): 06.03°C, Relay State: 0
2024-06-21 06:53:28 - BME280 Temperature: 09.48°C, Pressure: 982.71hPa, Humidity: 65.33%, Dew Point: 02.55°C, Dome Temperature (avg 30s): 05.93°C, Relay State: 0
2024-06-21 06:53:58 - BME280 Temperature: 09.48°C, Pressure: 982.71hPa, Humidity: 65.33%, Dew Point: 02.55°C, Dome Temperature (avg 30s): 05.68°C, Relay State: 0
2024-06-21 06:54:28 - BME280 Temperature: 09.53°C, Pressure: 982.71hPa, Humidity: 65.29%, Dew Point: 02.59°C, Dome Temperature (avg 30s): 05.68°C, Relay State: 0
2024-06-21 06:54:58 - BME280 Temperature: 09.53°C, Pressure: 982.71hPa, Humidity: 65.29%, Dew Point: 02.59°C, Dome Temperature (avg 30s): 05.73°C, Relay State: 0
2024-06-21 06:55:28 - BME280 Temperature: 09.59°C, Pressure: 982.70hPa, Humidity: 64.95%, Dew Point: 02.58°C, Dome Temperature (avg 30s): 05.53°C, Relay State: 0
2024-06-21 06:55:58 - BME280 Temperature: 09.59°C, Pressure: 982.70hPa, Humidity: 64.95%, Dew Point: 02.58°C, Dome Temperature (avg 30s): 05.38°C, Relay State: 0
2024-06-21 06:56:28 - BME280 Temperature: 09.64°C, Pressure: 982.70hPa, Humidity: 64.75%, Dew Point: 02.59°C, Dome Temperature (avg 30s): 05.33°C, Relay State: 0
2024-06-21 06:56:58 - BME280 Temperature: 09.64°C, Pressure: 982.70hPa, Humidity: 64.75%, Dew Point: 02.59°C, Dome Temperature (avg 30s): 05.28°C, Relay State: 0
2024-06-21 06:57:29 - BME280 Temperature: 09.68°C, Pressure: 982.71hPa, Humidity: 64.54%, Dew Point: 02.59°C, Dome Temperature (avg 30s): 05.33°C, Relay State: 0
2024-06-21 06:57:59 - BME280 Temperature: 09.68°C, Pressure: 982.71hPa, Humidity: 64.54%, Dew Point: 02.59°C, Dome Temperature (avg 30s): 05.13°C, Relay State: 0
2024-06-21 06:58:29 - BME280 Temperature: 09.72°C, Pressure: 982.71hPa, Humidity: 64.30%, Dew Point: 02.58°C, Dome Temperature (avg 30s): 05.08°C, Relay State: 0
2024-06-21 06:58:59 - BME280 Temperature: 09.72°C, Pressure: 982.71hPa, Humidity: 64.30%, Dew Point: 02.58°C, Dome Temperature (avg 30s): 04.98°C, Relay State: 0
2024-06-21 06:59:29 - BME280 Temperature: 09.74°C, Pressure: 982.71hPa, Humidity: 64.19%, Dew Point: 02.58°C, Dome Temperature (avg 30s): 04.93°C, Relay State: 0
2024-06-21 06:59:59 - BME280 Temperature: 09.74°C, Pressure: 982.71hPa, Humidity: 64.19%, Dew Point: 02.58°C, Dome Temperature (avg 30s): 04.88°C, Relay State: 0
2024-06-21 07:00:29 - BME280 Temperature: 09.78°C, Pressure: 982.66hPa, Humidity: 63.81%, Dew Point: 02.55°C, Dome Temperature (avg 30s): 04.88°C, Relay State: 0
2024-06-21 07:00:59 - BME280 Temperature: 09.78°C, Pressure: 982.66hPa, Humidity: 63.81%, Dew Point: 02.55°C, Dome Temperature (avg 30s): 04.88°C, Relay State: 0
[... cut ...]
BTW not sure if it's possible but I the analog temp sensor I use is connected via a Pimoroni explorerhat ( should probably switch to a digital sensor) - but where might I change some of your code to be able to use this temp sensor in your framework by either customising some of your code by importing the https://github.com/pimoroni/explorer-hat or even running an external script and just customising having your temp sensor module read a file generated by my external script, like you do for the camera temp option.

Thanks,
Ken.

aaronwmorris, 2024-06-21:
I have it in my plans to support analog temperature sensors such as TMP36 or LM35 via an ADC.  In fact, I wired up an LM35 to test with a Raspi, but realized the Pi does not have a native ADC... doh!  It looks like the explorer hat just has an ADS1015 which should not be complicated.

Let me think about the day time offset for the dew heater.

aaronwmorris, 2024-06-22:
I have added the code to support the TMP36 module via the ADS1x15 ADC that is on the Explorer HAT.  If you get a chance, can you validate it works and the temperature is correct?

keno242, 2024-06-23:
Nice one Aaron - with a bit of troubleshooting I was able to get the TMP36 reading as a sensor through the explorerhat pro

It appears that for my explorerhat pro `explorerhat.analog.one.read()` maps to Channel 3, so using the following sensor config worked:

```
+------------------------------------------------------------+
|                                                            |
| Sensor B                                                   |
|                                                            |
| +--------------------------------------------------------+ |
| | TMP36 ADS1015 i2c - Temp (1)                         v | |
| +--------------------------------------------------------+ |
|                                                            |
| Pin                                                        |
| +----------------+  Example: D6 (GPIO6 / Pin 31 Raspi)     |
| | P3             |  Example: P0 - Pin for ADS1x15 ADC      |
| +----------------+  Unused for most i2c devices            |
|                                                            |
| I2C Address                                                |
| +----------------+  Hexidecimal address - Example: 0x76    |
| | 0x48           |                                         |
| +----------------+                                         |
|                                                            |
| Sensor B Slot                                              |
| +--------------------------------------------------------+ |
| | User Slot 15                                         v | |
| +--------------------------------------------------------+ |
|                                                            |
+------------------------------------------------------------+
```

For anyone else here is the code Claude helped me out with to see which channel (P?) worked with my setup...

`explorerhat_temp_i2c.py`
```
#!/usr/bin/env python3

import time
import sys
from smbus import SMBus

def i2c_bus_id():
    revision = ([l[12:-1] for l in open('/proc/cpuinfo', 'r').readlines() if l[:8] == "Revision"] + ['0000'])[0]
    return 1 if int(revision, 16) >= 4 else 0

# ADS1015 configuration
address = 0x48
REG_CONV = 0x00
REG_CFG = 0x01

# Initialize I2C bus
i2c = SMBus(i2c_bus_id())

def read_adc(channel):
    # Configuration settings
    config = 0x0003 | 0x0100  # COMP_QUE and COMP_LAT
    config |= 0x0020  # 250 samples per second
    config |= (0x4000 | (channel << 12))  # Set channel
    config |= 0x0200  # ±4.096V range
    config |= 0x8000  # Start a single conversion

    # Write configuration
    i2c.write_i2c_block_data(address, REG_CFG, [(config >> 8) & 0xFF, config & 0xFF])

    # Wait for conversion to complete
    time.sleep(1.0 / 250 + 0.001)  # Wait a bit more than the conversion time

    # Read conversion result
    data = i2c.read_i2c_block_data(address, REG_CONV, 2)
    value = (data[0] << 4) | (data[1] >> 4)

    # Convert to voltage
    voltage = value * 4.096 / 2047.0

    return voltage

def calculate_temperature(voltage):
    # ExplorerHAT Pro temperature calculation
    return ((voltage * 1000) - 500) / 10

print("ExplorerHAT Pro All Channel Reader")
print("Press Ctrl+C to exit")

try:
    while True:
        voltages = [read_adc(channel) for channel in range(4)]
        temperatures = [calculate_temperature(v) for v in voltages]

        print("\nChannel Readings:")
        for i, (voltage, temperature) in enumerate(zip(voltages, temperatures)):
            print(f"Channel {i}: Voltage: {voltage:.3f}V, Temperature: {temperature:.2f}°C")

        time.sleep(1)

except KeyboardInterrupt:
    print("\nScript terminated by user")
except Exception as e:
    print(f"An error occurred: {e}")
    sys.exit(1)

```

aaronwmorris, 2024-06-23:
Is the temperature correct?  It looks like my algorithm is basically the same, where mine is based on volts and the above example is milli-volts.

As an FYI, you can use the script `./misc/sensor_test.py` script to test the sensors without starting indi-allsky.  This script uses the same code and config as indi-allsky to ensure the behavior matches.

```
source virtualenv/indi-allsky/bin/activate

./misc/sensor_test.py
```

closed 2024-06-26 by aaronwmorris