aaronwmorris/indi-allsky #1347: Timelapses failing to generate - choosing wrong timelapse to generate (night/day)

zaphus, 2024-06-23:
I've just started a clean install of indi-allsky on a larger SD card, and am trying to do both day and night timelapses.
When it comes to the time to create the timelapse, indi-allsky is failing and appears to be trying to generate the wrong video (ie night when it should be day, and day when it should be night) and it has the new-days date on it rather than yesterdays date (since the night just finished it should be yesterdays date)
Everything else is behaving - moon mode, sunrise, sunset.  Time and Location are correct.
I can manually create the videos, but not automatically
In the log below you can see 5168 failed to generate the day sequence, however I am in Australia, that time is GMT, and it is actually 7am here and it was a night sequence.  The manual run 
[allsky-support-info.txt](https://github.com/user-attachments/files/15946174/allsky-support-info.txt)
5183 generated the files correctly.

I am running 2024.05.01, and have also done an update to the latest as of 23rd June 2024 

Running Date on the Raspberry Pi shows the correct time and timezone
zaph@allskycam:~ $ date
Mon 24 Jun 07:15:43 AEST 2024

Tasks
ID Date Queue Action State Result
5183 2024-06-23 21:08:20 VIDEO generateVideo SUCCESS Generated timelapse: /var/www/html/allsky/images/ccd_c015092d-ba0f-4339-8152-9811f7314f80/timelapse/**20240623/allsky-timelapse_ccd1_20240623_night.mp4**
5184 2024-06-23 21:08:20 VIDEO generateKeogramStarTrails RUNNING None
5174 2024-06-23 21:06:09 VIDEO updateAuroraData SUCCESS  Aurora data updated
5168 2024-06-23 21:05:23 VIDEO generateVideo FAILED Failed to generate timelapse: /var/www/html/allsky/images/ccd_c015092d-ba0f-4339-8152-9811f7314f80/timelapse/**20240624/allsky-timelapse_ccd1_20240624_day.mp4**
5169 2024-06-23 21:05:23 VIDEO generateKeogramStarTrails RUNNING None
5170 2024-06-23 21:05:23 VIDEO expireData SUCCESS Expired data
5044 2024-06-23 20:03:45 VIDEO updateAuroraData SUCCESS Aurora data updated

indi-allsky time details:
NTP Available True
NTP Enabled True
NTP Synchronized True
RTC Available False
Timezone Australia/Melbourne
System time    06 / 24 / 2024, 07 : 18 : 31 am

aaronwmorris, 2024-06-23:
I just discovered this regression myself.  I have been recently working on fixes for latitudes where the sun never rises/sets and introduced this bug.

I am working on a fix.

aaronwmorris, 2024-06-24:
Merged #1348 to fix the problem with the wrong timelapses being generated.

zaphus, 2024-06-24:
Excellent, I have grabbed the update (slightly too late to test on the day-night transition) - will confirm that it works in the morning!

Dentonknifeworks, 2024-06-24:
I did a git pull this morning and getting an error
Error: While importing 'app', an ImportError was raised:
 
Traceback (most recent call last):
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/cli.py", line 245, in locate_app
    __import__(module_name)
  File "/home/pi/indi-allsky/app.py", line 3, in <module>
    from indi_allsky.flask import create_app
  File "/home/pi/indi-allsky/indi_allsky/flask/__init__.py", line 17, in <module>
    from .views import bp_allsky  # noqa: E402
  File "/home/pi/indi-allsky/indi_allsky/flask/views.py", line 22, in <module>
    from ..processing import ImageProcessor
  File "/home/pi/indi-allsky/indi_allsky/processing.py", line 11, in <module>
    import cv2
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/cv2/__init__.py", line 153, in bootstrap
    native_module = importlib.import_module("cv2")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: numpy.core.multiarray failed to import
 
 
Usage: flask [OPTIONS] COMMAND [ARGS]...
Try 'flask --help' for help.
 
Error: No such command 'db'.
 
 
###############
###  ERROR  ###
###############
 
The setup script exited abnormally, please try to run again...

aaronwmorris, 2024-06-24:
Did you receive this when running setup.sh ?

Dentonknifeworks, 2024-06-24:
Yes

aaronwmorris, 2024-06-24:
I am going to move this to a new issue.

zaphus, 2024-06-24:
Just confirming that#1348 has fixed the timelapses not generating issue that was recently introduced

henrikeri, 2024-07-05:
@aaronwmorris it works for me as well. But I am getting this annoying purple tint when there are scattered clouds at night time:
[image]

aaronwmorris, 2024-07-21:
FYI, there was a bug reported by another user due to the code refactored for this fix.

#1414

closed 2024-08-10 by aaronwmorris