aaronwmorris/indi-allsky #1357: Install error

pierrebilb, 2024-06-24:
I try to install on orange pi 3B with debian bookworm; i get that in the terminal:
The following packages have unmet dependencies:
 libmount-dev : Depends: libblkid-dev but it is not going to be installed
                Depends: libmount1 (= 2.38.1-5+b1) but 2.38.1-5+deb12u1 is to be installed
 libxft-dev : Depends: libfontconfig1-dev
E: Unable to correct problems, you have held broken packages.

aaronwmorris, 2024-06-24:
You may try doing a full package upgrade first and then see if it resolves.

```
sudo apt-get update

sudo apt-get dist-upgrade
```

pierrebilb, 2024-06-25:
Hi Aaron,
still the same error:
The following packages have unmet dependencies:
 libmount-dev : Depends: libblkid-dev but it is not going to be installed
                Depends: libmount1 (= 2.38.1-5+b1) but 2.38.1-5+deb12u1 is to be installed
 libxft-dev : Depends: libfontconfig1-dev
E: Unable to correct problems, you have held broken packages.

aaronwmorris, 2024-06-26:
Is this a 32-bit distribution or 64-bit?

pierrebilb, 2024-06-27:
It'a 64 bit distribution; the debian image found at this page:
http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-3B.html
Thanks for your help!
Pierre

pierrebilb, 2024-06-29:
I installed instead the ubuntu distribution and everything went fine! No more errors.

closed 2024-06-29 by pierrebilb