aaronwmorris/indi-allsky #1366: Expose latest generated timelapse as "latest.mp4"

henrikeri, 2024-06-27:
Hi Aaron,

It is possible relatively easily to expose the latest timelapse as "latest.mp4" similar to the latest photo? I expose the photo on a separate web page and would like to add the video as well if there is an easy way for indi-allsky to put it as the name above in a folder. I use this to show the data to non-tech-savvy users whom one could not expect to be able to open it from the timelapse menu after entering user names and passwords :)

Best regards,
Henrik

aaronwmorris, 2024-06-28:
Merged #1368 .  This uses the "watch timelapse" view to show the latest timelapse if no ID is specified:

https://hostname/indi-allsky/watch_timelapse

henrikeri, 2024-06-28:
Thanks Aaron this is awesome
Is there a meaningful way to access this server side? I am exposing this on a separate webpage to my non-tech users

aaronwmorris, 2024-06-28:
Querying the database is how that view does it.  What programming language would you want to do this in?

henrikeri, 2024-07-05:
@aaronwmorris I hacked together a bash script that finds the last changed folder and file and combines it to one with ffmpeg. Works for now so I'll close this.

aaronwmorris, 2024-07-05:
Let me know if you want me to write something in python, it should be pretty simple

henrikeri, 2024-08-19:
@aaronwmorris any chance we could add an option to generate a timelapse of the last 24h and store it as yesterday.mp4? The bash script seems to be misbehaving a bit :-/

henrikeri, 2024-08-20:
I would prefer noon-to-noon, but why not let the user set the time from a dropdown? Then it will not have to be picked up again if someone wants midnight-to-midnight or similar

aaronwmorris, 2024-08-20:
I just developed the mini-timelapse generator that does just that.  I also just submitted a change to increase the window.

If you update to the latest code, find the midpoint for the video you want to create (say like at midnight) and select a window of 12 hours before and 12 hours after, set a FPS, name, and generate.

henrikeri, 2024-08-21:
I cannot seem to find where you configure the mini panorama, I only find where you view them. Is it not yet in the main branch? Will it be able to run automatically in the background?

aaronwmorris, 2024-08-21:
Sorry, I meant to add this.  It is in the Image viewer.  The original intent of the Mini-Timelapse generator is picking an event you want to showcase, generating a video with frames before and after the event.  However, with the extended time options, you could pick midnight as the image and add 12-hours before and after the event for a full 24-hour timelapse.

henrikeri, 2024-08-22:
Tested it now, it creates a time lapse that works as intended. Is it a large work/any way to run this on a schedule to generate daily with a known file name? I use this outside of indi-allsky. I am ok triggering it from the terminal if reqd.

aaronwmorris, 2024-08-23:
There's nothing in the code that would limit the time to before or after noon.

Ultimately, it would probably be easier to write an external script that generates a timelapse.  The trouble is dealing with timestamps on the command line.  Let me sit with this for a few days.

aaronwmorris, 2024-12-09:
Not yet.