aaronwmorris/indi-allsky #1376: Light Sensor TSL2591 - Night to day Transition doesn't seem to work

Vertikar, 2024-07-04:
Hey Aaron, the day to night sensor gain seemed to work, ok, but doesn't seem to have worked the other way.
Either that or the sensor died overnight

### Relevant logs:

```
Jul  4 07:05:13 allsky [WARNING] MainProcess-11701/Sensor-3 sensor.night_day_change() [154]: Day/Night change
Jul  4 07:05:13 allsky [INFO] MainProcess-11701/Sensor-3 lightSensorTsl2591.update_sensor_settings() [58]: [B] Switching TSL2591 to day mode
Jul  4 07:05:14 allsky [INFO] MainProcess-11701/Sensor-3 lightSensorTsl2591.update() [29]: [B] TSL2591 - lux: 0.4734, visible: 126159178, ir: 1925, full: 126161103
Jul  4 07:05:14 allsky [INFO] MainProcess-11701/Sensor-3 tempSensorMlx90614.update() [22]: [C] MLX90614 - ambient temp: -1.7c, object temp: -6.5c
```

<snip>

```
Jul  4 07:13:32 allsky [INFO] MainProcess-11701/Sensor-3 lightSensorTsl2591.update() [29]: [B] TSL2591 - lux: 2.6513, visible: 913652716, ir: 13941, full: 913666657
Jul  4 07:13:32 allsky [INFO] MainProcess-11701/Sensor-3 tempSensorMlx90614.update() [22]: [C] MLX90614 - ambient temp: -1.6c, object temp: -6.0c
Jul  4 07:13:32 allsky [INFO] MainProcess-11701/Sensor-3 sensor.saferun() [146]: Sensor SQM: 11.52490
Jul  4 07:13:47 allsky [INFO] MainProcess-11701/Sensor-3 lightSensorTsl2591.update() [29]: [B] TSL2591 - lux: 2.7858, visible: 969817942, ir: 14798, full: 969832740
Jul  4 07:13:47 allsky [INFO] MainProcess-11701/Sensor-3 tempSensorMlx90614.update() [22]: [C] MLX90614 - ambient temp: -1.6c, object temp: -5.8c
Jul  4 07:13:47 allsky [INFO] MainProcess-11701/Sensor-3 sensor.saferun() [146]: Sensor SQM: 11.47118
Jul  4 07:14:02 allsky [INFO] MainProcess-11701/Sensor-3 lightSensorTsl2591.update() [29]: [B] TSL2591 - lux: 2.9349, visible: 1028408059, ir: 15692, full: 1028423751
Jul  4 07:14:02 allsky [INFO] MainProcess-11701/Sensor-3 tempSensorMlx90614.update() [22]: [C] MLX90614 - ambient temp: -1.5c, object temp: -5.9c
Jul  4 07:14:02 allsky [INFO] MainProcess-11701/Sensor-3 sensor.saferun() [146]: Sensor SQM: 11.41459
Jul  4 07:14:18 allsky [INFO] MainProcess-11701/Sensor-3 lightSensorTsl2591.update() [29]: [B] TSL2591 - lux: 3.0912, visible: 1091651295, ir: 16657, full: 1091667952
Jul  4 07:14:18 allsky [INFO] MainProcess-11701/Sensor-3 tempSensorMlx90614.update() [22]: [C] MLX90614 - ambient temp: -1.6c, object temp: -5.9c
Jul  4 07:14:18 allsky [INFO] MainProcess-11701/Sensor-3 sensor.saferun() [146]: Sensor SQM: 11.35824
Jul  4 07:14:33 allsky [ERROR] MainProcess-11701/Sensor-3 sensor.update_sensors() [389]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jul  4 07:14:33 allsky [INFO] MainProcess-11701/Sensor-3 tempSensorMlx90614.update() [22]: [C] MLX90614 - ambient temp: -1.6c, object temp: -5.9c
Jul  4 07:14:33 allsky [INFO] MainProcess-11701/Sensor-3 sensor.saferun() [146]: Sensor SQM: 11.35824
Jul  4 07:14:48 allsky [ERROR] MainProcess-11701/Sensor-3 sensor.update_sensors() [389]: SensorReadException: Overflow reading light channels!, Try to reduce the gain of#012 the sensor using adafruit_tsl2591.GAIN_LOW
Jul  4 07:14:48 allsky [INFO] MainProcess-11701/Sensor-3 tempSensorMlx90614.update() [22]: [C] MLX90614 - ambient temp: -1.5c, object temp: -5.8c
Jul  4 07:14:48 allsky [INFO] MainProcess-11701/Sensor-3 sensor.saferun() [146]: Sensor SQM: 11.35824

```

### Support info:
/opt/indi-allsky/misc/support_info.sh: line 61: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3930360 kB

System: Raspberry Pi 4 Model B Rev 1.1

Linux allsky 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

Time
Thu  4 Jul 14:36:25 AEST 2024

System timezone
Australia/Melbourne

Uptime
 14:36:25 up 1 day, 20:29,  5 users,  load average: 0.18, 0.19, 0.34

Memory
               total        used        free      shared  buff/cache   available
Mem:         3930360      555496      194444        3336     3180420     3294308
Swap:         102396        1792      100604

Filesystems
[... 1200 lines cut ...]
  "GPS_TIMESYNC": false,
  "IMAGE_LABEL": true,
  "SQLALCHEMY_DATABASE_URI": "sqlite:////var/lib/indi-allsky/indi-allsky.sqlite",
  "comment_SCNR_ALGORITHM": "empty string, average_neutral, or maximum_neutral",
  "comment_TEMP_DISPLAY": "c = celcius, f = fahrenheit, k = kelvin",
  "comment_TARGET_ADU": "Average channel brightness target",
  "comment_TARGET_ADU_DEV": "Allowed deviation from the mean before recalculating",
  "comment_ADU_ROI": "Region of Interest for ADU calculations",
  "comment_DETECT_STARS": "Enable Star detection",
  "comment_DETECT_METEORS": "Enable Meteor detection",
  "comment_DETECT_DRAW": "Enable drawing detections on original image",
  "comment_SQM_ROI": "Region of Interest for SQM and Star detection",
  "comment_IMAGE_FILE_TYPE": "jpg, png, or tif",
  "comment_IMAGE_FILE_COMPRESSION": "0-100 for jpg, 0-9 for png",
  "comment_IMAGE_DIR": "local base folder for images, empty for current dir",
  "comment_IMAGE_EXTRA_TEXT": "File containing extra text to add to image",
  "comment_IMAGE_EXPORT_RAW": "png or tif (or empty)",
  "VERSION": 20230109
}

```
#################################
###     end support info      ###
#################################
```

Vertikar, 2024-07-04:
Found it I think -

https://github.com/aaronwmorris/indi-allsky/blob/5292d054733a86fc37766ae893133eb7a88c385c/indi_allsky/devices/sensors/lightSensorTsl2591.py#L104

Should be GAIN_DAY not NIGHT

aaronwmorris, 2024-07-04:
Yep, I just found the typo myself.   Merged #1377 for the fix.

closed 2024-07-05 by aaronwmorris

WirthmU, 2024-07-15:
I just installed the latest version of indi-allsky. I have a number of sensors of which I started with the TSL2561. I could configure it using the I2C correct I2C address. I can see the four readout graphs in the graph section. However all values are 0.
Then I did run the sensor_test.py provided and I get the following result:
(indi-allsky) pi@meteotux:~/indi-allsky $ ./misc/sensor_test.py
[WARNING]: Initializing [Sensor A] TSL2561 I2C light sensor device @ 0x29
[WARNING]: No sensor B - Initializing sensor simulator
[WARNING]: No sensor C - Initializing sensor simulator
[INFO]: [Sensor A] Switching TSL2561 to day mode - Gain: 0, Integration: 1
[ERROR]: SensorReadException: float() argument must be a string or a real number, not 'NoneType'
[INFO]: Sensor 1: {'data': ()}
[INFO]: Sensor 2: {'data': ()}