aaronwmorris/indi-allsky #1381: Git password requested

Bodmer, 2024-07-05:
I tried this on my RPi as per "getting started" instructions on this site:

git clone https://github.com/aaronwmorris/indi-allsky.git

However I get asked for a login and password.

This seems to be the correct command line:

git clone git://github.com/aaronwmorris/indi-allsky.git

but then I get the error:

fatal: unable to connect to github.com:
github.com[0: 20.26.156.215]: errno=Connection timed out

Bodmer, 2024-07-05:
Probably a typo on my part!

aaronwmorris, 2024-07-05:
I am not sure if you meant to close this, but I believe the URL is correct in the getting started page.  You may have experienced a temporary network issue trying to perform the clone.

closed 2024-07-05 by aaronwmorris