aaronwmorris/indi-allsky #1383: I am getting this annoying purple tint

aaronwmorris, 2024-07-05:
@aaronwmorris it works for me as well. But I am getting this annoying purple tint when there are scattered clouds at night time:
[image]

_Originally posted by @henrikeri in https://github.com/aaronwmorris/indi-allsky/issues/1347#issuecomment-2211464420_

aaronwmorris, 2024-07-05:
@hengkit can you post the support info for your system?

henrikeri, 2024-07-05:
@aaronwmorris 

Note that I got this error while running the script, might be related to the fact that the camera is on  a remote computer:

ls: cannot access '/dev/video*': No such file or directory
./support_info.sh: line 169: v4l2-ctl: command not found
2024-07-06 00:42:42,254 [INFO] MainProcess config._dump() [942]: Dumping config

Any way I can send the file to you in an email or similar? Contains info I'd rather not put on the open internet :)

aaronwmorris, 2024-07-05:
Those are normal messages.  You can redact whatever you want from the support info.  I have already done a lot of work to remove sensitive info from the dump such as passwords and the latitude and longitude is reduced to an integer from a floating point to reduce the identifying info.  I am going to also add the OWNER field to the redacted info.

aaronwmorris, 2024-07-06:
Do you have a timelapse you could share somewhere?  Its fine if not.

henrikeri, 2024-07-06:
Will this do? 
https://imgur.com/a/TOPoXqY

aaronwmorris, 2024-07-06:
My guess is it is probably the stretch standard deviation cutoff.  A value of `1` will cause a fairly extreme stretch.  You might try increasing to `1.5`

aaronwmorris, 2024-07-23:
@henrikeri Has there been any change?

aaronwmorris, 2024-08-10:
Please re-open this if the issue persists.

closed 2024-08-10 by aaronwmorris