aaronwmorris/indi-allsky #1386: Possible?

sonny486, 2024-07-06:
Hey thanks for the addition of the DAY/NIGHT GPIO!

It will workout great!

One thing I had a question on.

Is it possible to change the camera file from DWG to JPG during the day?  It seems the DWG format has a contrast issue I cannot resolve, however if I use JPG with the AWB on camera it works great.

At night, obviously the DWG (RAW) file works great!  But for me, the day is too hard to fix with the contrast enhancement and the streach controls.

aaronwmorris, 2024-07-06:
I am working on this, but I will need to run some tests before I release this.

aaronwmorris, 2024-07-06:
Merged #1387 with separate day and night settings for the file type.

sonny486, 2024-07-07:
Wow, works great!  Had to delete on my primary and re-install, but working fantastic so far!

Sent you a little donation.  Thank you for the hard work on this!

aaronwmorris, 2024-07-07:
That was very kind!  Thank you!

sonny486, 2024-07-09:
Thanks works perfectly!

closed 2024-07-09 by sonny486