aaronwmorris/indi-allsky #1392: Two items - Just requests

sonny486, 2024-07-09:
Is it possible to create videos from the loop tab?

It would be nice to create short videos to upload when storms pass

Something that would take the current loop time and create an MP4 from those frames would be cool.

Or could it be possible to overwrite the timelapse instead of erroring out because there is already a time lapse present?

I would put a bounty on that feature!

aaronwmorris, 2024-07-10:
Creating mini-movies is definitely on my radar.  Let me think about how to implement this.

I will consider the overwrite.  I will have to think if there was a reason I did it this way.  I do not remember anything specifically.

aaronwmorris, 2024-08-09:
Merged #1446 for both requests.

There is now a new file type in indi-allsky, the mini-timelapse.  Browse images using the Image Viewer and find the image you want to base the timelapse on.  Click the `Mini Timelapse` icon at the bottom.  This will take you to the Mini Timelapse Generator view.  This view is similar to the Loop view, and you can select a time frame before and after the selected image.  You may also select the Framerate for the timelapse.  As you make changes, the animation should roughly match what will be generated with the timelapse.

There is also a Mini-Timelapse viewer to view the media.

There is now also an option in the Admin Tab of the config view to allow direct overwriting of timelapses if you wish.  The default is to NOT allow overwriting (aka the current behavior).

sonny486, 2024-08-09:
Wow, will test when I get home!

sonny486, 2024-08-11:
Worked as expected!  Great work!

closed 2024-08-11 by sonny486