aaronwmorris/indi-allsky #1399: TSL2561 not working

WirthmU, 2024-07-15:
I just installed the latest version of indi-allsky. I have a number of sensors of which I started with the TSL2561. I could configure it using the I2C correct I2C address. I can see the four readout graphs in the charts section. However all values are 0. The other sensors are fine (i.E. camera temperature or jSQM).

Then I did run the sensor_test.py script provided and I get the following result:
```
(indi-allsky) pi@meteotux:~/indi-allsky $ ./misc/sensor_test.py
[WARNING]: Initializing [Sensor A] TSL2561 I2C light sensor device @ 0x29
[WARNING]: No sensor B - Initializing sensor simulator
[WARNING]: No sensor C - Initializing sensor simulator
[INFO]: [Sensor A] Switching TSL2561 to day mode - Gain: 0, Integration: 1
[ERROR]: SensorReadException: float() argument must be a string or a real number, not 'NoneType'
[INFO]: Sensor 1: {'data': ()}
[INFO]: Sensor 2: {'data': ()}

```

Any help to this?

_Originally posted by @WirthmU in https://github.com/aaronwmorris/indi-allsky/issues/1376#issuecomment-2228337297_

WirthmU, 2024-07-15:
It is working now - it just needed a re-boot.

closed 2024-07-15 by WirthmU

aaronwmorris, 2024-07-15:
Just an FYI, it is noted in the library for the TSL2561 that the lux value can return a `None` value, so you may occasionally see this error, but indi-allsky should handle it gracefully.