aaronwmorris/indi-allsky #1400: DHT22 not working

WirthmU, 2024-07-15:
I am using a GrovePi Plus HAT with several sensors attached. The board has both digital and analog ports. I could readout the sensor values without any problem with previous OS versions of the RPI. Now after switching to bookworm the board seems no longer supported because of missing python drivers. The digital sensors still are visible via I2C and can be used. However the analogue DHT22 (AM2302; attached to D4 / Pin 12 / PWM0) and the GUVA-S12D (attached to A0) sensors can no longer be read.

Is there any chance the two sensors can get read?

aaronwmorris, 2024-07-15:
For the DHT22, from what I have read on the Seeedstudio wiki, it is wired directly to the D4 pin of the Raspberry Pi, so you should be able to read that using the existing DHT22 integration in indi-allsky.

I am looking for references for the GUVA-S12D

Just putting this here for reference.  https://wiki.seeedstudio.com/GrovePi_Plus/

aaronwmorris, 2024-07-22:
Do you have the pull up resistor installed for the DHT22?

WirthmU, 2024-07-22:
No - I have not. The Seeedstudio DHT22 pro comes with a cable which one uses to plug it to the D4 port of GrovePi_Plus HAT (https://wiki.seeedstudio.com/Grove-Temperature_and_Humidity_Sensor_Pro/)

aaronwmorris, 2024-07-22:
Can you confirm which hat you have?  `GrovePi Plus for Raspberry Pi` or `GrovePi+`

WirthmU, 2024-07-22:
It's a GrovePi+ (the blue one).

aaronwmorris, 2024-07-22:
Okay, that explains my confusion.  I got the module conflated when I was researching.  For the GrovePi+, the digital pins are hooked to the atmega328 and you have to access the values via the I2C connection between the atmel chip and the raspberry pi.  This is not going to work with the existing code.

Let me do some more research.

WirthmU, 2024-09-18:
Any news about the Seeedstudio DHT22 pro sensor?

aaronwmorris, 2024-09-19:
Just adding some links for notes...

https://github.com/DexterInd/GrovePi/blob/master/Software/Python/grovepi.py
https://github.com/DexterInd/RFR_Tools/blob/master/miscellaneous/di_i2c.py

aaronwmorris, 2025-02-20:
I do not think I am going to get to adding support for this board.  The library code is a bit of a mess.  The [code] developer only put enough effort into this for one-off type builds and it makes it difficult to support as a 3rd party.

closed 2025-02-20 by aaronwmorris

WirthmU, 2025-04-19:
Does this also apply for the GUVA-S12D (attached to A0/analogue) sensor?