aaronwmorris/indi-allsky #1404: Documentation of Sensor Implementation and imperial conversion

DamoclesSword, 2024-07-18:
I am not sure if this the correct place to do this, but there seems to be a lack of documentation of how to implement data from the Wunderground API via the sensors. I figured out how to do it, but I think clear steps should be in the wiki. 

The thing I still cannot find is how to report these values in imperial units.

aaronwmorris, 2024-07-18:
I will add a note in the wiki and the config form.  You can change from metric to imperial by changing the Temperature display from Celcius to Fahrenheit on the Camera tab.

aaronwmorris, 2024-07-18:
As a side note, the Weather Underground API is always queried for metric values.  indi-allsky converts the values based on the Temperature Display value in the config.

DamoclesSword, 2024-07-18:
Awesome. With a little help from ChatGPT, I got it figured out. I have only a very low level understanding of all of this, so maybe more experienced users wouldnt struggle. Either way, adding to the wiki cant hurt! 

Is there anyway to keep the camera sensor in Celsius, but just have the weather API in Imperial? I know it is weird, but I am used to evaluating electronics on the 0-100C scale.

aaronwmorris, 2024-07-18:
Not without manually editing the code.

aaronwmorris, 2024-07-18:
Nevermind, I will work on this.  It should be pretty simple.

aaronwmorris, 2024-07-18:
Merged #1405 to allow all of the system temperatures to be available as a static unit.

https://github.com/aaronwmorris/indi-allsky/wiki/Image-Labels#system-temperature-sensors

DamoclesSword, 2024-07-30:
Wow! Sorry for the delay but it is super cool that you went ahead and did this. This might be a dumb question, but how do I implement this? Do I have to reinstall everything?

aaronwmorris, 2024-07-30:
You just need to update the code to the latest level.  Below are the instructions:

https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky

closed 2024-09-14 by aaronwmorris

ayryq, 2024-11-20:
Sorry to reopen this but I don't understand. Is there a way to show wind speed from wunderground in knots, for example?