aaronwmorris/indi-allsky #1409: Internal Server Error

Dentonknifeworks, 2024-07-19:
Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

webapp-indi-allsky.log shows
Jul 18 22:18:31 120NoIR [ERROR] MainProcess-1506 app.log_exception() [838]: Exception on /indi-allsky/ [GET]#012Traceback (most recent call last):#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1473, in wsgi_app#012    response = self.full_dispatch_request()#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 882, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 880, in full_dispatch_request#012    rv = self.dispatch_request()#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 865, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]#012  File "/home/pi/indi-allsky/indi_allsky/flask/misc.py", line 22, in decorated_view#012    return app.ensure_sync(func)(*args, **kwargs)#012  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/views.py", line 107, in view#012    self = view.view_class(  # type: ignore[attr-defined]#012  File "/home/pi/indi-allsky/indi_allsky/flask/base_views.py", line 301, in __init__#012    super(TemplateView, self).__init__(**kwargs)#012  File "/home/pi/indi-allsky/indi_allsky/flask/base_views.py", line 63, in __init__#012    self.getSunSetDate()#012  File "/home/pi/indi-allsky/indi_allsky/flask/base_views.py", line 195, in getSunSetDate#012    obs.elevation = self.camera.elevation#012TypeError: Elevation must be numeric

Looking for a suggestion Brad

aaronwmorris, 2024-07-19:
The `elevation` attribute for the camera is null.  If you restart indi-allsky from the command line, it should initialize the elevation as an integer.

I will put in a workaround as well.

aaronwmorris, 2024-07-19:
Merged #1410 as a fix in case elevation is null.

Dentonknifeworks, 2024-07-19:
Tried restart indi-allsky same issue

Dentonknifeworks, 2024-07-19:
I did an update with your change its working now...
Thanks

closed 2024-07-24 by Dentonknifeworks