aaronwmorris/indi-allsky #1411: Install fails on Stellarmate X

gordtulloch, 2024-07-19:
After the pip installs the setup.sh script fails with the error:

Package libindi was not found in the pkg-config search path.
Perhaps you should add the directory containing `libindi.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libindi' found

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...

Issue was resolved by manually installing vi sudo apt install libindi-dev

gordtulloch, 2024-07-19:
Also had to set the ports for Apache to 8080 and 4430 but that's because they run NGINX

aaronwmorris, 2024-07-19:
Interesting.  I figured StellarmateX would be based on Ubuntu.  Can you tell where the indi packages might be installed?

The setup script tries to detect the version of indilib by using the `pkg-config --modversion libindi` command, but obviously it is not finding it.

You can find this around line 1851 in the setup script.
```bash
# try to detect installed indiversion
#DETECTED_INDIVERSION=$(${INDI_DRIVER_PATH}/indiserver --help 2>&1 | grep -i "INDI Library" | awk "{print \$3}")
DETECTED_INDIVERSION=$(pkg-config --modversion libindi)
echo
echo
echo "Detected INDI version: $DETECTED_INDIVERSION"
sleep 3
```

You can see a commented out command that I used to use to detect the version.  You can try using the old command in the script, or just manually set the version in that variable.  It expects something like `2.0.8`

gordtulloch, 2024-07-19:
Stellarmate is Kubuntu 22,04 currently.  Would it be easier just to install libindi-dev when you do the apt package installs? As I said the easy fix was to just install libindi-dev manually and rerun setup.sh

aaronwmorris, 2024-07-20:
Yeah, give the manual install a shot.  I am not sure what to think about this.

closed 2024-09-14 by aaronwmorris