aaronwmorris/indi-allsky #1412: Moon mode gain is set to --gain -1 even though I have a value of 15

sonny486, 2024-07-20:
Moon Mode makes gain go to -1, even though I have a valid gain value of 15 in the system.

I can put a value of 1 and gain still defaults to -1.

Don't understand why.

Rebooted, reloaded, restarted, same result.

aaronwmorris, 2024-07-20:
What kind of camera are you using?

sonny486, 2024-07-20:
Imx477 - I looked at the log and it shows the system is calling for --gain -1 in the rpi-capture command. 

Can't seem to locate why it's calling for that.  

Interesting note, during the twilight hours before morning it went back to gain 15 which it was setup for.

sonny486.myqnapcloud.com/8086 if you wanna take a look.

sonny486, 2024-07-20:
Here is the log:

https://drive.google.com/file/d/1RjFM_UW5g-_GHAo-X2wj7Mdfp4Z2lpYA/view?usp=drive_link

In the log it looks like it went back to gain 15 after the day/night mode change.

sonny486, 2024-07-20:
Well no, it looks like the system rebooted at that time, at 7am local time Central.

aaronwmorris, 2024-07-20:
Can you paste the support info here, as well?

sonny486, 2024-07-20:
/home/Sonny/indi-allsky/misc/support_info.sh: line 61: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3883052 kB

System: Raspberry Pi 4 Model B Rev 1.1

Linux SkyEye 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

Time
Sat 20 Jul 15:31:10 CDT 2024

System timezone
US/Central

Uptime
 15:31:10 up  2:45,  2 users,  load average: 0.15, 0.22, 0.29

Memory
               total        used        free      shared  buff/cache   available
Mem:         3883052      882268     1919692       17812     1184080     3000784
Swap:        1048572           0     1048572

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1668352        0   1668352   0% /dev
tmpfs             388308     1320    386988   1% /run
/dev/mmcblk0p2 245599888 45476956 187629620  20% /
tmpfs            1941524      312   1941212   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
/dev/mmcblk0p1    522230    76402    445828  15% /boot/firmware
tmpfs             388304       40    388264   1% /run/user/1000

sysctl info
vm.swappiness = 60

system python: Python 3.11.2

indiserver: /usr/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:0d:42:31 brd ff:ff:ff:ff:ff:ff
    inet 192.168.86.39/24 brd 192.168.86.255 scope global dynamic noprefixroute eth0
       valid_lft 854288sec preferred_lft 854288sec
    inet6 2600:1700:a301:7f0::3d/128 scope global dynamic noprefixroute 
       valid_lft 2205sec preferred_lft 2205sec
    inet6 2600:1700:a301:7f0:15a9:e655:d0f6:6299/64 scope global dynamic noprefixroute 
       valid_lft 3371sec preferred_lft 3371sec
    inet6 fe80::e87d:81e5:c89e:199/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

User info
uid=1000(Sonny) gid=1000(Sonny) groups=1000(Sonny),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),115(lpadmin),993(gpio),994(i2c),995(spi),999(systemd-journal)

[... 1139 lines cut ...]
    "TSL2561_GAIN_DAY": 0,
    "TSL2561_INT_NIGHT": 1,
    "TSL2561_INT_DAY": 1,
    "TSL2591_GAIN_NIGHT": "GAIN_MED",
    "TSL2591_GAIN_DAY": "GAIN_LOW",
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS",
    "A_LABEL": "Sensor A",
    "B_LABEL": "Sensor B",
    "C_LABEL": "Sensor C",
    "WUNDERGROUND_APIKEY": "REDACTED",
    "WUNDERGROUND_APIKEY_E": "REDACTED"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-07-20:
Okay, I was able to recreate this on my system.  I am still investigating why it happens.

aaronwmorris, 2024-07-20:
Merged #1414 

-1 is just the default value of the gain variable, so it was obviously not being updated for moon mode at startup.

I recently refactored and updated some of the basic functions related to transitions and I did not catch this being a possible condition.  Thank you for opening this!

sonny486, 2024-07-21:
Cool I will update and let you know.  I am sure it will work.

sonny486, 2024-07-21:
Looks good!  Thanks.

closed 2024-07-21 by sonny486