aaronwmorris/indi-allsky #1416: Web only setup failing on DBUS_SESSION_BUS_ADDRESS test

gordtulloch, 2024-07-23:
I updated the remote install on my Ubuntu 22.04 server (no GUI) via a "git pull origin main" but forgot that there's a web only setup, so I ran setup.sh. This failed on if [ -z "$DBUS_SESSION_BUS_ADDRESS" ]; then so I went Doh! and ran the web_only_setup.sh, but it fails on the same error. I tried installing the dbus-x11 package even tho this is a headless server but the error remained the same. I tried commenting out the test to see what would happen and got the following:

**** Enabling services ****
Failed to connect to bus: $DBUS_SESSION_BUS_ADDRESS and $XDG_RUNTIME_DIR not defined (consider using --machine=<user>@.host --user to connect to bus of other user)

Not sure where to go here, my remote server is still operating but I need to update it. Not urgent, but any advice would be appreciated thanks!

aaronwmorris, 2024-07-23:
Are you running your session through screen, tmux, or byobu?  I had someone using byobu and it caused the systemd environment variables to not be passed to the session.

gordtulloch, 2024-07-25:
Nope just putty

aaronwmorris, 2024-07-25:
The only package that should be necessary is `dbus-user-session`.  Did you reboot your VPS after receiving this error?

aaronwmorris, 2024-08-10:
@gordtulloch Did you get this worked out?

closed 2024-09-14 by aaronwmorris