aaronwmorris/indi-allsky #1417: Duplicated cameras in dropdown?

alexdumitriu, 2024-07-24:
Just a quick question - is there a way I can remove any of these cameras from the list here? The only one I actually have is the first one in the list. The moment I try to change to a different one from the list, the app crashes, so I'd like to avoid making a mistake when clicking on the cameras list, select an invalid one and then crashing the app.

[image]

Thanks a lot
Alex

aaronwmorris, 2024-07-24:
You can hide cameras using SQL from the command line.

https://github.com/aaronwmorris/indi-allsky/wiki/SQL-queries

alexdumitriu, 2024-07-24:
Amazing, thank you very much!

closed 2024-07-24 by alexdumitriu