aaronwmorris/indi-allsky #1418: Updating to the latest version - includiong indi - and setup.sh is failing.

vastmeridian, 2024-07-25:
Hi,

Upgrading indi-allsky to the latest version today and I'm getting the following error when setup.sh is running:

```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 indi-full : Depends: indi-avalonud but it is not installable
E: Unable to correct problems, you have held broken packages.

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...
```

aaronwmorris, 2024-07-25:
What distribution are you using?

aaronwmorris, 2024-07-25:
For the time being, I would avoid upgrading the INDI software.  Sometimes the INDI repo has broken dependencies.

vastmeridian, 2024-07-26:
> What distribution are you using?
Ubuntu 20.04 LTS

> For the time being, I would avoid upgrading the INDI software. Sometimes the INDI repo has broken dependencies. These are usually temporary, lasting a few days.
Taking your advice here and things work.  Therefore the issue lies with the indi installation, not allsky.  Thanks Aaron

vastmeridian, 2024-07-26:
So worth upgrading Allsky - my goodness you've added a lot of functionality!  Thanks.

vastmeridian, 2024-08-01:
Not going so well.  My indi-server is borked.  The camera isn't recognised when indi-allsky starts.  I've tested the camera on another system and it's OK, but I'm getting 'Camera not found' within indi-allsky.
indi-server is there, but I can't get through to the setup part to set up the camera.  The script 'Detected INDI version: 2.0.8'.
Is there a way to remove indi and re-install it (maybe not via indi-full, but with only the packages I need)?

aaronwmorris, 2024-08-01:
You can remove the indi-full package and only install the packages you need.  "indi-full" is really just a meta-package with dependencies for other packages to force them to install.  What camera do you have?

The other option is to entirely remove the indi PPA and compile indi from scratch.  https://github.com/aaronwmorris/indi-allsky/wiki/Uninstall-indi-repository  Let me know if you go this route, there are a couple of extra steps you will need to take to finish this.

vastmeridian, 2024-08-01:
Thanks Aaron.
I'm tempted to compile from scratch.  I have an Altair camera so I need indi_altair_ccd.  I'll take a look at the wiki link you posted.

I'm running : ubuntu 20.04 LTS 64-bit

vastmeridian, 2024-08-02:
> You could try the following. Altair cameras are part of the ToupTek family.
> 
> ```
> sudo apt-get uninstall indi-full
> 
> sudo apt-get install \
>  libindi-dev \
>  libaltaircam \
>  indi-toupbase \
>  libtoupcam
> ```

Thanks - I'll give this sequence a go later and let you know.

tjzepf, 2024-08-03:
@aaronwmorris I'm in the same boat. Clean install is broken and looks to be indi, not indi-allsky. How do I build for an ASI676MC? Thanks!

aaronwmorris, 2024-08-03:
@tjzepf You may want to opt for Ubuntu 22.04 or Debian 12 instead of 24.04.  I need to re-test with INDI 2.0.9, but the last time I tried, I experienced segfaults with Ubuntu 24.04 and pyindi-client.

The ASI676MC requires a minimum version of INDI 2.0.7 which has the support for the 676.

tjzepf, 2024-08-03:
Hmm... I tried 22.04 and still get the same failures:

Aug  3 17:41:08 polaris [INFO] MainProcess-12444/MainThread allsky._queueManualTasks() [1211]: Checking for manually submitted tasks
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception: Traceback (most recent call last):
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception:   File "/home/tjz/indi-allsky/indi_allsky/capture.py", line 175, in run
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception:     self.saferun()
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception:   File "/home/tjz/indi-allsky/indi_allsky/capture.py", line 187, in saferun
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception:     self._initialize()
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception:   File "/home/tjz/indi-allsky/indi_allsky/capture.py", line 704, in _initialize
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception:     ccd_info = self.indiclient.getCcdInfo()
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception:   File "/home/tjz/indi-allsky/indi_allsky/camera/indi.py", line 527, in getCcdInfo
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception:     ctl_CCD_EXPOSURE = self.get_control(self.ccd_device, 'CCD_EXPOSURE', 'number')
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception:                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception:   File "/home/tjz/indi-allsky/indi_allsky/camera/indi.py", line 1329, in get_control
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception:     raise TimeOutException('Timeout finding control {0}'.format(name))
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception: indi_allsky.exceptions.TimeOutException: Timeout finding control CCD_EXPOSURE
Aug  3 17:41:21 polaris [ERROR] MainProcess-12444/MainThread allsky._startCaptureWorker() [333]: Capture worker exception: 
Aug  3 17:41:21 polaris [INFO] MainProcess-12444/MainThread allsky._startCaptureWorker() [340]: Starting Capture-16 worker
Aug  3 17:41:21 polaris [INFO] MainProcess-12444/MainThread allsky._queueManualTasks() [1211]: Checking for manually submitted tasks
Aug  3 17:41:21 polaris [INFO] Capture-16-12661/MainThread indi.__init__() [122]: creating an instance of IndiClient
Aug  3 17:41:21 polaris [INFO] Capture-16-12661/MainThread indi.__init__() [129]: PyIndi version: 2.0.9
Aug  3 17:41:21 polaris [INFO] Capture-16-12661/MainThread capture._initialize() [564]: Connecting to indiserver
Aug  3 17:41:21 polaris [INFO] Capture-16-12661/MainThread indi.serverConnected() [349]: Server connected (localhost:7624)
Aug  3 17:41:21 polaris [INFO] Capture-16-12661/Dummy-4 indi.newDevice() [217]: new device Telescope Simulator
Aug  3 17:41:21 polaris [INFO] Capture-16-12661/Dummy-4 indi.newDevice() [217]: new device ZWO CCD

aaronwmorris, 2024-08-04:
<sigh>  Let me run some tests.

aaronwmorris, 2024-08-04:
@tjzepf I have run some tests with INDI 2.0.9 and I have not experienced any issues to date.  I have tested on Debian 12 (arm64) and Ubuntu 22.04 (x86_64).  My ASI678MC seems to function fine.

I do see an indication of a problem:

```
Aug  3 17:41:21 polaris [INFO] Capture-16-12661/Dummy-4 indi.newDevice() [217]: new device ZWO CCD
```

It only says `ZWO CCD` and not the full device name.  This means that the camera was not really detected.  There is one thing that can cause this... if you unplugged and replugged the camera, but you did not restart the indiserver, the camera will not work.  The indiserver is not resilient to physical disconnects.

tjzepf, 2024-08-04:
@aaronwmorris , that was it. Doh! I rebooted and sure enough, it's working now. Thank you for your time and expertise, much appreciated!

vastmeridian, 2024-08-08:
> > You could try the following. Altair cameras are part of the ToupTek family.
> > ```
> > sudo apt-get uninstall indi-full
> > 
> > sudo apt-get install \
> >  libindi-dev \
> >  libaltaircam \
> >  indi-toupbase \
> >  libtoupcam
> > ```
> 
> Thanks - I'll give this sequence a go later and let you know.

Tried this today and ... Phew!  We're back online.
[image]

From my perepsctive I'm working.  If @tjzepf is working too, then we can close this.

aaronwmorris, 2024-08-09:
Fantastic!

closed 2024-08-09 by aaronwmorris