aaronwmorris/indi-allsky #1419: Nothing is generating now

Dentonknifeworks, 2024-07-25:
[image]
I have been having an issue that nothing is being generated.
If I go to generate I get error XXX mpg exist.
I check on the PI and yes it is there but its not a valid MPG it wont play.
Yesterday I decided to clean out everything using the Utilities tab flushed everything.
That did not help.
I looked in the log for errors and not seeing anything that looks like would cause the issue. Just a few smoke data errors.
I did a update yesterday per the WIKI instructions no issues with that.
Need ideas as to what to do. Is there another log I am not seeing?
Thanks
Brad

aaronwmorris, 2024-07-25:
Can you provide the support info here?

Dentonknifeworks, 2024-07-25:
Telescope Simulator.TELESCOPE_PARK.PARK=On
Telescope Simulator.TELESCOPE_PARK.UNPARK=Off
Telescope Simulator.TELESCOPE_PARK_POSITION.PARK_HA=0
Telescope Simulator.TELESCOPE_PARK_POSITION.PARK_DEC=33.09999847412109375
Telescope Simulator.TELESCOPE_PARK_OPTION.PARK_CURRENT=Off
Telescope Simulator.TELESCOPE_PARK_OPTION.PARK_DEFAULT=Off
Telescope Simulator.TELESCOPE_PARK_OPTION.PARK_WRITE_DATA=Off
Telescope Simulator.TELESCOPE_PARK_OPTION.PARK_PURGE_DATA=Off
Telescope Simulator.TELESCOPE_PIER_SIDE.PIER_WEST=Off
Telescope Simulator.TELESCOPE_PIER_SIDE.PIER_EAST=On
Telescope Simulator.APPLY_SCOPE_CONFIG.SCOPE_CONFIG1=On
Telescope Simulator.APPLY_SCOPE_CONFIG.SCOPE_CONFIG2=Off
Telescope Simulator.APPLY_SCOPE_CONFIG.SCOPE_CONFIG3=Off
Telescope Simulator.APPLY_SCOPE_CONFIG.SCOPE_CONFIG4=Off
Telescope Simulator.APPLY_SCOPE_CONFIG.SCOPE_CONFIG5=Off
Telescope Simulator.APPLY_SCOPE_CONFIG.SCOPE_CONFIG6=Off
Telescope Simulator.USEJOYSTICK.ENABLE=Off
Telescope Simulator.USEJOYSTICK.DISABLE=On
Telescope Simulator.SNOOP_JOYSTICK.SNOOP_JOYSTICK_DEVICE=Joystick
Telescope Simulator.TELESCOPE_TIMED_GUIDE_NS.TIMED_GUIDE_N=0
Telescope Simulator.TELESCOPE_TIMED_GUIDE_NS.TIMED_GUIDE_S=0
Telescope Simulator.TELESCOPE_TIMED_GUIDE_WE.TIMED_GUIDE_W=0
Telescope Simulator.TELESCOPE_TIMED_GUIDE_WE.TIMED_GUIDE_E=0
Telescope Simulator.GUIDE_RATE.GUIDE_RATE_WE=0.5
Telescope Simulator.GUIDE_RATE.GUIDE_RATE_NS=0.5
Telescope Simulator.ACTIVE_DEVICES.ACTIVE_GPS=GPS Simulator
Telescope Simulator.ACTIVE_DEVICES.ACTIVE_DOME=Dome Simulator
Telescope Simulator.DOME_POLICY.DOME_IGNORED=On
Telescope Simulator.DOME_POLICY.DOME_LOCKS=Off
Telescope Simulator.TELESCOPE_INFO.TELESCOPE_APERTURE=10
Telescope Simulator.TELESCOPE_INFO.TELESCOPE_FOCAL_LENGTH=10
Telescope Simulator.TELESCOPE_INFO.GUIDER_APERTURE=120
Telescope Simulator.TELESCOPE_INFO.GUIDER_FOCAL_LENGTH=900
Telescope Simulator.SCOPE_CONFIG_NAME.SCOPE_CONFIG_NAME=indi-allsky
Telescope Simulator.USEJOYSTICK.ENABLE=Off
Telescope Simulator.USEJOYSTICK.DISABLE=On
Telescope Simulator.SNOOP_JOYSTICK.SNOOP_JOYSTICK_DEVICE=Joystick
Telescope Simulator.MOUNT_TYPE is write-only
Telescope Simulator.SIM_PIER_SIDE is write-only
Telescope Simulator.MOUNT_MODEL is write-only
Telescope Simulator.MOUNT_AXES.PRIMARY=1.008622319886853802e-06
Telescope Simulator.MOUNT_AXES.SECONDARY=33.099997999999999365
Telescope Simulator.FLIP_HA is write-only

libcamera: not detected

libcamera packages
ii  libcamera-apps                       1.2.1-1                          armhf        libcamera-apps
ii  libcamera-tools                      0~git20230124+9b860a66-1         armhf        complex camera support library (tools)
ii  libcamera0:armhf                     0~git20230720+bde9b04f-1         armhf        complex camera support library

libcamera cameras
libcamera-hello: /usr/bin/libcamera-hello
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
                             2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
                             4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]

python packages
ii  libpython3-dev:armhf                 3.9.2-3                          armhf        header files and a static library for Python (default)
ii  libpython3-stdlib:armhf              3.9.2-3                          armhf        interactive high-level object-oriented language (default python3 version)
ii  libpython3.9:armhf                   3.9.2-1+rpi1                     armhf        Shared Python runtime library (version 3.9)
ii  libpython3.9-dev:armhf               3.9.2-1+rpi1                     armhf        Header files and a static library for Python (v3.9)
ii  libpython3.9-minimal:armhf           3.9.2-1+rpi1                     armhf        Minimal subset of the Python language (version 3.9)
ii  libpython3.9-stdlib:armhf            3.9.2-1+rpi1                     armhf        Interactive high-level object-oriented language (standard library, version 3.9)
ii  python-apt-common                    2.2.1                            all          Python interface to libapt-pkg (locales)
ii  python-is-python3                    3.9.2-1                          all          symlinks /usr/bin/python to python3
[... 737 lines cut ...]
    "B_I2C_ADDRESS": "0x76",
    "C_CLASSNAME": "",
    "C_PIN_1": "D16",
    "C_USER_VAR_SLOT": 20,
    "C_I2C_ADDRESS": "0x40",
    "OPENWEATHERMAP_APIKEY": "REDACTED",
    "OPENWEATHERMAP_APIKEY_E": "REDACTED",
    "WUNDERGROUND_APIKEY": "REDACTED",
    "WUNDERGROUND_APIKEY_E": "REDACTED",
    "MQTT_PASSWORD": "REDACTED",
    "MQTT_PASSWORD_E": "REDACTED"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  },
  "GPS_TIMESYNC": false
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-07-25:
Check your `/var/log/syslog` and `/var/log/kern.log` to see if you see any messages related to segfaults.

Dentonknifeworks, 2024-07-25:
None found

aaronwmorris, 2024-07-25:
Ouch.

Dentonknifeworks, 2024-07-25:
Such as life LOL

closed 2024-07-25 by Dentonknifeworks

aaronwmorris, 2024-07-25:
Have you checked your thermals?  Has your fan failed and perhaps it is rebooting due to heat?

Dentonknifeworks, 2024-07-25:
I did check them and they looked good.