aaronwmorris/indi-allsky #1428: Complete crash of indiallsky last night due to Capture-3 process in syslog "sqlite3.OperationalError: database is locked" - indi_v2024.05.1-636-gd0c47508

gaitskell, 2024-07-27:
Last night indiallsky crashed completely. Running on raspi4. The indi-allsky logfile (see below) just stopped with last entry being the "S3 bucket upload" - no errors shown. 

The syslog shows crash occurred in the Capture-3 process (Jul 26 22:57:21 PiAllSky python3). Cause was that the "sqlite3.OperationalError: database is locked" (see below).

Additonally, according to GCP S3 file list the available fits files uploads had actually stopped successfully appearing in GCP an hour prior to the complete crash. Files were stored locally ion raspi. No upload errors were being reported in the indi-allsky log - looked for [ERROR] or [WARNING]. 

I should acknowlege we are running indi-allsky pretty hard with 1 sec exposures every 5 secs, uploading 2.6 MB fits to GCP.  Using hard ethernet connection with more than adequate bandwidth. 

I used './misc/upload_sync.py sync' to get the last hour of files uploaded to GCP.

Can you see why this major crash might have not been caught? I can provide any additonal log files you need.

As a secondary questions - should the GCP uploads show in log if unsuccessful? 

Should I modify our setup to reduce errors?

Thanks, 

DETAILS FOLLOW:

Version:
```
$ git describe --tags
indi_v2024.05.1-636-gd0c47508
```

/var/log/indi-allsky/indi-allsky.log-20240727
 The last entries in logfile were :
```
Jul 26 22:57:24 PiAllSky [INFO] Image-3-19622/MainThread processing.apply_image_circle_mask() [1479]: Image circle mask in 0.0157 s
Jul 26 22:57:24 PiAllSky [INFO] Image-3-19622/MainThread image.processImage() [625]: Image processed in 5.2085 s
Jul 26 22:57:24 PiAllSky [INFO] Image-3-19622/MainThread image.write_img() [1312]: Image compressed in 0.0270 s
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread miscDb.addImage() [200]: Adding image ccd_f3ee6189-ab30-44e0-a7a7-75ec905af83c/exposures/20240726/night/26_22/ccd1_20240726_225645.jpg to DB
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread miscDb.addThumbnail() [1010]: Adding thumbnail to DB: /var/www/html/allsky/images/ccd_f3ee6189-ab30-44e0-a7a7-75ec905af83c/exposures/20240726/night/26_22/thumbnails/6cfb21bc-dc00-47db-aca3-6c25598eecf3.jpg
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread miscUpload.s3_upload_asset() [478]: Uploading to S3 bucket
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread miscUpload.s3_upload_asset() [478]: Uploading to S3 bucket
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread processing.add() [520]: Image bits: 16, cfa: None
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread processing._detectBitDepth() [607]: Image max value: 25985
Jul 26 22:57:38 PiAllSky [WARNING] Image-3-19622/MainThread processing._detectBitDepth() [638]: *** DETECTED BIT DEPTH (15) IS DIFFERENT FROM CONFIGURED BIT DEPTH (16) ***
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread processing._detectBitDepth() [640]: Overriding bit depth to 16 bits
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread processing._apply_calibration() [704]: Searching for bad pixel map: gain -1, exposure >= 1.0, temp >= -273.1c
Jul 26 22:57:38 PiAllSky [WARNING] Image-3-19622/MainThread processing._apply_calibration() [722]: Temperature matched bad pixel map not found: -273.15c
Jul 26 22:57:38 PiAllSky [WARNING] Image-3-19622/MainThread processing._apply_calibration() [741]: Bad Pixel Map not found: ccd1 16bit 1.0000000s gain -1 bin 1 -273.15c
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread processing._apply_calibration() [753]: Searching for dark frame: gain -1, exposure >= 1.0, temp >= -273.1c
Jul 26 22:57:38 PiAllSky [WARNING] Image-3-19622/MainThread processing._apply_calibration() [771]: Temperature matched dark not found: -273.15c
Jul 26 22:57:38 PiAllSky [WARNING] Image-3-19622/MainThread processing._apply_calibration() [790]: Dark not found: ccd1 16bit 1.0000000s gain -1 bin 1 -273.15c
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread miscDb.addFitsImage() [687]: Adding fits image ccd_f3ee6189-ab30-44e0-a7a7-75ec905af83c/fits/20240726/night/26_22/ccd1_20240726_225647.fit to DB
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread image.write_fit() [1058]: fit filename: /var/www/html/allsky/images/ccd_f3ee6189-ab30-44e0-a7a7-75ec905af83c/fits/20240726/night/26_22/ccd1_20240726_225647.fit
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread miscUpload.s3_upload_asset() [478]: Uploading to S3 bucket
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread sqm.calculate() [21]: Exposure: 1.000000, gain: -1
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread sqm.calculate() [37]: Raw SQM average: 1658.06
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread sqm.calculate() [42]: Weighted SQM average: 18404.50
Jul 26 22:57:38 PiAllSky [ERROR] Image-3-19622/MainThread processing.debayer() [1006]: No bayer pattern detected
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread image.processImage() [374]: Image: 1392 x 1040
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread processing._calculate_8bit_adu() [883]: ADU average: 1588.9 (6)
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread stretch.main() [43]: Using image stretch mode 1
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread stretch.mode1_apply_gamma() [65]: Applying gamma correction
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread stretch.mode1_apply_gamma() [85]: Image gamma in 0.0127 s
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread stretch._get_image_stddev() [168]: Mean and std dev in 0.0737 s
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread stretch.mode1_adjustImageLevels() [94]: Mean: 19196.79, StdDev: 922.04
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread stretch.mode1_adjustImageLevels() [131]: Image levels in 0.0098 s
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread processing.contrast_clahe_16bit() [1395]: Performing 16-bit CLAHE contrast enhance
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread processing._convert_16bit_to_8bit() [1084]: Resampling image from 16 to 8 bits
Jul 26 22:57:38 PiAllSky [WARNING] Image-3-19622/MainThread image.recalculate_exposure() [1660]: New calculated exposure: 1.00000000
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread processing.apply_image_circle_mask() [1479]: Image circle mask in 0.0156 s
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread image.processImage() [625]: Image processed in 0.4400 s
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread image.write_img() [1312]: Image compressed in 0.0273 s
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread miscDb.addImage() [200]: Adding image ccd_f3ee6189-ab30-44e0-a7a7-75ec905af83c/exposures/20240726/night/26_22/ccd1_20240726_225647.jpg to DB
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread miscDb.addThumbnail() [1010]: Adding thumbnail to DB: /var/www/html/allsky/images/ccd_f3ee6189-ab30-44e0-a7a7-75ec905af83c/exposures/20240726/night/26_22/thumbnails/26f69d49-10c5-4fae-82d1-6967bd0323f0.jpg
Jul 26 22:57:38 PiAllSky [INFO] Image-3-19622/MainThread miscUpload.s3_upload_asset() [478]: Uploading to S3 bucket
[... 72 lines cut ...]
Jul 26 22:57:21 PiAllSky python3[19958]:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1412, in execute
Jul 26 22:57:21 PiAllSky python3[19958]:     return meth(
Jul 26 22:57:21 PiAllSky python3[19958]:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/sql/elements.py", line 516, in _execute_on_connection
Jul 26 22:57:21 PiAllSky python3[19958]:     return connection._execute_clauseelement(
Jul 26 22:57:21 PiAllSky python3[19958]:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1635, in _execute_clauseelement
Jul 26 22:57:21 PiAllSky python3[19958]:     ret = self._execute_context(
Jul 26 22:57:21 PiAllSky python3[19958]:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1844, in _execute_context
Jul 26 22:57:21 PiAllSky python3[19958]:     return self._exec_single_context(
Jul 26 22:57:21 PiAllSky python3[19958]:   File "/home/admin/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1984, in _exec_single_context
Jul 26 22:57:21 PiAllSky python3[19958]:     self._handle_dbapi_exception(
[... cut ...]
Jul 26 22:57:22 PiAllSky python3[19958]: [SQL: UPDATE state SET "createDate"=?, value=? WHERE state."key" = ?]
Jul 26 22:57:22 PiAllSky python3[19958]: [parameters: ('2024-07-26 22:56:49.468698', '1722049009', 'WATCHDOG')]
Jul 26 22:57:22 PiAllSky python3[19958]: (Background on this error at: https://sqlalche.me/e/20/e3q8)
Jul 26 23:17:01 PiAllSky CRON[25424]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Jul 26 23:27:42 PiAllSky rngd[526]: stats: bits received from HRNG source: 1100064
```

Let me know if you need any other log files

aaronwmorris, 2024-07-27:
Can you post the support info here.  I just want to double check your python modules.

aaronwmorris, 2024-08-10:
Sorry for taking so long on this, I have been investigating possible sources of this.

A while back, I wrote a test program (`testing/sqlite_flask_multithread.py`) to torture test the SQLite database because a bunch of users were experiencing locked databases.  The test script creates 100 threads that write to the same database simultaneously, and 10 threads that only read.  This very quickly exposed the database locking issue.  I ultimately found that increasing the busy_timeout to 10s allowed the 100 workers to work concurrently.

When I re-tested my script now, I was once again getting locked errors and connection pooling errors.  When I originally wrote the torture script, SQLAlchemy was at 1.4, where now 2.0 is released.  After much research, I discovered that while connection pooling was already the default for most databases in 1.4, it was not for SQLite.  Connection pooling was now a default for SQLite as of 2.0.

https://docs.sqlalchemy.org/en/20/changelog/changelog_20.html#change-2.0.0b1-sqlite

The default connection pool is 5, the first change I am going to make is increase the pool_size to 25 and see if that helps.

Merged #1448 to increase connection pool

aaronwmorris, 2024-09-14:
Closing this for now.  Please re-open if this is still an issue.

closed 2024-09-14 by aaronwmorris