aaronwmorris/indi-allsky #1431: find camera name

ecalandrini, 2024-07-28:
Hi! i have to usb camera connected (I know you disencourage the use of these type of cameras). how can i pass from one to the other?
indi webcam ccd connect directly to /dev/video0, i want to use video1 or 5. thank you for your help!

aaronwmorris, 2024-07-28:
Can you post the output of the support info view?

aaronwmorris, 2024-08-10:
@ecalandrini Did you manage to work through this?

ecalandrini, 2024-08-11:
Sorry I didn't understand what you need. Do you want the result of the
indi_getprop command?

aaronwmorris, 2024-08-11:
Under the Tools drop down, there is a `Support Info` view that has data about your system.  You can just paste that here.

ecalandrini, 2024-08-13:
thank you! here you are!

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command
substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3883064 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1
(2024-05-29) aarch64 GNU/Linux

Time
Tue 13 Aug 17:14:27 CEST 2024

System timezone
Europe/Madrid

Uptime
 17:14:27 up 17 min,  4 users,  load average: 0.10, 0.17, 0.11

Memory
               total        used        free      shared  buff/cache
available
Mem:         3883064      527000     2595872        1892      839568
3356064
Swap:         204796           0      204796

Filesystems
Filesystem     1K-blocks    Used Available Use% Mounted on
udev             1671304       0   1671304   0% /dev
tmpfs             388308    1204    387104   1% /run
/dev/mmcblk0p2  30109280 8506724  20054268  30% /
tmpfs            1941532     160   1941372   1% /dev/shm
tmpfs               5120      16      5104   1% /run/lock
/dev/mmcblk0p1    522230   64628    457602  13% /boot/firmware
tmpfs             388304       0    388304   0% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
49173

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group
default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group
default qlen 1000
    link/ether e4:5f:01:a5:b8:f4 brd ff:ff:ff:ff:ff:ff
[... 1339 lines cut ...]
    "TSL2561_GAIN_NIGHT": 1,
    "TSL2561_GAIN_DAY": 0,
    "TSL2561_INT_NIGHT": 1,
    "TSL2561_INT_DAY": 1,
    "TSL2591_GAIN_NIGHT": "GAIN_MED",
    "TSL2591_GAIN_DAY": "GAIN_LOW",
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS",
    "VEML7700_GAIN_NIGHT": "ALS_GAIN_1",
    "VEML7700_GAIN_DAY": "ALS_GAIN_1_8",
    "VEML7700_INT_NIGHT": "ALS_100MS",
    "VEML7700_INT_DAY": "ALS_100MS"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-09-14:
You can probably change the video device by using something like this syntax:

```json
{
    "PROPERTIES" : {},
    "SWITCHES" : {},
    "TEXT" : {
        "DEVICE_PORT" : {
            "PORT" : "/dev/video5"
        }
    }
}
```

closed 2025-02-20 by aaronwmorris