aaronwmorris/indi-allsky #1443: FTP transfer doens't work

giorgio-grag, 2024-08-08:
I configured the INDI-AllSky with Oculus StarLight on a Raspberry Pi 4, and it functions properly. However, I encounter issues when attempting to upload files to a web server using FTP. Manual uploads via command line with a standard FTP connection are successful, but using Python FTP or PyCURL FTP results in timeout or network errors. I am utilizing the standard port 21.

```
Aug  8 09:37:38 raspberrypi [INFO] MainProcess-5551/Upload-46 generic.put() [66]: Uploading /var/www/html/allsky/images/ccd_35205d82-dacc-43b7-ab43-17dd50483a83/exposures/20240808/day/08_09/ccd1_20240808_093736.jpg
Aug  8 09:37:38 raspberrypi [INFO] MainProcess-5551/Upload-46 pycurl_ftp.put() [100]: pycurl URL: ftp://ftp.grag.org:21//home/allsky/upload/allsky/image.jpg
Aug  8 09:37:46 raspberrypi [INFO] MainProcess-5551/MainThread allsky._queueManualTasks() [1211]: Checking for manually submitted tasks
Aug  8 09:37:49 raspberrypi [INFO] Capture-63-25694/MainThread capture.saferun() [218]: Camera last ready: 13.1s
Aug  8 09:37:49 raspberrypi [INFO] Capture-63-25694/MainThread capture.saferun() [219]: Exposure state: OK
Aug  8 09:37:49 raspberrypi [INFO] Capture-63-25694/MainThread capture.detectNight() [1315]: Sun altitude: 36.3
Aug  8 09:37:49 raspberrypi [INFO] Capture-63-25694/MainThread capture.detectNight() [1322]: Moon altitude: -4.5, phase 13.4%
Aug  8 09:37:49 raspberrypi [WARNING] Capture-63-25694/MainThread indi.getCcdTemperature() [884]: Camera temperature not supported
Aug  8 09:37:49 raspberrypi [INFO] Capture-63-25694/MainThread indi.getTelescopeRaDec() [874]: Telescope Coord: RA 5.58, Dec 42.30
Aug  8 09:37:53 raspberrypi [ERROR] MainProcess-5551/Upload-46 uploader.processUpload() [456]: Connection failure: Connection timed out after 15001 milliseconds
```

aaronwmorris, 2024-08-08:
This is a connection timeout, so it is unable to make the connection to the host or IP address.  I would just look for typos in the hostname.

Can you provide the data from the support info view?  All passwords will be redacted.

giorgio-grag, 2024-08-12:
> This is a connection timeout, so it is unable to make the connection to the host or IP address. I would just look for typos in the hostname.
> 
> Can you provide the data from the support info view? All passwords will be redacted.

I double check and use the data in the configuration (ctrl-c ctrl-v) and it works from command line
The support view info remaing in loading, here is the screenshoot of the configuration
[image]

aaronwmorris, 2024-08-12:
You are not the first to experience an issue like this.  I am starting to suspect that the issue is you are connecting via IPv6 and that requires the use the EPSV (Extended Passive Mode) with FTP.  Let me look at adding an option for this.

aaronwmorris, 2024-08-12:
Actually, try using the `PycURL Options` option to add the setting:

```json
{
    "FTP_USE_EPSV": 1
}
```

giorgio-grag, 2024-08-12:
> ```json
> {
>     "FTP_USE_EPSV": 1
> }
> ```

I try, but I always get a timeout error
[image]

aaronwmorris, 2024-08-12:
Make sure you reload/restart indi-allsky.

giorgio-grag, 2024-08-12:
> Make sure you reload/restart indi-allsky.

I try restart indi-all-sky and even if reaspberry restart

aaronwmorris, 2024-08-12:
Try adding the verbose option, maybe we can see what is going on in the background:

```json
{
    "VERBOSE": 1,
    "FTP_USE_EPSV": 1
}
```

giorgio-grag, 2024-08-12:
> ```json
> {
>     "VERBOSE": 1,
>     "FTP_USE_EPSV": 1
> }
> ```

Aug 13 00:02:31 raspberrypi python3[1500]: *   Trying 185.116.60.12...
Aug 13 00:02:31 raspberrypi python3[1500]: * TCP_NODELAY set
Aug 13 00:02:31 raspberrypi python3[1500]: *   Trying 2a0c:79c0:1:c::4...
Aug 13 00:02:31 raspberrypi python3[1500]: * TCP_NODELAY set
Aug 13 00:02:31 raspberrypi python3[1500]: * Immediate connect fail for 2a0c:79c0:1:c::4: Network is unreachable
Aug 13 00:02:32 raspberrypi python3[1500]: *   Trying 2a0c:79c0:1:c::4...
Aug 13 00:02:32 raspberrypi python3[1500]: * TCP_NODELAY set
Aug 13 00:02:32 raspberrypi python3[1500]: * Immediate connect fail for 2a0c:79c0:1:c::4: Network is unreachable
Aug 13 00:02:33 raspberrypi python3[1500]: *   Trying 2a0c:79c0:1:c::4...
Aug 13 00:02:33 raspberrypi python3[1500]: * TCP_NODELAY set
Aug 13 00:02:33 raspberrypi python3[1500]: * Immediate connect fail for 2a0c:79c0:1:c::4: Network is unreachable
Aug 13 00:02:34 raspberrypi python3[1500]: *   Trying 2a0c:79c0:1:c::4...
Aug 13 00:02:34 raspberrypi python3[1500]: * TCP_NODELAY set
Aug 13 00:02:34 raspberrypi python3[1500]: * Immediate connect fail for 2a0c:79c0:1:c::4: Network is unreachable
Aug 13 00:02:35 raspberrypi python3[1500]: *   Trying 2a0c:79c0:1:c::4...
Aug 13 00:02:35 raspberrypi python3[1500]: * TCP_NODELAY set
Aug 13 00:02:35 raspberrypi python3[1500]: * Immediate connect fail for 2a0c:79c0:1:c::4: Network is unreachable
Aug 13 00:02:36 raspberrypi python3[1500]: *   Trying 2a0c:79c0:1:c::4...

here is the problem: the application try to reach the ftp server using the IP instead the name. It is on a shared server and works only with the name ftp.grag.org

aaronwmorris, 2024-08-12:
The IP is perfectly normal, I am wondering if it is trying to connect via IPv6 and your system is not able to connect via IPv6.

Can you try to just put in the IPv4 address of `185.116.60.12` in the hostname and see if that works?

giorgio-grag, 2024-08-13:
> The IP is perfectly normal, I am wondering if it is trying to connect via IPv6 and your system is not able to connect via IPv6.
> 
> Can you try to force IPv4?
> 
> ```json
> {
>     "VERBOSE": 1,
>     "IPRESOLVE": 1
> }
> ```

Fantastic, it's working now! Thank you so much. I believe I can remove the verbose options.

aaronwmorris, 2024-08-13:
That is great news!  I think I will add options for this directly in the interface.

aaronwmorris, 2024-08-15:
Merged #1458 with explicit settings.  You do not have to alter your current config method, it will still work.

closed 2024-08-21 by aaronwmorris