aaronwmorris/indi-allsky #1449: Improvements ideas.

giangitesta, 2024-08-10:
Hi @aaronwmorris ,
i would like to submit if possible a couple of improvements that can increase the ability to make Indi AllSky interoperate with other software.
The first is the addition of a LastWillMessage with related topic in the MQTT functionality. This would allow to understand when IndiAllSky is online and therefore the data posted in the broker is up to date.
The second is the addition of a parameter to set the altitude to consider the Moon above the site's horizon (aka Moon Mode). For example, in my observatory the geographical features screen the Moon when it is below a certain altitude. Perhaps you might consider entering a different altitude for the east and one for the west.
Otherwise I use the software a lot and think it is outstanding.
Best regards
G

aaronwmorris, 2024-08-10:
I like the idea of the LastWillMessage, that should be easy enough to manage.

There is already a setting to indicate the altitude of the moon for starting moon mode.  It is on the Location tab at the bottom.  I am not sure about having separate east/west settings.  Because the moon's rise/set varies a lot in azimuth, it would be limited to situations where the entire west or east is occluded and not partially occluded.

aaronwmorris, 2024-08-11:
Unfortunately, I do not think I can implement the LastWillMessage in indi-allsky due to the way I implemented the MQTT functionality.

indi-allsky is a multi-process/multi-threaded application.  Image capture, image processing, file transfers all happen in separate threads.  This is because doing the file transfers programmatically, if you do not catch all possible exceptions, you program can just stop... and there are a lot of errors that can happen with file transfers.  This prevents file transfer error from causing capture and image processing failures.

The file transfer threads handle most of the network I/O, and a thread may handle an SFTP transfer for one transaction, then an S3 transfer, then an MQTT transfer.  Each transfer is an exclusive transaction, meaning there are no persistent connections.  Each MQTT message that is sent happens under a new connection when is then closed after it is sent.  The LastWillMessage assumes that a persistent connection is maintained for the lifetime of the application.

The best I can do in this case is just publish a topic with the latest date so you know the data is fresh.

Vertikar, 2024-08-12:
> The LastWillMessage assumes that a persistent connection is maintained for the lifetime of the application.
> 
> The best I can do in this case is just publish a topic with the latest date so you know the data is fresh.

I was curious about this, as I'd read about the LastWill message a few times before and didn't fully understand it, so I had a quick dig, and it looks like you might be able to implement it:

> The Will Message MUST be published after the Network Connection is subsequently closed and either the Will Delay Interval has elapsed or the Session ends, unless the Will Message has been deleted by the Server on receipt of a DISCONNECT packet with Reason Code 0x00 (Normal disconnection) or a new Network Connection for the ClientID is opened before the Will Delay Interval has elapsed [MQTT-3.1.2-8].
https://docs.oasis-open.org/mqtt/mqtt/v5.0/os/mqtt-v5.0-os.html#_Toc479576982

So if the connection is closed cleanly, the Will message is deleted. Alternatively, it could hit the Will Delay interval, which I guess could tie into the capture intrerval?

aaronwmorris, 2024-08-12:
Oh yeah, I could definitely implement the LastWill message, but based on how I implemented the MQTT connection, the LastWill functionality would only produce a result if it failed during the MQTT connection which is a very small window.  If indi-allsky fails, it will just not make any new connections, so the LastWill functionality would never be utilized.

closed 2024-09-14 by aaronwmorris