aaronwmorris/indi-allsky #1451: Upgrade Failing

StarGeezerPhil, 2024-08-12:
Hi Aaron, hope all is well.

I'm just trying to update my install and I'm getting failure messages as below:

```
Detected INDI version: 1.9.7
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyindi-client
  Cloning https://github.com/indilib/pyindi-client.git (to revision ffd939b) to /tmp/pip-install-y81fek66/pyindi-client_1fc000c3289f43068cad9887e8303bc9
  Running command git clone --filter=blob:none --quiet https://github.com/indilib/pyindi-client.git /tmp/pip-install-y81fek66/pyindi-client_1fc000c3289f43068cad9887e8303bc9
  WARNING: Did not find branch or tag 'ffd939b', assuming revision or ref.
  Running command git checkout -q ffd939b
  Resolved https://github.com/indilib/pyindi-client.git to commit ffd939b
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [41 lines of output]
      /home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/dist.py:289: InformationOnly: Normalizing 'v1.9.1' to '1.9.1'
        self.metadata.version = self._normalize_version(self.metadata.version)
      running egg_info
      creating /tmp/pip-pip-egg-info-i17qbuc5/pyindi_client.egg-info
      writing /tmp/pip-pip-egg-info-i17qbuc5/pyindi_client.egg-info/PKG-INFO
      writing dependency_links to /tmp/pip-pip-egg-info-i17qbuc5/pyindi_client.egg-info/dependency_links.txt
      writing top-level names to /tmp/pip-pip-egg-info-i17qbuc5/pyindi_client.egg-info/top_level.txt
      writing manifest file '/tmp/pip-pip-egg-info-i17qbuc5/pyindi_client.egg-info/SOURCES.txt'
      file PyIndi.py (for module PyIndi) not found
      reading manifest file '/tmp/pip-pip-egg-info-i17qbuc5/pyindi_client.egg-info/SOURCES.txt'
      adding license file 'LICENSE'
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-y81fek66/pyindi-client_1fc000c3289f43068cad9887e8303bc9/setup.py", line 99, in <module>
          setup(
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/__init__.py", line 108, in setup
          return distutils.core.setup(**attrs)
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 184, in setup
          return run_commands(dist)
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 200, in run_commands
          dist.run_commands()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 970, in run_commands
          self.run_command(cmd)
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/dist.py", line 945, in run_command
          super().run_command(command)
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 989, in run_command
          cmd_obj.run()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/command/egg_info.py", line 310, in run
          self.find_sources()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/command/egg_info.py", line 318, in find_sources
          mm.run()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/command/egg_info.py", line 544, in run
          self.prune_file_list()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/command/egg_info.py", line 610, in prune_file_list
          base_dir = self.distribution.get_fullname()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/_core_metadata.py", line 266, in get_fullname
          return _distribution_fullname(self.get_name(), self.get_version())
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/_core_metadata.py", line 284, in _distribution_fullname
          canonicalize_version(version, strip_trailing_zero=False),
      TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...
```

I've done a full update and also cleared the PIP cache as you mentioned in another thread, but no success here.

StarGeezerPhil, 2024-08-12:
Thanks @aaronwmorris I did try as part of the upgrade process - but it fails.

aaronwmorris, 2024-08-12:
You would need to compile a new version of the INDI library with `./misc/build_indi.sh`.  I am not sure how it is currently installed, so you may need to remove the version you currently have if it was installed from a repository.  Do you know if the indiserver is installed at /usr/bin/ or /usr/local/bin/ ?

StarGeezerPhil, 2024-08-12:
Thanks again @aaronwmorris, it was installed as part of the indi-allsky service (a while back now) and is located in `/usr/bin`

Running the `./misc/build_indi.sh` script quits as it's already installed.

Am I best to remove indi independently and then running that script to reinstall it?

aaronwmorris, 2024-08-12:
If indi was installed using packages, yes, it would be best to uninstall the indi related packages and then re-install via the build_indi.sh script.

If indi was custom compiled and installed to /usr as the PREFIX, then recompiling and installing over what is already there.

There is a variable in build_indi.sh `INSTALL_PREFIX` which you can modify to do this.

StarGeezerPhil, 2024-08-12:
I managed to uninstall libindi etc, then recompile with the misc script, but unfortunately hitting the same errors in the update of indi-allsky:

```
Detected INDI version: 2.0.9
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyindi-client
  Cloning https://github.com/indilib/pyindi-client.git (to revision 6f8fa80) to /tmp/pip-install-t86xulii/pyindi-client_1df7ce0abacb4a29999e441d040e86c6
  Running command git clone --filter=blob:none --quiet https://github.com/indilib/pyindi-client.git /tmp/pip-install-t86xulii/pyindi-client_1df7ce0abacb4a29999e441d040e86c6
  WARNING: Did not find branch or tag '6f8fa80', assuming revision or ref.
  Running command git checkout -q 6f8fa80
  Resolved https://github.com/indilib/pyindi-client.git to commit 6f8fa80
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [41 lines of output]
      /home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/dist.py:289: InformationOnly: Normalizing 'v1.9.1' to '1.9.1'
        self.metadata.version = self._normalize_version(self.metadata.version)
      running egg_info
      creating /tmp/pip-pip-egg-info-417jxio1/pyindi_client.egg-info
      writing /tmp/pip-pip-egg-info-417jxio1/pyindi_client.egg-info/PKG-INFO
      writing dependency_links to /tmp/pip-pip-egg-info-417jxio1/pyindi_client.egg-info/dependency_links.txt
      writing top-level names to /tmp/pip-pip-egg-info-417jxio1/pyindi_client.egg-info/top_level.txt
      writing manifest file '/tmp/pip-pip-egg-info-417jxio1/pyindi_client.egg-info/SOURCES.txt'
      file PyIndi.py (for module PyIndi) not found
      reading manifest file '/tmp/pip-pip-egg-info-417jxio1/pyindi_client.egg-info/SOURCES.txt'
      adding license file 'LICENSE'
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-t86xulii/pyindi-client_1df7ce0abacb4a29999e441d040e86c6/setup.py", line 99, in <module>
          setup(
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/__init__.py", line 108, in setup
          return distutils.core.setup(**attrs)
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 184, in setup
          return run_commands(dist)
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 200, in run_commands
          dist.run_commands()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 970, in run_commands
          self.run_command(cmd)
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/dist.py", line 945, in run_command
          super().run_command(command)
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 989, in run_command
          cmd_obj.run()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/command/egg_info.py", line 310, in run
          self.find_sources()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/command/egg_info.py", line 318, in find_sources
          mm.run()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/command/egg_info.py", line 544, in run
          self.prune_file_list()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/command/egg_info.py", line 610, in prune_file_list
          base_dir = self.distribution.get_fullname()
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/_core_metadata.py", line 266, in get_fullname
          return _distribution_fullname(self.get_name(), self.get_version())
        File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/setuptools/_core_metadata.py", line 284, in _distribution_fullname
          canonicalize_version(version, strip_trailing_zero=False),
      TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...
```

aaronwmorris, 2024-08-12:
Try upgrading the following package and try again.

```
source virtualenv/indi-allsky/bin/activate

pip3 install --upgrade packaging
```

StarGeezerPhil, 2024-08-12:
You're a star - thank you for that, absolutely brilliant - seems to have worked. Thank you Aaron

closed 2024-08-12 by StarGeezerPhil

Vertikar, 2024-10-12:
Ran into the same issue myself while trying to update, upgrading packaging fixed it for me too

aaronwmorris, 2024-10-12:
Merged #1568 to always upgrade the packaging module