aaronwmorris/indi-allsky #1452: Docker build fails at libcamera

vercastro, 2024-08-13:
Hello!

I am working on configuring the docker based instillation of the ind-allsky software. I am therefore following the guide in the wiki. For the most part there are no issues until towards the end of the "docker compose build" command run, where the following error is observed:

```
37.33 Cloning into '/home/allsky/Projects/src/libcamera'...
37.61 remote: Repository not found.
37.61 fatal: repository 'https://github.com/raspberrypi/libcamera=/' not found
------
failed to solve: process "/bin/sh -c ./build_libcamera.sh $LIBCAMERA_TAG $RPICAM_APPS_TAG" did not complete successfully: exit code: 128
```

When I check that URL I can see that it is indeed 404. https://github.com/raspberrypi/libcamera appears to be the correct URL.

I tried to find the script that calls this URL so I could resolve the issue, however I have as of this moment been unsuccessful.

The host for this install is amd64 based Debian 12.

As an aside, the purpose of this installation is to act as a remote host for public access of my allsky via the Sync API, thereby avoiding directly exposing the local RPi installation with the actual camera. I suspect that this means libcamera wouldn't be necessary anyways. Although I am definitely not the expert on this topic.

aaronwmorris, 2024-08-13:
Yeah, you can comment out the libcamera compile from the Docker file.  The latest version will not compile for some reason.

I will update the repo soon with this.

aaronwmorris, 2024-08-15:
Merged #1457 to disable the libcamera build by default

vercastro, 2024-08-15:
I presume you are referring to this line in Dockerfile.indi_base_debian12:
`RUN ./build_libcamera.sh $LIBCAMERA_TAG $RPICAM_APPS_TAG`

closed 2024-09-14 by aaronwmorris