aaronwmorris/indi-allsky #1460: Problem with square blocks in timelapse

Marchal1977, 2024-08-15:
Hi,

Noticed having square blocks in all of my timelapses (see attached screenshot of timelapse)
[image]

I changed the FFMPEG Bitrate from 5000K to 20000, 30000 or 40000K, but no luck.

Any idea?

Thanks Marchal

aaronwmorris, 2024-08-15:
Do you see those artifacts when viewing the video that comes out of indi-allsky, or is it when viewing the video on Facebook or another service?

Marchal1977, 2024-08-15:
The artifacts are already visible in indi-allsky itself, they are also visible in other players when viewed there.

aaronwmorris, 2024-08-15:
Try adding `-level 3.1` to the ffmpeg options on the Image tab in the config.  Restart indi-allsky and try generating again.  It might help, might not.

Marchal1977, 2024-08-15:
Going to try this, I will let you know the outcome!

Marchal1977, 2024-08-16:
Had alot of issues with re-generating (alot of QUEUED/FAILED attempts). After a full reboot/restart I got it working with one timelapse so far. I added -level 3.1 to the ffmpeg options. Now I will to wait until it makes a timelapse of its own the next day. Will keep you updated!

aaronwmorris, 2024-08-16:
The default settings require you to first delete the timelapse before you regenerate it again.  There is a setting on the Admin tab that allows you to directly overwrite and existing timelapse.

Marchal1977, 2024-08-16:
Cannot find that setting on the Admin tab...

aaronwmorris, 2024-08-16:
If its there, it would be at the top of that tab, under the expiration settings.  It was a recent update, so you may need to pull down the latest code.

Marchal1977, 2024-08-16:
[image]

aaronwmorris, 2024-08-16:
Look like I added the functionality in #1446 on August 9th.

Marchal1977, 2024-08-16:
[image]

this one?

aaronwmorris, 2024-08-16:
Affirmative.

aaronwmorris, 2024-08-17:
Fantastic news!  Let it run a few more days to make sure the clean result was not just a fluke.

Marchal1977, 2024-08-29:
Still clean! Thanks!

aaronwmorris, 2024-08-29:
Great news!  `-level 3.1` is now a default setting in indi-allsky.

closed 2024-08-29 by Marchal1977