aaronwmorris/indi-allsky #1461: Pi 5 "Sensor worker exception: RuntimeError: Cannot determine SOC peripheral base address"

bendenajones, 2024-08-16:
I had gpio working with "AllSky"; decided to check out "Indi-Allsky" and am getting the following error (I have a dew heater and fan devices installed and setup):

2024-08-15T22:06:42.396616-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception: 
2024-08-15T22:06:42.396575-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception: RuntimeError: Cannot determine SOC peripheral base address
2024-08-15T22:06:42.396541-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:     GPIO.setup(self.id, GPIO.IN)
2024-08-15T22:06:42.396506-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:   File "/home/matt/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm283x/pin.py", line 40, in init
2024-08-15T22:06:42.396474-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:     self._pin.init(mode=Pin.IN)
2024-08-15T22:06:42.396439-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:   File "/home/matt/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/digitalio.py", line 223, in direction
2024-08-15T22:06:42.396407-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:     ^^^^^^^^^^^^^^
2024-08-15T22:06:42.396373-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:     self.direction = Direction.INPUT
2024-08-15T22:06:42.396337-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:   File "/home/matt/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/digitalio.py", line 193, in __init__
2024-08-15T22:06:42.396306-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-08-15T22:06:42.396271-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:     self.pin = digitalio.DigitalInOut(pin1)
2024-08-15T22:06:42.396234-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:   File "/home/matt/indi-allsky/indi_allsky/devices/generic/gpioStandard.py", line 25, in __init__
2024-08-15T22:06:42.396197-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-08-15T22:06:42.396160-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:     self.gpio = a_gpio_class(self.config, pin_1_name=a_gpio_pin_1, invert_output=a_gpio_invert_output)
2024-08-15T22:06:42.396126-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:   File "/home/matt/indi-allsky/indi_allsky/sensor.py", line 207, in init_gpio
2024-08-15T22:06:42.396089-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:     self.init_gpio()
2024-08-15T22:06:42.396055-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:   File "/home/matt/indi-allsky/indi_allsky/sensor.py", line 110, in saferun
2024-08-15T22:06:42.396017-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:     self.saferun()
2024-08-15T22:06:42.395971-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:   File "/home/matt/indi-allsky/indi_allsky/sensor.py", line 96, in run
2024-08-15T22:06:42.395808-07:00 AllSky [ERROR] MainProcess-5186/MainThread allsky._startSensorWorker() [511]: Sensor worker exception: Traceback (most recent call last):

bendenajones, 2024-08-16:
Support Info:
/home/matt/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 4142224 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux AllSky 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

Time
Thu Aug 15 10:10:14 PM PDT 2024

System timezone
US/Pacific

Uptime
 22:10:14 up 29 min,  3 users,  load average: 0.41, 0.50, 0.38

Memory
               total        used        free      shared  buff/cache   available
Mem:         4142224     1230400     1160400       23040     1841184     2911824
Swap:        1048560           0     1048560

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1896144        0   1896144   0% /dev
tmpfs             414224     6288    407936   2% /run
/dev/mmcblk0p2  60846312 14327344  43408272  25% /
tmpfs            2071104      944   2070160   1% /dev/shm
tmpfs               5120       48      5072   1% /run/lock
tmpfs              76800        0     76800   0% /home/matt/allsky/tmp
/dev/mmcblk0p1    522230    76426    445804  15% /boot/firmware
tmpfs             414208      160    414048   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
49050

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 2c:cf:67:03:5e:cc brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:03:5e:cd brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.6/24 brd 192.168.0.255 scope global dynamic noprefixroute wlan0
       valid_lft 84640sec preferred_lft 84640sec
    inet6 fe80::b2ed:e7c8:618b:7f33/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
[... 1195 lines cut ...]
    "TSL2561_GAIN_NIGHT": 1,
    "TSL2561_GAIN_DAY": 0,
    "TSL2561_INT_NIGHT": 1,
    "TSL2561_INT_DAY": 1,
    "TSL2591_GAIN_NIGHT": "GAIN_MED",
    "TSL2591_GAIN_DAY": "GAIN_LOW",
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS",
    "VEML7700_GAIN_NIGHT": "ALS_GAIN_1",
    "VEML7700_GAIN_DAY": "ALS_GAIN_1_8",
    "VEML7700_INT_NIGHT": "ALS_100MS",
    "VEML7700_INT_DAY": "ALS_100MS"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  }
}

```
#################################
###     end support info      ###
#################################

bendenajones, 2024-08-16:
Thanks, I performed those steps and the error changed. I now get "GPIO busy" errors:

2024-08-16T08:25:36.573695-07:00 AllSky [INFO] MainProcess-2235/MainThread allsky._startSensorWorker() [518]: Starting Sensor-2485 worker
2024-08-16T08:25:36.573659-07:00 AllSky [ERROR] MainProcess-2235/MainThread allsky._startSensorWorker() [511]: Sensor worker exception: 
2024-08-16T08:25:36.573624-07:00 AllSky [ERROR] MainProcess-2235/MainThread allsky._startSensorWorker() [511]: Sensor worker exception: lgpio.error: 'GPIO busy'
2024-08-16T08:25:36.573584-07:00 AllSky [ERROR] MainProcess-2235/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:     raise error(error_text(v))
2024-08-16T08:25:36.573545-07:00 AllSky [ERROR] MainProcess-2235/MainThread allsky._startSensorWorker() [511]: Sensor worker exception:   File "/home/matt/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/lgpio.py", line 458, in _u2i

using gpioinfo for the pins I have specified (D22 for Dew, D27 for fan):

 line  22:     "GPIO22"         "lg"  output  active-high [used bias-disabled]
        line  23:     "GPIO23"       unused   input  active-high
        line  24:     "GPIO24"       unused   input  active-high
        line  25:     "GPIO25"       unused   input  active-high
        line  26:     "GPIO26"       unused   input  active-high
        line  27:     "GPIO27"         "lg"  output  active-high [used bias-disabled]

aaronwmorris, 2024-08-16:
You may just need to reboot.

bendenajones, 2024-08-16:
I did so and eventually the errors start showing up in the log; however, I think gpio is working as the fan is on. Heater is on even though I have "day time" heater disabled.

aaronwmorris, 2024-08-16:
You might have another service that has locked the GPIO ports or it is a desktop utility.

bendenajones, 2024-08-17:
When I stop the indi-allsky service, gpioinfo show that the target gpios are not in use; when I start the service gpioinfo shows the following: 

        line  21:     "GPIO21"       unused   input  active-high
        line  22:     "GPIO22"         "lg"  output  active-high [used bias-disabled]
        line  23:     "GPIO23"       unused   input  active-high
        line  24:     "GPIO24"       unused   input  active-high
        line  25:     "GPIO25"       unused   input  active-high
        line  26:     "GPIO26"       unused   input  active-high
        line  27:     "GPIO27"         "lg"  output  active-high [used bias-disabled]

I then start seeing the errors in the log file; I also believe that these errors are causing my WeatherUnderground to be used excesssively as within a couple minutes the WU website reports over 20 queries.

aaronwmorris, 2024-08-17:
Let me see if I can replicate this on my Pi5.  So far I have done almost all of my GPIO testing on my Pi5 system, so there is something else involved.

bendenajones, 2024-08-17:
I don't trust my SD card; I am getting a new one in today; will setup the OS fresh and then re-install indi-allsky and report back.

bendenajones, 2024-08-18:
Yes, nothing changed. Got the "Cannot determine SOC peripheral base address" error initially after fresh OS/indi-allsky installation; then performed the python gpio setup steps (noted by above) and now again get the "lgpio.error: 'GPIO busy" error in the log. My devices appear to working - though they are almost always both on when they shouldn't be. I have the heater disabled for day use but it's on. I have the Fan disabled for night use but it stays on all the time.

aaronwmorris, 2024-08-19:
I have not found a cause yet.  Can you show the output of the following command?

You may have to install lsof using `sudo apt-get install lsof`

```
sudo lsof -n | grep -i gpio
```

bendenajones, 2024-08-19:
I get nothing with that command (I had to install lsof on my x64 bookworm lite OS install). Removing the grep for gpio I do get output.

aaronwmorris, 2024-08-19:
Oh well.  Can you send me another support info dump?  I want to compare your python modules now that you swapped them.  I will be back in my office tomorrow where I have my Pi5.

bendenajones, 2024-08-19:
NP; Thanks
[Fresh _OS_indiallsky_install_configured.txt](https://github.com/user-attachments/files/16655590/Fresh._OS_indiallsky_install_configured.txt)

bendenajones, 2024-08-19:
I had success using gpiozero when developing a modified plug-in for allsky dew heater control. Just an fyi - it seems as if Pi made a mess of gpio w/ the 5 series.

bendenajones, 2024-08-24:
> Oh well. Can you send me another support info dump? I want to compare your python modules now that you swapped them. I will be back in my office tomorrow where I have my Pi5.

Did you replicate the issue?

aaronwmorris, 2024-08-25:
Not yet which is really strange.  I did almost all of my development of this code on Rpi5.

Can you send me the contents of your `/boot/firmware/config.txt`?

bendenajones, 2024-09-08:
Sorry for the delay - I am not getting notified of thread updates. Here is the file you requested.
[config.txt](https://github.com/user-attachments/files/16923131/config.txt)

bendenajones, 2024-09-15:
Some followup: I hooked up some test leds to the IOs configured for use for my dew heater and fan control: 22 and 27 respectively on my Pi 5. Using 'gpioset' command: 'gpioset 4 22=1' and 'gpioset 4 27=1' both work as expected (after stopping the indi all-sky service) to turn on the IOs.

bendenajones, 2024-09-15:
It would great if I was able to test ios via the device tab; a simple on/off or "pulse" capability would go a long ways to help trouble-shoot these types of issues.

I am trying to get the fan to turn off; I had to disable thresholds for the dew heater to get turned off which worked. I am seeing the following given my threshold setting for the fan:

check_fan_thresholds() [502]: Fan threshold current: 123.3, target: 140.0, delta: -16.7

If I read that correctly the fan should be turned off. The idea is that unless the target sensor temp (CPU in my case) is less than 140 it should turn off the fan?

bendenajones, 2024-09-15:
I am starting to think the gpio "busy" errors I reported earlier are related to restarting indi-allsky via teh "Save - reload on Save" option. I do not see those errors in the log until after I make config changes... still investigating.

bendenajones, 2024-09-16:
I can confirm that the "gpio busy" bug is related to "Save - reload on Save" function. I went over 4 hrs with no gpio errors; as soon as I performed a config update w/ reload option enabled - the errors show up in the log.

aaronwmorris, 2024-09-16:
Okay.... I will try to focus on the reload functionality.

aaronwmorris, 2024-09-16:
I am definitely seeing exceptions on reload.

aaronwmorris, 2024-09-17:
This was definitely a PITA to track down.  It appears the lgpio module which is used for the GPIO related functionality on the Raspberry Pi 5 may not be threadsafe.

indi-allsky uses the python multiprocessing library to split processing tasks between multiple processes.  There is also a mixture of threads for some workers where tasks are not CPU bound (network transfers, etc).  The sensor and GPIO functionality was implemented as of of those threads.  Threads are much less memory heavy than processes, so it is ideal in situations where there is more limited memory (like a Raspberry Pi 3 with 1GB of memory).

Unfortunately, the error you experienced appears to be caused by esoteric python functionality on how multiprocessing works.  When you spawn a separate Process, python copies all of the memory (and other things) from the parent process into the memory of the new process.  The sub-processes and threads were all handled by the main process thread so something was being maintained across process reloads.  (The reload restarts all of the sub-processes and threads, but the main process does not restart).

The TLDR is it appears that the locked pin from one of the threads was being persisted by one of the other processes when the threads and processes were restarted on a reload.

This also only happened when using lgpio which is only required on Raspberry Pi 5.  Raspberry Pi 4 uses RPi.GPIO normally which does not exhibit this behavior.  Swapping RPi.GPIO for lgpio on Rpi4 exposes the same problem.  There is something up with the lgpio module.

Merged #1511 to convert the sensor worker from thread based to process based.  This increases the memory a little, but there is really no other choice.

bendenajones, 2024-09-17:
Wow, I need to beef up on my python multi-threading knowledge - that's far more complex than C++, C# or any other language i know - but of course in these languages one must manage the TLS or cross-thread data yourself - something to be said for having full control of the functionality vs 4GLs like python.

Regardless happy you found the issue and the solution; appreciate it.

aaronwmorris, 2024-09-17:
indi-allsky was always an experiment to see if an allsky system could even be managed in Python.  All of the common criticisms of Python are generally true, mainly that it is slow.

What is really interesting about indi-allsky is while it is written in Python, most of the modules that provide the core functionality are backed by C/C++ code.   pyindi-client, opencv, numpy, etc are all C-based.  Python is just a glue between these C libraries.  Even the lgpio module is actually a SWIG based module (same as pyindi-client) which means the python interface is just a thin wrapper around the C library.

This may be part of the problem.  The fact that some of these modules are really C code may have strange interactions when using multiprocessing.

bendenajones, 2024-09-21:
Pulled change #1511 and the issue has not re-occured.

closed 2024-09-21 by bendenajones