aaronwmorris/indi-allsky #1477: Disable day/night timelapse but always save files

alexhelms, 2024-08-21:
I would like to disable timelapse generation but always save files. Currently if I disable daytime timelapse generation it does not save files. The pi takes way too long to generate timelapse files because of memory constraints with a 533 sensor so I'm going to just generate the timelapse myself with the files that get FTP'd out but this isn't possible because no files are saved if I disable timelapse. I'd like to request decoupling this behavior so that timelapse and saving files are separate configuration options.

aaronwmorris, 2024-08-22:
I have been thinking about this.  There is a downside to not generating a timelapse video.  The Timelapse Viewer interface, which encompasses Timelapse, Keograms, and Star Trails is all based on the existence of the timelapse video asset.  Without the video, the keogram and startrail will not show up in the interface.

If memory is a problem, you can always set the video scaling to 50% or even 25%.  The amount of memory needed to generate the timelapse is all based on the output resolution (the input files do not matter).  Setting to 50% will cut memory usage by approximately 50% and 25% will be reduced by 75%.

alexhelms, 2024-08-22:
What's wrong with not having any timelapses in the UI? If I turn off timelapses I would expect an empty timelapse UI. It appears there is already a config toggle to turn off timelapses. If I turn that off, does ti still save day and night images?

I'm using a 533 for the resolution so I do not want to scale anything. The real fix is to swap out the pi for one with more memory, but even then creating the timelapses with ffmpeg on another machine is not a problem.

aaronwmorris, 2024-08-25:
Merged #1479 which includes a new option to save daytime images, but generate no timelapse.

alexhelms, 2024-08-26:
Thank you, I will update soon and try this out. In the mean time I had just turned off timelapse entirely.