aaronwmorris/indi-allsky #1485: Exception during MTF Stretch

alexhelms, 2024-08-27:
Just pulled main to try out MTF stretch and I'm getting an exception when processing the image:

```
2024-08-27T14:30:38.607752-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception: 
2024-08-27T14:30:38.607509-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception: IndexError: index 65535 is out of bounds for axis 0 with size 65535
2024-08-27T14:30:38.607303-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception:                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-08-27T14:30:38.607087-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception:     stretched_image = self._mtf_lut.take(data, mode='raise')
2024-08-27T14:30:38.606872-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception:   File "/home/alex/indi-allsky/indi_allsky/stretch/mode2_mtf.py", line 66, in stretch
2024-08-27T14:30:38.606649-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception:                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-08-27T14:30:38.606469-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception:     stretched_image = self._stretch.stretch(self.image, self.max_bit_depth)
2024-08-27T14:30:38.606255-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception:   File "/home/alex/indi-allsky/indi_allsky/processing.py", line 2411, in stretch
2024-08-27T14:30:38.606041-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception:     self.image_processor.stretch()
2024-08-27T14:30:38.605836-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception:   File "/home/alex/indi-allsky/indi_allsky/image.py", line 490, in processImage
2024-08-27T14:30:38.605617-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception:     self.processImage(i_dict)
2024-08-27T14:30:38.605405-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception:   File "/home/alex/indi-allsky/indi_allsky/image.py", line 249, in saferun
2024-08-27T14:30:38.605119-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception:     self.saferun()
2024-08-27T14:30:38.604802-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception:   File "/home/alex/indi-allsky/indi_allsky/image.py", line 220, in run
2024-08-27T14:30:38.604293-07:00 indiallsky [ERROR] MainProcess-32807/MainThread allsky._startImageWorker() [388]: Image worker exception: Traceback (most recent call last):
```

I turned on MTF and changed midtones to 0.25.

alexhelms, 2024-08-27:
```
alex@indiallsky:~/indi-allsky $ git diff
diff --git a/indi_allsky/stretch/mode2_mtf.py b/indi_allsky/stretch/mode2_mtf.py
index 706a982e..dae12f89 100644
--- a/indi_allsky/stretch/mode2_mtf.py
+++ b/indi_allsky/stretch/mode2_mtf.py
@@ -41,7 +41,7 @@ class IndiAllSky_Mode2_MTF_Stretch(IndiAllSky_Stretch_Base):
             data_max = (2 ** image_bit_depth) - 1

-            range_array = numpy.arange(0, data_max, dtype=numpy.float32)
+            range_array = numpy.arange(0, data_max + 1, dtype=numpy.float32)
             shadows_val = int(self.shadows * data_max)
             highlights_val = int(self.highlights * data_max)
```

Seems to resolve the exception.

aaronwmorris, 2024-08-27:
I do not know why this did not come up in my testing.  Merged #1486 to fix.

closed 2024-09-14 by aaronwmorris