aaronwmorris/indi-allsky #1487: Pi HQ camera controlled remotely from Ekos is not working

quanticleap, 2024-08-29:
Hello,
when I connect from remote with Ekos, indiserver provide the "indi_simulator_ccd" and not the IMX 477 or libcamera or indi_rpicamera and photos are all black:
pi@astro:~ $ ps ax | grep indiserver
    802 ?        Ss     0:14 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_simulator_ccd

The goal is to put indi-allsky on ubuntu PC and control the indi-allsky on raspberry PI with Pi HQ camera, in order to unload the CPU of raspberry.
The Ubuntu PC with indi-allsky will act like client for indiserver on indi-allsky installed on raspberry pi 3 with Pi HQ camera connected.

Thanks

Next all the output support:
/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 929092 kB

System: Raspberry Pi 3 Model B Plus Rev 1.3

Linux astro 6.6.47-v8+ #1792 SMP PREEMPT Thu Aug 22 15:11:43 BST 2024 aarch64 GNU/Linux

Time
Thu 29 Aug 09:59:45 CEST 2024

System timezone
Europe/Rome

Uptime
 09:59:45 up  1:59,  1 user,  load average: 0.15, 0.05, 0.14

Memory
               total        used        free      shared  buff/cache   available
Mem:          929092      592328      209984        3220      186964      336764
Swap:        1073148       15616     1057532

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev              325148        0    325148   0% /dev
tmpfs              92912     1072     91840   2% /run
/dev/mmcblk0p2  60222412 19550088  37595556  35% /
tmpfs             464544      168    464376   1% /dev/shm
tmpfs               5120       12      5108   1% /run/lock
/dev/mmcblk0p1    522230    65994    456236  13% /boot/firmware
tmpfs              92908        4     92904   1% /run/user/1000

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
44546

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
[... 1165 lines cut ...]
    "TSL2561_GAIN_NIGHT": 1,
    "TSL2561_GAIN_DAY": 0,
    "TSL2561_INT_NIGHT": 1,
    "TSL2561_INT_DAY": 1,
    "TSL2591_GAIN_NIGHT": "GAIN_MED",
    "TSL2591_GAIN_DAY": "GAIN_LOW",
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS",
    "VEML7700_GAIN_NIGHT": "ALS_GAIN_1",
    "VEML7700_GAIN_DAY": "ALS_GAIN_1_8",
    "VEML7700_INT_NIGHT": "ALS_100MS",
    "VEML7700_INT_DAY": "ALS_100MS"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-08-29:
My INDI build script does not include indi_libcamera_ccd because it is not yet stable.  There is some manual patches to the libcamera source that is required.

You may try to build that yourself.  The source for indi should still be located at `~/Projects/src/indi_3rdparty/indi-libcamera`.  I have not attempted this build yet.  The other option would be to use indi_pylibcamera which is an alternative method for managing a libcamera camera remotely.  https://github.com/aaronwmorris/indi-allsky/wiki/indi_pylibcamera-setup

quanticleap, 2024-08-29:
Great, thank you for the answer, I will take a look to the documentation provided.
The idea is to control a remote PI Hq camera connected to raspberry from indi-allsky installed on Ubuntu server and unload the raspberrpy pi 3 CPU when doing timelapses.
With Altair astro, already is working in this way.
Consider the issue closed on my side.Thanks again!

aaronwmorris, 2024-08-29:
That makes perfect sense.  A few people have already done this with indi_pylibcamera, so hopefully it will work well for you.

We can keep this open if you have any other issues.  I will close it out after a few weeks.

quanticleap, 2024-08-29:
all working fine with indi_pylibcamera! 
Test done with Indi-all sky on Ubuntu and the PI HQ camera connected on remote Pi 3 with indi_pulibcamera installed.
I had just an issue, a library was missed "Couldn't open libGLESv2.so.2", when I issued the commands to verify that libcamera is fine before to use indi_pylibcamera:
 libcamera-still -r --mode 1332:990 --shutter 100000 --gain 1 --awbgains 1,1 --immediate -o test.jpg
and fixed with command:
sudo apt-get install libgles2-mesa

Thanks!Have a good day!

closed 2024-09-14 by aaronwmorris