aaronwmorris/indi-allsky #1491: unable to connect QHY5iii715c after building fxload for USB3

broz22, 2024-09-01:
The log indicated that the camera was found, another message indicating connecting to it, and another that it was using the qhy sdk. But no pictures cane up. I can try to attach the log and the config if needed. Thanks,
John

aaronwmorris, 2024-09-01:
Please do upload some of the log from `/var/log/indi-allsky/indi-allsky/.log`

Also, if you could upload the information from the Support Info view.

broz22, 2024-09-01:
[indi-allsky.log-20240830.gz](https://github.com/user-attachments/files/16829392/indi-allsky.log-20240830.gz)

Where do I find the support info? Thanks (and btw, your program and support are amazing).
John

aaronwmorris, 2024-09-01:
In the web view (after you login), you can find it under Tools -> Support Info.  You can also run `./misc/support_info.sh`

broz22, 2024-09-01:
It looked like an abnormal termination:
john@pi4-4g:~/indi-allsky/misc $ ./support_info.sh >support
./support_info.sh: line 62: warning: command substitution: ignored null byte in input
Cannot open device /dev/video0, exiting.
2024-09-01 12:24:03,288 [INFO] MainProcess config._dump() [1029]: Dumping config

output file zipped:
[support.gz](https://github.com/user-attachments/files/16829725/support.gz)

aaronwmorris, 2024-09-01:
Perfect.  That helped.

I think the problem is because you had the ASI camera connected first, you still have the ASI specific INDI config defined.  Set that back to the default, and everything should work.

```
{
    "PROPERTIES" : {},
    "SWITCHES" : {}
}
```

broz22, 2024-09-01:
Sorry to keep bothering you, but how do I reset to default? Thanks,
John

aaronwmorris, 2024-09-01:
Under the Camera tab, near the bottom, just set it to the text I commented above.

broz22, 2024-09-01:
That did it - many thanks. Now I just have to focus and set up all my config parameters for the new camera. Thanks again.
John

closed 2024-09-01 by broz22