aaronwmorris/indi-allsky #1492: Time lapse video "clipping" issue

AstroAlan, 2024-09-02:
Hello,
I'm hoping someone can help please.
Firstly well done to AM for a great (and free) piece of software. Had no real issues once I increased the memory swap size to 5gb. But I've been trying to sort an issue out for about a month now. I'm a novice with the raspberry pi so I'm now just guessing at the cause. When a timelaose has been created there's an artifact on the video that appears to be clipping the brightness of the video making it appear blocky and pixelated in patches. This dances around the screen as the video progresses.. Non of these artifacts are present on the individual frames taken throughout the night, so I believe it's a time lapse processing error/setting fault. 
Here's a Dropbox link to the time lapse video (340mb so exceeds the 10mb allowed to post here) and below the support info.

https://www.dropbox.com/scl/fi/xzemmgw6cisk6hcpcz0km/allsky-timelapse_ccd1_20240901_night-2.mp4?rlkey=v4ezhx1e9eqni1npkp3jsdhm3&st=q81ho60s&dl=0

Support info (Sorry it's so long!!!)

/home/alans/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 4142224 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux raspberrypi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

Time
Mon  2 Sep 07:00:23 BST 2024

System timezone
Europe/London

Uptime
 07:00:23 up 2 days, 12:25,  2 users,  load average: 0.05, 0.07, 0.09

Memory
               total        used        free      shared  buff/cache   available
Mem:         4142224      923984     2445792       34688      873824     3218240
Swap:        2097136      940032     1157104

Filesystems
Filesystem     1K-blocks      Used Available Use% Mounted on
udev             1896160         0   1896160   0% /dev
tmpfs             414224      6000    408224   2% /run
/dev/mmcblk0p2 491542480 162978220 303580784  35% /
tmpfs            2071104      1056   2070048   1% /dev/shm
tmpfs               5120        48      5072   1% /run/lock
/dev/mmcblk0p1    522230     76398    445832  15% /boot/firmware
tmpfs             414208       160    414048   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
39700

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
[... 1260 lines cut ...]
    "TSL2561_GAIN_NIGHT": 1,
    "TSL2561_GAIN_DAY": 0,
    "TSL2561_INT_NIGHT": 1,
    "TSL2561_INT_DAY": 1,
    "TSL2591_GAIN_NIGHT": "GAIN_MED",
    "TSL2591_GAIN_DAY": "GAIN_LOW",
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS",
    "VEML7700_GAIN_NIGHT": "ALS_GAIN_1",
    "VEML7700_GAIN_DAY": "ALS_GAIN_1_8",
    "VEML7700_INT_NIGHT": "ALS_100MS",
    "VEML7700_INT_DAY": "ALS_100MS"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-09-02:
indi-allsky balances the exposure by measuring the average brightness of the central 50% of the image.  Pixels outside the central region are not used for brightness calculations.  When you have those extremely bright clouds, they do not affect the exposure until they approach the central region of the image.

On the Image tab, you can see the default setting of 50% for `ADU FoV`.  You can expand this range to 66% or 100%.

A better option would be to use a detection mask, where you can draw arbitrary boundaries for the brightness calculations (and other functions).  https://github.com/aaronwmorris/indi-allsky/wiki/Detection-Masks

AstroAlan, 2024-09-04:
Hello, Thanks for your reply. Just an update.  I've tried altering the ADU FoV and I'm still getting the effect. Probably worse. I've looked at the mask but I keep getting a line on the images which I don't want. I'm going to try re-installing the software. I've been altering the settings to try and find a cure and may have changed some detrimentaly. 
ATB 
Alan

aaronwmorris, 2024-09-04:
Do you have an example image with the line?

AstroAlan, 2024-09-05:
Weird thing happening here. When I view the timelapse via the local network I get the artifacts, both on my Samsung  S24 or my HP laptop. When I load the video to Dropbox and view it from there the video is perfect. No flaws or clipping at all. How can that be? Don't understand that at all. 
Thanks 
Alan

aaronwmorris, 2024-09-08:
That makes more sense.  I just viewed the video on dropbox and I did not see the problem.  I just assumed the problem was the bright areas.

Try adding `-level 3.1` to the extra ffmpeg options and restart indi-allsky.  Other users have reported this fixes the artifacts in the videos.  This is actually now a default in indi-allsky, but you installed before it became a default option.

AstroAlan, 2024-09-10:
Hello,
That's cured it. I've done 2 nights timelapse and it's perfect. Many thanks for your help. It's very appreciated. It's a superb price of software. Well done. 

ATB
Alan

closed 2024-09-13 by aaronwmorris